dresden-elektronik/deconz-rest-plugin #3542: Develco Air Quality Sensor (AQSZB-110)

michael666nan, 2020-10-30:
## Device

- Product name: Air Quality Sensor
- Manufacturer: Develco Products A/S
- Model identifier: AQSZB-110
- Device type :
  - Sensor
  
## Screenshots
Basic cluster:
[image]
[image]

Power Configuration:
[image]
[image]

Identify:
[image]

Poll Control:
[image]

Temperature measurement:
[image]

Relative humidity measurement:
[image]

Node Info Panel:
[image]

cllatser, 2020-10-31:
I am looking for this product to buy, but Develco is a white label. Where did you buy it? Thanks a lot.

michael666nan, 2020-10-31:
Sorry I can't help since I did not buy these sensors myself, but received some in a project. Best regards.

OsteHovel, 2020-11-15:
I bought one from [Elektroimportøren ](https://www.elektroimportoren.no/develco-air-quality-sensor/4514806/Product.html) in Norway

closed 2020-11-15 by manup