dresden-elektronik/deconz-rest-plugin #3548: Can't find Phillips Hue Bulb

simplenotezy, 2020-10-31:
## Describe the bug
I have paired almost 10 bulbs, but theres ONE bulb that I removed from Hue app, and which I now want to connect to phoscon, however it is not showing up.

I have heard that I could use touchlink to reset it. Is this possible? if so, how do I find it?

I have tried resetting using remote, however, that makes it blink and seems like it's been reset, however, I still cant find it when searching. The stick is 5 centimers away from the bulb..

## Steps to reproduce the behavior
1. Reset the bulb (using the Hue App), by clicking delete
2. Search for the bulb -- no results
3. Powercycle the bulb. Perform a reset of the bulb using the Remote, holding on and off button for 10 seconds. Bulb will blink.
4. Reset the bulb (on/off 3 times) then perform reset of bulb using remote. Bulb will blink. No bulbs found.
5. Try to connect the bulb using Phillips Hue - success! No issues
6. Tried resetting via TouchLink - cannot be found.

## Expected behavior
Bulb should show up

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Homebridge
 - Firmware version: (26660700)
 - deCONZ version: Don't know where to find this in the Phoscon Web UI. I installed yesterday so must be latest.
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No.

## deCONZ Logs
Not sure what logs you would want; I am happy to provide logs if I am given instructions on how to do so on a Homebridge device.

## Additional context
I can pair it using phillips hue app/bridge.

simplenotezy, 2020-10-31:
Something really weird; while I was pairing a Motion Sensor and a Remote from IKEA, I had to put them into pairing mode. Once they were succesfully paired using the Phoscon Web UI the Phillips Hue Bulb that I cannot seem to Find, was starting to blink 2-3 times.

Mimiix, 2020-11-02:
What happens if you turn the hue bridge off?

simplenotezy, 2020-11-05:
@Mimiix it turned out that the light was already linked. Just renamed another bulb and for some reason that was the wrong bulb - so it had always been there just hiding in diquise under a different name!

Case closed ;)

Mimiix, 2020-11-05:
He was just playing hide and seek. You won!

closed 2020-11-05 by Mimiix