dresden-elektronik/deconz-rest-plugin #3550: Thermostat 902010/32 (mode support)

keesj, 2020-11-01:
## Device

The device itself it already added in  #1003 and is the 
Bitron Thermostat 902010/32. For better support however it would be good to be able to set the heating mode 

## Screenshots

[image]

[image]
[image]

Smanar, 2020-11-01:
Hello again.

So to test it, you need an Unix OS (if you have one ?), and take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only with a modification for step 1

```
git clone --branch bitron https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

keesj, 2020-11-01:
[image]

keesj, 2020-11-01:
Some more notes on things dicovered:
from deCONZ ui it is possible to set the mode (attribute 0x001C) but only to 2 values , either OFF or Heat, setting to an other value like "AUTO" results in a write failed. but overall I can now set the device off in domoticz
[image]

closed 2020-11-08 by manup