dresden-elektronik/deconz-rest-plugin #3555: Why is ID 1 Sensor updated every minute?

TTP1106, 2020-11-01:
## Describe the question or issue you are having

 made a Websock which connects to Deconz WS via Browser.
I have a few switches in my envoirment.

If I do nothing then every minute appears new Event from ID: 1 (daylight sensor). But if I request the Sensors´ ID then there is no change. Why is this announced every minute?

## Screenshots

[image]
[image]
[image]

## Environment

 - Host system: OrangePi Zero / Orange Pi One
 - Running method: Marthoc Docker container
 - Firmware version: 264A0700
 - deCONZ version: 2.05.86 / 15.10.2020
 - Device: ConBee II
 - Do you use an USB extension cable: NO
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs

`   [marthoc/deconz] Starting deCONZ...                                                                                             │
│   [marthoc/deconz] Current deCONZ version: 2.05.86                                                                                │
│   [marthoc/deconz] Web UI port: 8081                                                                                              │
│   [marthoc/deconz] Websockets port: 8443                                                                                          │
│   [marthoc/deconz] VNC Disabled                                                                                                   │
│   QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'                                                      │
│   libpng warning: iCCP: known incorrect sRGB profile                                                                              │
│   This plugin does not support propagateSizeHints()                                                                               │
│   This plugin does not support propagateSizeHints()                                                                               │
│   This plugin does not support propagateSizeHints()                                                                               │
│   [marthoc/deconz] Starting deCONZ...                                                                                             │
│   [marthoc/deconz] Current deCONZ version: 2.05.86                                                                                │
│   [marthoc/deconz] Web UI port: 8081                                                                                              │
│   [marthoc/deconz] Websockets port: 8443                                                                                          │
│   [marthoc/deconz] VNC Disabled                                                                                                   │
│   QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'                                                      │
│   libpng warning: iCCP: known incorrect sRGB profile                                                                              │
│   This plugin does not support propagateSizeHints()                                                                               │
│   This plugin does not support propagateSizeHints()                                                                               │
│   This plugin does not support propagateSizeHints()     `

## Additional context

Why is this close bot so freaky?

Smanar, 2020-11-01:
Hello, you have it only for this sensor ?

If I m right all devices send this notification "attr" every minute, to update the "lastseen"

TTP1106, 2020-11-01:
No I also have temperature Sensors which are only sending notification on change, not time based.
My Aqara Switches and My Osram AC/Plug are also not sending every minute. Its only the Daylight Sensor which is an virtual sensors. So in my opinion this does not make sense...

Mimiix, 2020-11-02:
@TTP1106 

The bot closes if you are not following the template.

The sensors decide when there is a update, not deCONZ.  The virtual sensor does so apparently.  The update is the timestamp.

As far as i know, @ebaauw implemented this announcement. He might be able to provide some information on this.

ebaauw, 2020-11-02:
> If I do nothing then every minute appears new Event from ID: 1 (daylight sensor). 

No.  The log shows events for `/lights/1`, not for `/sensors/1`.  The notification is triggered by `lastseen` being updated.  Set `/config/websocketnotifyall` to `false` to receive only the updated attributes in the event.

TTP1106, 2020-11-02:
That´s strange because I do not have a light connected:
[image]

I´ll test the /config/websocketnotifyall false flagging

TTP1106, 2020-11-02:
Why is Conbee II added as light ressource?

Smanar, 2020-11-02:
Long story but it s normal too, since this version https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_74

>Support RaspBee, ConBee, ConBee II, XBee under lights resources 8c8e4b

TTP1106, 2020-11-07:
Why is it saying that it does not exists?

> No. The log shows events for `/lights/1`, not for `/sensors/1`. The notification is triggered by `lastseen` being updated. Set `/config/websocketnotifyall` to `false` to receive only the updated [...]

`[{"error":{"address":"/config/websocketnotifyall","description":"resource, /config/websocketnotifyall, not available","type":3}}]`

Smanar, 2020-11-07:
Do you have the complete request you have used ?
But for me it s useless (you will have shorter notifications but all will be same, sd write/red, memory usage, process used, ect ...), and your third app can need the websocketnotifyall set to true to work correctly.

TTP1106, 2020-11-09:
I just made a quick GET via Browser on this URL:
http://mydevice/api/D00000C086/config/websocketnotifyall

Smanar, 2020-11-09:
You need to make a PUT with a form, for exemple with curl

`curl -H 'Content-Type: application/json' -X PUT -d '{"websocketnotifyall ": false}' http://IP:PORT/api/API_KEY/config`

TTP1106, 2020-11-09:
Okay, my Mistake.
`curl -X PUT -H "Content-Type: application/json" -d '{"websocketnotifyall":false}' http://localhost:8081/api/D60849C086/config
[{"success":{"/config/websocketnotifyall":false}}]`
This worked to change the Attribute. But it will not change the announcements

[image]

As @Smanar  mentioned, it will not work to reduce it.

Smanar, 2020-11-09:
For me it s normal, you have this notification on every "lastseen" change, so every minut if I remember.

closed 2020-11-15 by TTP1106