dresden-elektronik/deconz-rest-plugin #3559: Tint Müller Licht RGB Colorloop und weitere Effekte

BigHub40, 2020-11-02:
## Feature request type
Erweiterung der Effekte bei Müller Licht Tint

## Description
Hallo,
ich nutze den Conbee 2 Stick in Verbindung mit dem IOBROKER deconz adapter.
In dem Adapter selbst gibt es einen Datenpunkt "Effekt" und einen Datenpunkt "Colorspeed".
In Effekt kann man "Colorloop" oder "None" eintragen. In Colorspeed einen Wert für die Geschwindigkeit des Colorloops.
So weit sehr gut!
Was natürlich toll wäre, wären die weiteren Effekte, die man über die TINT Fernbedienung auslösen kann.
Nachtlicht, Romantisch, etc...
Die statischen Effekte kann man ja eigentlich mehr als Szene ansehen, daher kann man sich diese ja theoretisch auch selber als "Befehl" basteln und anwenden.
Was jedoch nicht so ohne weitere geht sind die dynamischen Effekte, wie eben der Colorloop.
Ich finde zum Beispiel den Feuer Effekt der ein Lagerfeuer oder Kerzenflackern simuliert hilfreich. Ebenso wäre es cool wenn man so etwas wie diesen "Romantik" Effekt umsetzen kann, bei dem innerhalb der Rot und Pink Töne langsam "geloopt" wird.
Ideal wäre natürlich wenn man zwei Datenpunkte hätte in denen man jeweils einen Farbwert einträgt und dann bei Effekt z.B
"2 Color Loop " anklicken könnte, wobei dann zwischen den beiden Farbwerten hin und her geloopt wird. Im Prinzip genau wie beim Colorloop, nur das nicht durch alle Farben sondern zwischen den beiden vordefinierten hin und her geloopt wird.
Lassen sich die Fernebdingungseffekte zusätzlich zum Colorloop integrieren?
Ließe sich so ein "2 Farbwerte Loop" integrieren?
Danke sehr und viele Grüße!

## Considered alternatives
Alternativen würden mir keine Einfallen.

## Additional context
(Beim Adapter Programmierer hatte ich schon angefragt. Dort war die Antwort, dass der Adapter selbst es nicht kann, wenn es nicht bei deConz integriert ist.)

Mimiix, 2020-11-02:
Hi,

As we are fully English here, could you please translate to English? This would help users to understand better :)

Thanks.

BigHub40, 2020-11-02:
Done :-)

Mimiix, 2020-11-02:
Thanks :) I just saw it.

Was it this one?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1209

That was just closed due to inactivity.

BigHub40, 2020-11-02:
No, not #1209 
#1209 was from somebody who wants to integrate the tint remote into the adapter.

My request was this one:
https://github.com/iobroker-community-adapters/ioBroker.deconz/issues/183

Its only about the above named Color Effekt feature.
Thanks

Mimiix, 2020-11-02:
Let me check with @jey-cee

Jey-Cee, 2020-11-02:
I am not sure if this effects are already implemented in the lamp its self like the colorloop, but it sounds so.
If it so deConz has to export it like the colorloop feature.

ebaauw, 2020-11-02:
The REST API plugin already exposes the built-in "effect scenes" of Tint Müller lights through `state.effect`, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1209#issuecomment-648931529.

I don't think the plugin should support effects that need to be programmed using multiple commands (either using rules, or using external logic, like Node Red, HA, ...).  Maybe Phoscon could, but it would seem they have plenty of other priorities.

BigHub40, 2020-11-02:
Hi all, i have testet what ebaauw has written. The effects are working if you set them in the "effect" datapoint of the light in the deconz adapter for iobroker. 
First of all that really great !!! (i have overseen this in the long remote #1209 request.)
What i dont understand is: when you klick on the effect datapoint in the deconz adapter in iobroker, only none(none) and colorloop(colorloop) comes up as a drop down. Why does it not shown the other already implemented effects?

If my request about looping to customer defined colours should not be implemented, i will try to solve this with a blockly script in iobroker.

BigHub40, 2020-11-02:
"sunset", "party", "worklight", "campfire", "romance", or "nightlight" ... are the implemented effects...

scobby, 2021-01-03:
Still in developement?

irekzielinski, 2021-01-03:
> Still in developement?

I think it's all done. I am able to use all functionality of this light via REST api (including effects)

scobby, 2021-01-03:
Can you close it?

Mimiix, 2021-01-03:
@BigHub40 case closed?

BigHub40, 2021-01-05:
yes

 
 

Gesendet: Sonntag, 03. Januar 2021 um 23:20 Uhr

closed 2021-01-05 by Mimiix