dresden-elektronik/deconz-rest-plugin #3561: HomeSeer plugin calling DeConz cresating error message in log

PrimaryInc, 2020-11-02:
## Describe the question or issue you are having
Hi, Im using DeConz via HomeSeer 4 and Jowihue plugin. 
This works great :) During Halloween I'æve setup Scenes and animation via jowihue plugin and most of the time it works great.
Somethimes i can see that error message is loged in HomeSeer. See screenshot.
I guess this is because of to many requests happening.
So question is, is ther any hard limit to amount of requests that can happen?

## Screenshots

Scene createt johihuw plugin, changing color of light bulb.

[image]

Animation - Called to automate change fo color, HomeSeer event cals the nimation to start and end.

[image]

Log error 

[image]

## Environment

 - Host system: PC
 - Running method: Windows Server 2019 Standard runing as a Virtual Machine in Vmware
 - Firmware version: 26390500
 - deCONZ version: 2.05.86
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: RFXtrx433 and Sigma Design z-wave stick

Mimiix, 2020-11-02:
Hi @PrimaryInc :)

We do not maintain homeseer or jowie hue. You should ask in their repository/help channels.

Sorry! :(

Kind regards,
Dennis

closed 2020-11-02 by Mimiix

PrimaryInc, 2020-11-02:
@Mimiix  Seems you missed the point but yeah...

Kane610, 2020-11-02:
There is a queue which is limited to 20 requests, more than that and you will get errors

Mimiix, 2020-11-02:
> @Mimiix Seems you missed the point but yeah...

I did get it, but either way we cannot fix it. The limit is 20 requests, thus they need to adjust this.

PrimaryInc, 2020-11-02:
Hi, Kane610 @Mimiix thanks for reply.

Well it is my setup that create the amount of request so limit is DeConz not HS og Jowihue. 
Que limit shoud have been your answer in the first place @Mimiix and not a "not our problem and close" response.

Supplemental question
Is the 20 requests inside a time frame or continuous requests?
Any way to enable more requests, or is this a hard limit? 
Reson for me asking is to tune my requests to DeConz.
Looking at my automation it should not hit the 20 limit, but mulitiple sceens and animations was in effect and probaply is the reason.

Kane610, 2020-11-02:
AFAIK its hardcoded. Its a queue so it manages the requests in it and if you're too quick at any given point you will get a 901 message and retry sending. You should create an issue with the plugin and ask them to implement a retry mechanism

PrimaryInc, 2020-11-02:
Thanks @Kane610, 

To bad it is hardcoded, but I'm sure it have its reason :)
I'll lett JowiHue deleveoper know about this limit.

Thanks again :)

Mimiix, 2020-11-02:
> Que limit shoud have been your answer in the first place @Mimiix and not a "not our problem and close" response.

Not intended to give you that feeling. Sorry if i did so.

PrimaryInc, 2020-11-02:
@Mimiix no problem m8 :)