dresden-elektronik/deconz-rest-plugin #3562: Updated firmware and now Conbee II is not reachable

Staphylococcus, 2020-11-02:
## Describe the question or issue you are having
I've just updated the firmware via Phoscon and the Conbee II now can't control any devices. In the gateway settings firmware is reported as "Not connected". I've downloaded backups before performing the upgrade, but restoring from backup seems to do nothing. Resetting the gateway results in failure.

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Homebridge Raspbian image
 - Firmware version: Not connected
 - deCONZ version: 2.05.84 / 9/14/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2020-11-02:
Hi!

I'd like to ask you to follow this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Then use version 26650700 .

See if that helps :)

Thanks!

Mimiix, 2020-11-02:
Looks like : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3387

Staphylococcus, 2020-11-02:
```
pi@homebridge:~ $ sudo systemctl stop deconz
Warning: The unit file, source configuration file or drop-ins of deconz.service changed on disk. Run 'systemctl daemon-reload' to reload units.
pi@homebridge:~ $ wget https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26650700.bin.GCF
--2020-11-02 12:48:50--  https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26650700.bin.GCF
Resolving deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)... 84.19.195.242
Connecting to deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)|84.19.195.242|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 164104 (160K) [application/x-graphing-calculator]
Saving to: 'deCONZ_RaspBeeII_0x26650700.bin.GCF'

deCONZ_RaspBeeII_0x26650700.b 100%[=================================================>] 160.26K   401KB/s    in 0.4s    

2020-11-02 12:48:51 (401 KB/s) - 'deCONZ_RaspBeeII_0x26650700.bin.GCF' saved [164104/164104]

pi@homebridge:~ $ sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_RaspBeeII_0x26650700.bin.GCF
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 26660700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164090 bytes: |=
dev: ERROR: INVALID FIRMWARE TYPE
 (10 ms)
Reboot device /dev/ttyACM0 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164090 bytes: |=
dev: ERROR: INVALID FIRMWARE TYPE
 (10 ms)
Reboot device /dev/ttyACM0 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164090 bytes: |=error: timeout flashing firmware after 3000 ms
^C
pi@homebridge:~ $ sudo systemctl stop ModemManager
pi@homebridge:~ $ sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_RaspBeeII_0x26650700.bin.GCF
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
retry, failed
pi@homebridge:~ $ sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_RaspBeeII_0x26650700.bin.GCF
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
retry, failed
pi@homebridge:~ $ 
```
Is it a brick?

Staphylococcus, 2020-11-02:
I tried correcting my confusion in naming `RaspBee` -> `ConBee`, meaning downloading the right firmware and trying to flash with that, but the flasher still exits with `retry, failed`.

Staphylococcus, 2020-11-02:
Managed to breathe life into the darn thing by flashing from Windows, now I'm running the `26650700` firmware and Zigbee connectivity is back online. Is the issue in the new firmware? Should I stick to the current one for the time being and what is the team's thinking on the handling of these scenarios? One learning I personally take out of this experience is that the branding of `ConBee` and `RaspBee` is easy to mix up from a glance; another is that updating the firmware is probably not something that should be done when one has a lot of devices connected -> too much time spent afterwards trying to restore functionality.

Mimiix, 2020-11-02:
I think it is @Staphylococcus . I've already forwarded it.

closed 2020-12-19 by stale[bot]