dresden-elektronik/deconz-rest-plugin #3568: OTA Firmware Update for Legrand devices (#

PGU79, 2020-11-02:
## Describe the question or issue you are having
Issue #883 was closed with a hope for public firmware availability from Legrand.

For testing purpose i am using a single wireless switch from legrand with old firmware (1e). I don't plan to buy legrand gateway just to update one switch. 

Old firmware doesn't seem to allow update of FC01 cluster attributes : attributes are grayed out after clicking on read button in deconz app. 

ls this behavior expected or am I doing something wrong ?

Do you have any news from Legrand concerning firmware availability ? 

## Screenshots

N/A
## Environment

 - Host system: VM on Mac
 - Running method: Ubuntu on Virtual Machine, Homebridge with hue plugin
 - Firmware version: 26660700
 - deCONZ version: 2.05.86
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
N/A

## Additional context

Smanar, 2020-11-02:
Hello, so first some information about legrand firmware.
- ATM deconz have a problem to update Legrand firmware, will be corrected soon > https://github.com/dresden-elektronik/deconz-ota-plugin/pull/19
- From someone from Netatmo, the firmware will be officially public in november, but long time we are waiting them, and we don't know the procedure to obtain them, free access / developper reserved for exemple.

Don't buy the gateway just to update 1 device, ask first for it, I have some of them (old firmwares for the majority, but possible to have more recent one from other users, for exemple from pipiche)

But for you it s useless ^^, IDK wich one option you want to enable on the cluster FC01, but wireless device don't use "permanent LED", to prevent the  battery use. And I m perhaps wrong but the long press (dim feature) is enabled by defaut.

PGU79, 2020-11-02:
Thanks Smanar.

I just want to enable dim feature on my switch. When I try to read FC01 cluster all attributes previously visible are grayed out. Write doesn't work either. And dim feature is not enable by default.

That's why I want to upgrade firmware (version 33 or 34). From my understanding (I read it somewhere) it will enable dim feature.

Smanar, 2020-11-02:
Yep for the wired device, not the wireless, I already have this disscution why someone else, I can search it tommorow.
But there is an easy way to be sure, just look at return value when you use the device

```
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Dimm up"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Dimm up stop"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Dimm down"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Dimm down stop"]
```

Make short/long press and look at the value returned, you will see if the long press is working or not.

PGU79, 2020-11-02:
OK,

long press or hold is working (found some INFO message about that), but It seems dimming is going too fast when binding legrand switch to a group using deconz. I read this somewhere too. Dimming was actually working but due to default offset it was mostly like on/off during long press or hold.

Will try to find out where I read this.

PGU79, 2020-11-02:
I found where I read info about dimming step : comments from jd1900 on issue #883. He had this problem with IKEA Bulbs and my test is with hue bulbs !

Smanar, 2020-11-03:
Ha yes, right.

> As it is not defined, the bulbs go as fast as possible, therefore, going directly to "maximum" or "minimum". So there are three solutions:
> 1.- Change the rate, which doesn't seem to be easy or supported by Legrand (at least in the official app).
> 2.- Change the 0x0014 attribute in the bulb. I'll have to research if that is possible.
> 3.- Use rules, as you say. Which I don't like because if the system is down, I can't switch on/off the lights.

> Maybe I explained wrongly. Using the Legrand remote bound to a Legrand light, works fine. Regulation takes a few seconds to reach the maximum or minimum level.
> Using the Legrand remote bound to a Ikea bulb, it does not work fine, it jumps from the current value to the maximum or minimum values.

So I thing it s from the firwmare, it was made for Legrand, not sure you can use direct connexion for level control with a Legrand remote.

PGU79, 2020-12-05:
OK to close.

closed 2020-12-05 by PGU79

fvodola, 2020-12-10:
> Don't buy the gateway just to update 1 device, ask first for it, I have some of them (old firmwares for the majority, but possible to have more recent one from other users, for exemple probably from pipiche)

@Smanar , may you please share a repository where I can retrieve these firmwares? With the recent deconz version the OTA issue should have been fixed. THanks

Smanar, 2020-12-10:
Netatmo want to make something clean, so they are making a portal to download firmwares. (delayed because of Zigbee 3.0 certifications)
And during this time they have ask us to wait for public share.
They are realy "fair play", so for the moment we respect their conditions, and there is no public repositary. But we can share them, just pm me on discord with the device name and the firmware version you have.

vsaugey, 2021-12-12:
Sorry do you still think the guy are really "fair play" ?
One year after still nothing happened ...
I very disappointed by this brand. 
Regards,

Smanar, 2021-12-13:
They are starting to make their firmware public https://developer.legrand.com/documentation/operating-manual/ (ok only 2 for the moment but it s a good starting point)

Rjevski, 2023-07-22:
@Smanar any chance you can revisit your stance on this and release any firmwares you got? It doesn't look like Legrand is being fair-play anymore? It's been over a year now and not only is that link above dead but any firmware-related links on their developer portal are now behind a login screen and there doesn't seem to be a way to self-service register (there's a contact form, but is unlikely to grant access to end-users).

Smanar, 2023-07-22:
You have this information https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-1585596489

And yes on their site you need a dev account to have last official one, but you can take them on the z2m page too 

https://github.com/Koenkk/zigbee-OTA/tree/master/images/Legrand