dresden-elektronik/deconz-rest-plugin #3570: Conbee II : problem devices connection after latest upgrade 26660700 firmware

baldus88, 2020-11-02:
# Describe the bug
Hi.
I  have the Conbee II plugged to my raspberry (with raspbian os) and home assistant software for home automation.
Before this double upgrade (firmware & deconz) the integration of my zigbee devices is ok.

First I have installed the firmware (26660700) via this guide:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows
After I have installed the latest version of deconz 2.05.86.

The old firmware and not update needed.
After I tried to plug the conbee ii on my laptopt and I am successfully update the conbee to the last firmware.

When try to connect the new device (ex. osram light) the phosocon app found and connect the device. But I am trying to integrate this in home assistant o restart the device the phosocon app have more different  behaviors.

_Firmware Not connected:_

_Or Firmware_: 000000

If i try to reconnect the light the conbee not found nothing.

Probably the deconz software is corrupted? Or other reasons?

I am already removed the software (via "sudo apt remove deconz") and reinstalled but the problem remain the same.

Thanks for the support

## Steps to reproduce the behavior

  1. Go to Phoson App
  2. Click on Gateway 
  3. The version of firmware is different
  4. Try to connect the devices
  5. Nothing found

## Expected behavior
The regular connected of my zigbee devices

## Screenshots
If I will try to update firmware via raspbian in phoscon app view this:
[image]

Firmware not connected:
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian with Home Assistant Core
 - Firmware version: (26660700)
 - deCONZ version: (2.05.86)
 - Device: ConBee II
 - Do you use an USB extension cable: no)
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
```
17:37:18:637 HTTP Server listen on address 0.0.0.0, port: 40851, root: /usr/share/deCONZ/webapp/
17:37:18:659 CTRL. 3.27.217:37:18:964 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:37:18:964 COM: /dev/ttyS0 :  (0x0000/0x0000)
17:37:18:964 dev /dev/ttyS0 (/dev/ttyS0)
17:37:18:964 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
17:37:18:965 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
17:37:19:390 parent process bash
17:37:19:390 gw run mode: normal
17:37:19:390 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
17:37:19:390 sd-card cid: 9c534f4c5836344710ffffffff013aad
17:37:19:393 DB sqlite version 3.27.2
17:37:19:394 DB PRAGMA page_count: 33
17:37:19:394 DB PRAGMA page_size: 4096
17:37:19:394 DB PRAGMA freelist_count: 0
17:37:19:394 DB file size 135168 bytes, free pages 0
[... 357 lines cut ...]
16:38:54:878 COM: /dev/ttyS0 :  (0x0000/0x0000)
16:38:54:878 dev /dev/ttyS0 (/dev/ttyS0)
16:38:54:878 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
16:38:55:317 device state timeout ignored in state 2
16:38:55:817 try to reconnect to network try=4
16:38:57:317 MASTER kill cmd 0x08 (TIMEOUT)
16:38:57:473 MASTER kill cmd 0x08 (ERROR)
16:38:57:473 MASTER kill cmd 0x08 (ERROR)
16:38:57:473 MASTER kill cmd 0x08 (ERROR)
16:38:57:476 saved node state in 0 ms
16:38:57:540 sync() in 64 ms
16:38:57:541 REST API plugin shutting down
16:38:57:541 DB save zll database items 0x000000C1
16:38:57:543 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:21:2e:ff:ff:04:e2:10-01', '1.0', '{"dark":true,"daylight":false,"lastupdated":"2020-11-02T16:37:19.420","status":200,"sunrise":"2020-11-02T06:05:34","sunset":"2020-11-02T16:10:33"}', '{"configured":true,"lat":"45.4707","long":"9.1889","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
16:38:57:560 DB saved in 18 ms
16:38:57:582 sync() in 21 ms
16:38:57:583 ~Resource() /sensors 0x15c8160
16:38:57:583 ~Resource() /groups 0x150b4b0
16:38:57:583 ~Resource() /groups 0x150b538
16:38:57:583 ~Resource() /groups 0x150b5c0
16:38:57:583 ~Resource() /config 0x14d19dc

```

## Additional context

Mimiix, 2020-11-02:
Hi! Can you try going to version 26650700 ?

If that works, you have the same issue as https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3387

baldus88, 2020-11-03:
> Hi! Can you try going to version 26650700 ?
> 
> If that works, you have the same issue as #3387

Thanks for the suggestion. I am downgraded the firmware and now the conbee working properly.
For the future waiting 2/3 months befor to upgrade :)

Mimiix, 2020-11-04:
@baldus88 I suggest re-opening this one.

Mimiix, 2020-11-04:
Can you provide logs?

baldus88, 2020-11-04:
Yes.
When try to launch the command the gui is not open correctly:
[image]

This is the log produce:
```
11:57:25:606 HTTP Server listen on address 0.0.0.0, port: 40851, root: /usr/share/deCONZ/webapp/
11:57:25:706 CTRL. 3.27.211:57:25:995 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
11:57:25:995 COM: /dev/ttyS0 :  (0x0000/0x0000)
11:57:25:995 dev /dev/ttyS0 (/dev/ttyS0)
11:57:25:995 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
11:57:25:996 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
11:57:26:417 parent process bash
11:57:26:417 gw run mode: normal
11:57:26:417 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
11:57:26:418 sd-card cid: 9c534f4c5836344710ffffffff013aad
11:57:26:420 DB sqlite version 3.27.2
11:57:26:421 DB PRAGMA page_count: 35
11:57:26:422 DB PRAGMA page_size: 4096
11:57:26:422 DB PRAGMA freelist_count: 0
11:57:26:422 DB file size 143360 bytes, free pages 0
11:57:26:422 DB PRAGMA user_version: 7
11:57:26:422 DB cleanup
11:57:26:423 DB create temporary views
11:57:26:424 DB view [0] created
11:57:26:424 DB view [1] created
11:57:26:425 DB view [2] created
11:57:26:426 DB view [3] created
11:57:26:426 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
11:57:26:428 sql exec SELECT key FROM config2
11:57:26:429 sql exec SELECT key,value FROM config2
11:57:26:429 Load config UTC: 2020-11-02T19:06:25 from db.
11:57:26:429 Load config announceinterval: 45 from db.
11:57:26:429 Load config announceurl: https://phoscon.de/discover from db.
11:57:26:429 Load config apiversion: 2.05.86 from db.
11:57:26:429 Load config bridgeid: 00212EFFFF04E210 from db.
11:57:26:429 Load config datastoreversion: 93 from db.
11:57:26:429 Load config dhcp: true from db.
11:57:26:429 Load config discovery: true from db.
11:57:26:429 Load config factorynew: false from db.
11:57:26:429 Load config fwneedupdate: false from db.
11:57:26:429 Load config fwupdatestate: idle from db.
11:57:26:429 Load config fwversion: 0x26650700 from db.
11:57:26:429 Load config gateway: 127.0.0.1 from db.
11:57:26:429 Load config group0: 65520 from db.
11:57:26:429 Load config groupdelay: 50 from db.
11:57:26:429 Load config gwpassword: $1$8282jdkm$h6vxvMoj2g6UlHvcUGYsI0 from db.
11:57:26:429 Load config gwusername: delight from db.
11:57:26:429 Load config homebridge: not-managed from db.
11:57:26:429 Load config homebridge-pin:  from db.
11:57:26:429 Load config homebridgepin:  from db.
11:57:26:429 Load config homebridgeupdate: false from db.
11:57:26:429 Load config homebridgeupdateversion:  from db.
11:57:26:429 Load config homebridgeversion:  from db.
11:57:26:429 Load config ipaddress: 192.168.x.x. from db.
11:57:26:429 Load config linkbutton: false from db.
11:57:26:429 Load config localtime: 2020-11-02T20:06:25 from db.
11:57:26:429 Load config mac: 38:60:77:7c:53:18 from db.
11:57:26:429 Load config modelid: deCONZ from db.
11:57:26:429 Load config name: Phoscon-GW from db.
11:57:26:429 Load config netmask: 255.0.0.0 from db.
11:57:26:429 Load config networkopenduration: 180 from db.
11:57:26:429 Load config otauactive: false from db.
11:57:26:430 Load config otaustate: off from db.
11:57:26:430 Load config panid: 0 from db.
11:57:26:430 Load config permitjoin: 0 from db.
11:57:26:430 Load config permitjoinfull: 0 from db.
11:57:26:430 Load config port: 40851 from db.
11:57:26:430 Load config portalservices: false from db.
11:57:26:430 Load config proxyaddress: none from db.
[... 506 lines cut ...]
10:58:22:879 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:21:2e:ff:ff:04:e2:10-01', '1.0', '{"dark":false,"daylight":true,"lastupdated":"2020-11-04T10:57:26.475","status":160,"sunrise":"2020-11-04T06:08:24","sunset":"2020-11-04T16:07:47"}', '{"configured":true,"lat":"45.4707","long":"9.1889","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
10:58:22:894 DB saved in 15 ms
10:58:22:914 sync() in 19 ms
10:58:22:915 ~Resource() /sensors 0x1e948f8
10:58:22:915 ~Resource() /sensors 0x1e949e0
10:58:22:915 ~Resource() /sensors 0x1e94ac8
10:58:22:915 ~Resource() /sensors 0x1e94bb0
10:58:22:915 ~Resource() /sensors 0x1e94c98
10:58:22:915 ~Resource() /sensors 0x1e94d80
10:58:22:915 ~Resource() /sensors 0x1e94e68
10:58:22:915 ~Resource() /sensors 0x1e94f50
10:58:22:915 ~Resource() /sensors 0x1e95038
10:58:22:915 ~Resource() /sensors 0x1e95120
10:58:22:915 ~Resource() /sensors 0x1e95208
10:58:22:915 ~Resource() /sensors 0x1e952f0
10:58:22:915 ~Resource() /sensors 0x1e953d8
10:58:22:915 ~Resource() /sensors 0x1e954c0
10:58:22:915 ~Resource() /sensors 0x1e955a8
10:58:22:915 ~Resource() /sensors 0x1e95690
10:58:22:915 ~Resource() /sensors 0x1e95778
10:58:22:915 ~Resource() /groups 0x1cee748
10:58:22:915 ~Resource() /groups 0x1cee7d0
10:58:22:915 ~Resource() /groups 0x1cee858
10:58:22:915 ~Resource() /config 0x1d8e224
```

Mimiix, 2020-11-04:
Can you try with a extension cable? It can't join the network.

SwoopX, 2020-11-04:
@Mimiix Raspbee, so that will be difficult... Except for there is any external device attached.

Mimiix, 2020-11-04:
> @Mimiix Raspbee, so that will be difficult... Except for there is any external device attached.

That must be the case then.

baldus88, 2020-11-04:
> Can you try with a extension cable? It can't join the network.

Before the upgrade the conbee work good without a problem. Only difference are upgrade deconz and upgrade firmware.

Mimiix, 2020-11-04:
@SwoopX Can it be that network key thing you had earlier?

vsaarinen, 2020-11-04:
I am also experiencing similar issues using a ConBee2 with a Raspberry Pi 3 B+ and an extension cable:

```
[...snip...]
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:38:702 failed to reconnect to network try=3
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:38:702 wait reconnect 5 seconds
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:39:702 wait reconnect 4 seconds
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:40:702 wait reconnect 3 seconds
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:41:702 wait reconnect 2 seconds
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:42:702 wait reconnect 1 seconds
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:42:792 dev /dev/ttyAMA0
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:42:792 COM: --dev: /dev/ttyACM1 (ConBee II)
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:42:834 device disconnected reason: 4, index: 0
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:43:709 failed to reconnect to network try=4
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:43:709 wait reconnect 15 seconds
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:43:794 dev /dev/ttyAMA0
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:43:794 COM: --dev: /dev/ttyACM1 (ConBee II)
Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:44:702 wait reconnect 14 seconds
[...snip...]
```

The weird thing is that the Zigbee connection to my devices still seems to work, but every now and then my lights disappear and the deCONZ web UI shows the firmware as "Not connected", after which I need to disconnect the ConBee 2 for 10 seconds, reconnect it and restart Home Assistant in order to make everything visible again. I also had an issue where my Trådfri remotes stopped working for a day or two until they started working again by themselves, but that might be an unrelated problem.

I was initially using the 26660700 firmware and downgraded to 26650700, but the same issue persists.

Mimiix, 2020-11-04:
@vsaarinen Can you open a seperate issue with your own details?  Also: In the addon make sure to use the full device serial and not the dev/ttyAMA0.

vsaarinen, 2020-11-04:
> @vsaarinen Can you open a seperate issue with your own details? Also: In the addon make sure to use the full device serial and not the dev/ttyAMA0.

In my deCONZ Home Assistant add-on, my configuration is:
```
device: >-
  /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2226650-if00
```

But sure, I'll open a new issue when I have time to dig into this further.

baldus88, 2020-11-04:
> I am also experiencing similar issues using a ConBee2 with a Raspberry Pi 3 B+ and an extension cable:
>
> ```
> [...snip...]
> Nov  4 15:44:45 raspberrypi 3e8797087857[744]: 15:44:38:702 failed to [...]

Hi @vsaarinen. I have the same situation. I have tested the same things in addition have a problem with light from osram.

vsaarinen, 2020-11-04:
Sorry to continue using this issue, but just in case it's relevant: could these error logs be due to `deCONZ` being launched twice for some reason?

```shell
pi@raspberrypi:~ $ ps ax | grep deCONZ
  325 ?        Ss    14:00 /bin/bash /usr/bin/deCONZ-WIFI2.sh
  329 ?        Ss     5:08 /bin/bash /usr/bin/deCONZ-update2.sh
 7423 ?        Ssl   78:59 /usr/bin/deCONZ -platform minimal --http-port=80
22737 ?        Ssl   47:28 deCONZ -platform xcb --auto-connect=1 --dbg-info=1 --dbg-aps=0 --dbg-otau=0 --dbg-zcl=0 --dbg-zdp=0 --upnp=0 --http-port=40850 --ws-port=8081 --dev=/dev/ttyACM1
22806 ?        S      0:00 /bin/sh /usr/bin/deCONZ-otau-dl.sh
23420 ?        Sl     0:24 /usr/bin/Xtigervnc :0 -desktop Home Assistant - deCONZ -auth /data/.Xauthority -geometry 1920x1080 -depth 16 -rfbwait 30000 -rfbport 5900 -pn -localhost -SecurityTypes None
23422 ?        S      0:00 /usr/bin/perl /usr/bin/tigervncserver -name Home Assistant - deCONZ -geometry 1920x1080 -depth 16 -localhost yes -SecurityTypes None --I-KNOW-THIS-IS-INSECURE :0
30454 pts/0    S+     0:00 grep --color=auto deCONZ
```

SwoopX, 2020-11-04:
What is it not actually? Above in the screenshot it shows Raspbee, the log files say Conbee II

baldus88, 2020-11-04:
No the conbee shown different firmware to do update:

[image]

Why 26390500 instead 26660700?

SwoopX, 2020-11-04:
Erm, good question. 26390500 is for Raspbee I, 26660700 for Conbee II (and Raspbee II as I recall).

baldus88, 2020-11-04:
> Erm, good question. 26390500 is for Raspbee I, 26660700 for Conbee II (and Raspbee II as I recall).

Yep, is not normal the conbee II signal the upgrade for firmware raspbee :(

closed 2020-12-19 by stale[bot]

Spirituss, 2022-01-25:
@Mimiix I propose to reopen the issue. I upgraded Conbee II to the newest firmware and it doesn't work after that. It appears in the system as USB device for very short time (1-2 seconds) and than disappear again. In Phoscon I see product/firmware info for this short time and than "Not connected". I tried to connect it in Windows/Debian - the same thing. It looks like infinite boot loop. Is any chance to re-flash it manually including option with solder?

foscoj, 2023-01-26:
same behaviour after latest update over here. short appearance and again, and again after latest update

foscoj, 2023-01-26:
Just did that within windows a few minutes ago and it's alive again, thanks!