dresden-elektronik/deconz-rest-plugin #3571: Sending a "effect":"colorloop" on group level does not work

wvuyk, 2020-11-02:
## Describe the bug
Never tested this yet before, but someone wants to send the `colorloop `command to a group. Including the `colorloopspeed` command. I checked today and the response on the `"effect":"colorloop"` is normal: `{"success":{"/groups/19/action/effect":"colorloop"}}`
But the lights do not respond. I also checked on a Philips Hue Bridge and the lights are responding correct. 

## Steps to reproduce the behavior

## Expected behavior

I would expect the colorloop to start on all applicable lights in the group. Also when `"colorloopspeed""=nnn` is send The speed should be set accordingly? 
## Screenshots

## Environment

 - Host system: (Raspberry Pi)
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: 2.05.86
 - Device: RaspBee I 

## deCONZ Logs

## Additional context

ebaauw, 2020-11-02:
It works for me on .84 (production network) and .86 (test with only a single Hue bulb).

What group are you using?  Can you confirm in the GUI that the lights subscribed to that group.  Make sure that _Saturation_ is 254 before starting the colour loop; it only loops through _Enhanced Hue_, so with a low _Saturation_, you won't see the colour change.

wvuyk, 2020-11-03:
Erik,

The lights are subscribed to the group indeed and working with all other commands. It seems I missed an issue that was reported already in #3146. I had included the "on" command as well. It seems deCONZ does not work well with this combination...

wvuyk, 2020-11-03:
Closing this one as it is double with #3146

closed 2020-11-03 by wvuyk