dresden-elektronik/deconz-rest-plugin #3573: deCONZ losing most lights and switches

erikproper, 2020-11-02:
## Describe the question or issue you are having

Tried to add two Busch & Jaeger switches (the two-gang ones with a built in relay).
This seems to have worked. However, now all other lights and switches (Phillips, Innr and OSRAM) have
become unreachable. When looking in the GUI, I indeed don't see any connections to these devices.

Any suggestions on how to remedy this?

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: 2.05.86
 - Device: ConBee I

erikproper, 2020-11-04:
Meanwhile, the wierd thing is that some of the sensors have kept on working. So, even though basically all lights dropped off the grid, the sensors kept connected.

I'm suspecting that the two Busch & Jaeger switches are causing the mayhem. May need to move them to an old HUE bridge, and let hem have their own Zigbee network.

Mimiix, 2020-11-04:
What happens if you remove them ?

erikproper, 2020-11-04:
Already tried that. Didn't make the other lights come back. 

Meanwhile:
- Re-added (touchlink; reset; add again) some of the lights & switches
- Then re-added the B&J's ... which resulted in some (not all) of the re-added  lights/switches to disappear again.

Mimiix, 2020-11-04:
Wait. I just noticed your on the latest firmware.
 
Did you update that too?

erikproper, 2020-11-04:
At least a few weeks ago. So, well before the issues appeared.

Mimiix, 2020-11-04:
Fair enough. Can you go to the latest stable version (84?)

erikproper, 2020-11-04:
Back to 84 now. Lights are not re-appearing yet ...

closed 2020-12-19 by stale[bot]