dresden-elektronik/deconz-rest-plugin #3575: Lights get uncontrollable, have to press 0 in deCONZ

yabbah, 2020-11-03:
## Describe the bug
I have to repeatedly press 0 on certain nodes in deconz-vnc for them to be controllable. In the api it always "reachable": true both before (uncontrollable) and after.

## Steps to reproduce the behavior
Cant be reproduced as its intermitent.

## Expected behavior
That the light should be controllable all the time?
## Screenshots
No screenshoots needed.

## Environment
deCONZ: 2.05.84
Conbee II: 0x26660700

Host system: ubuntu 18.0.4
Running method: HA Addon
Firmware version: 26660700
deCONZ version: 2.05.84
Device: ConBee II
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
The logs say it send to turn the light on. I cant find that specific line.

## Additional context
No additional context.

Mimiix, 2020-11-03:
Any context on what lights in particular?

yabbah, 2020-11-03:
IKEA, Philips hue. Totally random which and what become uncontrollable.

Mimiix, 2020-11-03:
Can you provide all logs? There must be some Zigbee error codes :)

yabbah, 2020-11-03:
No there is not. And I dont have any logs atm. I fix some logs. But all logs say its okey and it seems like it has turned the light on successfully too. But in IRL its off, until I press 0 on the node.

Mimiix, 2020-11-03:
There should be logs that it isn't replying with error codes from zigbee.

kmplngj, 2020-11-03:
I often have the same situation. Random times, random lights. Really often when I turn off groups of lights with Home Assistant or HomeKit/HomeBridge same lights do not react. 

I could also try to sent logs. 

Note: the turn off all rights button in Phoscon mostly works correct. I don't know if his uses another command to turn the lights off.

yabbah, 2020-11-06:
The light is OFF in HA, but ON irl. I pressed on then off, and get this in the log:

`2020-11-06 12:50:44 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True, 'transitiontime': 0}" to "172.30.33.5 /lights/17/state"
2020-11-06 12:50:44 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/17/state/on': True}}]
2020-11-06 12:50:44 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"17","r":"lights","state":{"alert":null,"bri":128,"on":true,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:95:0a:0a-01"}
2020-11-06 12:50:44 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"18","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
2020-11-06 12:50:47 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.30.33.5 /lights/17/state"
2020-11-06 12:50:47 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/17/state/on': False}}]
2020-11-06 12:50:47 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"17","r":"lights","state":{"alert":null,"bri":128,"on":false,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:95:0a:0a-01"}
2020-11-06 12:50:47 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"18","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
`

Still on IRL tho.

yabbah, 2020-11-06:
Here is the log after I pressed '0' in vnc, and the light is turning on / off IRL

`2020-11-06 13:04:09 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/17/state/on': True}}]
2020-11-06 13:04:09 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"17","r":"lights","state":{"alert":null,"bri":128,"on":true,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:95:0a:0a-01"}
2020-11-06 13:04:09 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"18","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
2020-11-06 13:04:09 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"17","lastannounced":"2020-11-03T16:41:59Z","lastseen":"2020-11-06T12:04Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E14 W op/ch 400lm","name":"melanies_fonsterlampa","swversion":"1.2.214","type":"Dimmable light","uniqueid":"00:0b:57:ff:fe:95:0a:0a-01"},"e":"changed","id":"17","r":"lights","t":"event","uniqueid":"00:0b:57:ff:fe:95:0a:0a-01"}
2020-11-06 13:04:14 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.30.33.5 /lights/17/state"
2020-11-06 13:04:14 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/17/state/on': False}}]
2020-11-06 13:04:14 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"17","r":"lights","state":{"alert":null,"bri":128,"on":false,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:95:0a:0a-01"}
2020-11-06 13:04:14 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"18","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
`

Mimiix, 2020-11-06:
That's the API logs. I'm looking for the deCONZ logs.

yabbah, 2020-11-06:
Where in the container can I found them? On the addon-page I cant scroll back so long.

yabbah, 2020-11-06:
`12:50:47:964 delay sending request 91 dt 3 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:052 delay sending request 91 dt 3 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:152 delay sending request 91 dt 3 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:252 delay sending request 91 dt 3 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:352 delay sending request 91 dt 3 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:452 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:552 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:652 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:753 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:852 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:48:952 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:49:052 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:49:152 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:49:252 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:49:352 delay sending request 91 dt 4 ms to 0x000B57FFFE950A0A, ep: 0x01 cluster: 0x0006 onAir: 1
12:50:57:449 0x000B57FFFE950A0A error APSDE-DATA.confirm: 0xA7 on task
12:51:02:707 0x000B57FFFE950A0A error APSDE-DATA.confirm: 0xA7 on task
`

Mimiix, 2020-11-06:
So there is some issue on that node:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log#0xa7-no-application-ack-received

yabbah, 2020-11-06:
I see it as an issue with deconz. As this goes from random lights all the time, no matter if its IKEA or Philips or any other brand. And more people have the same issue. Just press 0 on the node in vnc and it works again. If there a way to schedule a 0-press on all nodes each hour? A ugly workaround, but should work.

Lindansaren, 2020-11-13:
Seeing this issue myself as well. Random lights is not responding but pressing 0 i vnc makes it work again. Really annoying.

Mimiix, 2020-11-13:
@Lindansaren Please open a seperate issue and provide some logs :)

macmanroy, 2020-11-15:
I seem to be having the same problem. Lights still responds to group commands though.

`19:43:44:183 delay sending request 157 dt 3 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:286 delay sending request 157 dt 3 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:383 delay sending request 157 dt 3 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:484 delay sending request 157 dt 3 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:583 delay sending request 157 dt 3 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:683 delay sending request 157 dt 3 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:784 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:883 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:44:990 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:083 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:158 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:183 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:283 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:383 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:483 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:583 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:45:683 delay sending request 157 dt 4 ms to 0x001788010456F679, ep: 0x0B cluster: 0x0300 onAir: 1
19:43:48:567 0x001788010456F679 error APSDE-DATA.confirm: 0xA7 on task
19:43:48:568 max transmit errors for node 0x001788010456F679, last seen by neighbors 88 s`

manup, 2020-11-17:
Can you please provide a screenshot from deCONZ UI (visible via VNC) with the LQI button enabled, to see how the connection quality is?

yabbah, 2020-11-17:
[image]

manup, 2020-11-20:
Quite a large network, some links are weak but overall it looks fine.
I can only guess that when a light is not reachable via unicast that the route is broken (hard to tell if in the coordinator routing table or a hop in between).

As a test can you please create a source route to a light which is not reachable due the error.
**Note:** For the test please don't press 0.

To create a source route:

1. Select the blue coordinator node;
2. Press and **hold** Ctrl;
3. Select a likely good next hop node (it also gets the selection blue frame);
4. Select the destination node (the one which is not reachable);
5. Release Ctrl key;
6. With the 3 nodes selected, open the context menu of the not-reachable node and select **Add source route**.

This will create a blue-red'ish line representing the source route.

[image]

With the source route intact please try to control the light again.
This should overrule any dynamic routing issues for the destination node.

**Note:** Up to 8 hops can be part of a source route, and it can be deleted also via the context menu (important click in an empty area first to deselect all nodes).

yabbah, 2020-11-25:
Worked great, the problem is that I have to do this for each of my "problem-lights", as they change all the time. Gives alot of static routes. :/

Lindansaren, 2020-11-25:
I found a setting in deconz vnc called destination, that was set to unicast since I've changed it to "broadcast routers" i have not had a single issue. FYI i was facing the same issues you are having. Is it all placebo or did it actually make any difference?

manup, 2020-11-25:
Note the destination settings are only used for commands send via the Cluster Info Panel it has no effect on what the REST-API sends. Also Broadcast as the name suggests send a command to all devices which is mostly not what you want.

closed 2020-12-25 by stale[bot]