dresden-elektronik/deconz-rest-plugin #3576: Constant LOW Battery Warning on Sercomm SZ-DWS04 Micro Door Window Sensor

perfecblue, 2020-11-03:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi 4 with Homebridge)
 - Running method: (Raspbian)
 - Firmware version: (26660700)
 - deCONZ version: (2.05.84)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (YES) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs

## Additional context

SwoopX, 2020-11-03:
Hm, so then the device does indeed to seem to provide any battery readings whatsoever...

We could try to swtich that over to battery alarm mask (so the `lowbattery` entry for the sensor) in the hope that his is taken into account.

perfecblue, 2020-11-03:
Thank you for your fast prompt response!  Hope something can be done about it in future updates, thank you guys again!

SwoopX, 2020-11-04:
Just seen that the reporting configuration for battery voltage was not taken into account. Updated the code in this regard.

closed 2020-11-08 by manup

perfecblue, 2020-11-13:
I can now confirm that after the above fix that was merged into #3502, the Sercomm SZ-DWS04 Micro Door Window Sensor is now reporting the battery status correctly!  Like similar Zigbee devices, it’ll take an hour or so for the battery level to be correctly registered so please be patient.  Thank you again for your work on this issue!