dresden-elektronik/deconz-rest-plugin #3577: Livolo switches

sygys, 2020-11-03:
I would like to make a request for support for livolo switches with zigbee support. I fell in love with these wall switches the first time I saw them. They use their own hub or with an app over WiFi. I would really like them to join my  home assistant setup. Is there a way that their products get support by the conbee2 USB stick? That would make me very happy  

Link to the products:

https://www.livolo.nl/?gclid=Cj0KCQjwit_8BRCoARIsAIx3Rj6W3EHm90poOdFegLAMni0tmtagcReepLDjuvsbytxHOLQPVsLNrocaAhMyEALw_wcB

closed 2020-11-03 by close-issue-app[bot]

Mimiix, 2020-11-03:
@YKO-de User needs to use the template.

YKO-de, 2020-11-03:
Yeah. I didn't think of that. ^^

But what do users do who do not know what is meant by this? Or where to find the information they need.

Mimiix, 2020-11-03:
When opening a new issue, it asks for whatever they want to do. 

After that, the user is presented with a template :)

sygys, 2020-11-03:
Well. I don't know anything of templates and how to do a proper request... I'm new to all this. Just bought a livolo switch with zigbee and want to integrate it into HA. I have a conbee2 USB stick but this wasn't picking up the signal from the switch so I asked it here

Mimiix, 2020-11-03:
As i just said:

When clicking "new issue" , you are presented with 4 options. You need the device request one:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Device+Request&template=device-request.md&title=Device+name

In there, stuff is already filled in. That is the template. The template asks you details on how to proceed and where to get further info. However, to request support you do need the device itself.  If you have the wifi version: That won't work with deCONZ. deCONZ uses Zigbee.

sygys, 2020-11-03:
Never mind. Just remove it. I'm done with the switches and done with conbee2. Good luck.

Mimiix, 2020-11-03:
@sygys I won't remove anything. I am not sure why you are replying like this, but you've got to understand that we can't add stuff out of thin air.

sygys, 2020-11-03:
I understand. But I can't provide more info so this is going nowhere.

Mimiix, 2020-11-03:
Ah okay. Well, if you change your mind and have a device you know where to find us 🙂