dresden-elektronik/deconz-rest-plugin #3579: IKEA round remote extra ghost event for long press

PremiumUsername, 2020-11-03:
## Describe the bug
Using the IKEA round remote (5 buttons), the middle button (the power button) produces incorrect output when holding it in. It creates an extra event based on what button was used previously and places it between the two normal events. It's a very reproducible bug for me. 

## Steps to reproduce the behavior
  1. press the right button once
  2. Hold the middle button

## Expected behavior
1. normal output for the first button press produces: 5002
2.  holding produces the following three events:
1002 ---> 5003 --->1001

## Screenshots

## Environment
 - Host system: Raspberry Pi 3
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26390500
 - deCONZ version: 2.05.84
 - Device: ConBee I 
 - Do you use an USB extension cable:no
## deCONZ Logs
```
Event 2 fired 2:03 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "tradfri_remote_control",
        "unique_id": "90:fd:9f:ff:fe:d8:c5:a4",
        "event": 1001
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-03T13:03:34.306832+00:00",
    "context": {
        "id": "faceef331dd411eb92a4670c98429bd6",
        "parent_id": null,
        "user_id": null
    }
}
Event 1 fired 2:03 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "tradfri_remote_control",
        "unique_id": "90:fd:9f:ff:fe:d8:c5:a4",
        "event": 5003
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-03T13:03:33.266915+00:00",
    "context": {
        "id": "fa3041e11dd411eb8c8705ed2beea4e2",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 2:03 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "tradfri_remote_control",
        "unique_id": "90:fd:9f:ff:fe:d8:c5:a4",
        "event": 1002
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-03T13:03:30.256866+00:00",
    "context": {
        "id": "f864f6b01dd411eb9eddf1bedbaea09c",
        "parent_id": null,
        "user_id": null
    }
}
```
## Additional context
I think I have the first generation of IKEA round remotes if there are several

MattWestb, 2020-11-03:
Ikea have 2 hardware for the remote but they is running the same firmware with different data setting. Its looks the function its the same and only the device type is changes in the setting.

The long press the toggle button is generating one soft reset / sync command to the bunded group for syncing all the devices powerstate, light level and color.
The "decoding" the light command and doing HA events of them also some that is not existing is not very clever done but its done.
Its like trying imprinting something else is like try learning one kow to flye (5001).

PremiumUsername, 2020-11-05:
I'm not sure I understand. Are you saying this is a hardware limitation? I find it hard to believe that one button would create an event that belongs to another button without that being a bug of some sort. How does the newer HW version react to a long press of the middle button?

MattWestb, 2020-11-05:
No its not one hardware limitations its depends of the device type in the zigbee world.
Zigbee HA switches is sending codes as events as you have described above.
Zigbee Lights is sending command for string lights,

The IKEA 5 butt is one "Non-Colour Controller (Device ID is 0x0820.)" and is sending  (Ala Zigbee LL and Zigbee 3 standard) commands logged thru wireshark:

Up single: (Level control cluster) Step up with off (0x06), Step mode: Up (0x00), Step size: 43, Transition time 0.5 Seconds

Up dubble:  (Level control cluster) single, Step up with off (0x06), Step mode: Up (0x00), Step size: 43, Transition time 0.5 Seconds *2

Up long, (Level control cluster) Move with OnOff (0x05), Move mode: Up (0x00), rate 84

Up relese, (Level control cluster) Stop (0x03)

Down single, (Level control cluster) Step (0x02); Modde Down (0x01), Step size 43, Transition time 0.5 seconds

Down dubble, (Level control cluster) Step (0x02); Modde Down (0x01), Step size 43, Transition time 0.5 seconds *2

Down long, (Level control cluster) Move (0x01), Move mode: Up (0x01), rate 84

Down relese, (Level control cluster) Stop (0x03)

Right short, (Scene cluster) (0x07) 00010d00 (#)

Right dubble, (Scene cluster) (0x07) 00010d00  (#) * 2

Right long, (Scene cluster) (0x07) 00010d00 (#) repeated

Left short, (Scene cluster) (0x07) 01010d00 (#)

Left dubble, (Scene cluster) (0x07) 01010d00 (#) * 2

Left long, (Scene cluster) (0x07) 01010d00 (#) repeated

Down single, (Level control cluster) Step (0x02); Modde Down (0x01), Step size 43, Transition time 0.5 seconds

Down dubble, (Level control cluster) Step (0x02); Modde Down (0x01), Step size 43, Transition time 0.5 seconds

Down long, (Level control cluster) Move (0x01), Move mode: Up (0x01), rate 84

Down relese, (Level control cluster) Stop (0x03)

Middle short, (On/Off Cluster) Toggle (0x02)

Middle long, (On/Off Cluster) Toggle (0X02) + (scene cluster) (0x09) 0000 (#) + (Level control cluster) Move to level with OnOff (0x04), Level 256, Transition time 0.0 seconds, (scene cluster) (0x07) 02000000 (#)
This sequins is resetting all the bulbs in the group to default brightness and colour so they is synced.

Middle relese, nothing

(#) = Custom commands = Extended Zigbee command not standard

deCONZ is doing one "translating" of the LL commands to HA events and it cant do things that the device is not sending and the device is sending what it suod for string lights.

The devs can changing how they is translating the LL commands to HA events if they finding one better imitation of them but its still fundamental wrong using on LL switch as HA switch and opposite.

closed 2020-12-19 by stale[bot]