dresden-elektronik/deconz-rest-plugin #3586: Loosing contact with IKEA Trådfri controller after upgrade to deCONZ 6.4.1 (2.05.84)

baunan, 2020-11-03:
Upgrades my home assistant to 0.116.4 this weekend and at the same time updated deCONZ to the latest available version from 2.05.81 (I think).
I have two IKEA E1702 on/off controller units (https://www.ikea.com/no/no/p/tradfri-adapter-90356166/), where one is used as a repeater and the second used as a on/off switch. This unit is also farthest away in my network of 21 sensors, At the latest LQI check when they where available none of the sensors did have poor signals.

The IKEA unit farthest away was the first one I noticed to dropped in and out of the network and now the second unit used as a repeater. 

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi 3B
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26580700)
 - deCONZ version: (2.05.84)
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs (from latest startup)
I see the log Classifies 0x680ae2fffef06eca which is the IKEA unit I use as repeater as zombie

17:40:53:460 discovery for zombie 0x680ae2fffef06eca dropped, last try was 354 s ago
17:40:54:625 Websocket disconnected 127.0.0.1:37728, state: 0, close-code: 1000, reason: 
17:40:55:711 New websocket 127.0.0.1:37760 (state: 3) 
17:41:01:137 send permit join, duration: 59
17:41:13:140 discovery for zombie 0x680ae2fffef06eca dropped, last try was 374 s ago
17:41:32:820 discovery for zombie 0x680ae2fffef06eca dropped, last try was 394 s ago
17:41:37:155 Current channel 25
17:41:37:166 Device TTL 2253 s flags: 0x7
17:41:49:140 discovery for zombie 0x680ae2fffef06eca dropped, last try was 410 s ago
17:42:00:135 send permit join, duration: 59
17:42:37:154 Current channel 25
17:42:37:164 Device TTL 2193 s flags: 0x7
17:42:49:203 dev /dev/ttyAMA0
17:42:49:203 GW firmware version: 0x26580700
17:42:49:204 GW firmware version shall be updated to: 0x26660700
17:42:59:136 send permit join, duration: 59
17:43:37:154 Current channel 25
17:43:37:164 Device TTL 2133 s flags: 0x7
17:43:58:138 send permit join, duration: 0
17:43:58:838 Search sensors done
17:44:08:576 Websocket disconnected 127.0.0.1:37760, state: 0, close-code: 1000, reason: 
17:44:11:268 New websocket 127.0.0.1:38436 (state: 3) 
17:44:16:204 discovery for zombie 0x680ae2fffef06eca dropped, last try was 16 s ago
17:44:29:484 Websocket disconnected 127.0.0.1:38436, state: 0, close-code: 1000, reason: 
17:44:30:724 New websocket 127.0.0.1:38470 (state: 3) 
17:44:31:442 Websocket disconnected 127.0.0.1:38470, state: 0, close-code: 1000, reason: 
17:44:32:584 New websocket 127.0.0.1:38490 (state: 3) 
17:44:33:012 discovery for zombie 0x680ae2fffef06eca dropped, last try was 33 s ago
17:44:37:155 Current channel 25
17:44:37:165 Device TTL 2073 s flags: 0x7
17:44:49:332 discovery for zombie 0x680ae2fffef06eca dropped, last try was 49 s ago
17:44:59:199 dev /dev/ttyAMA0
17:44:59:200 GW firmware version: 0x26580700
17:44:59:200 GW firmware version shall be updated to: 0x26660700
17:45:05:652 discovery for zombie 0x680ae2fffef06eca dropped, last try was 65 s ago
17:45:21:972 discovery for zombie 0x680ae2fffef06eca dropped, last try was 82 s ago
17:45:37:154 Current channel 25
17:45:37:164 Device TTL 2013 s flags: 0x7
17:45:38:294 discovery for zombie 0x680ae2fffef06eca dropped, last try was 98 s ago
17:45:51:252 discovery for zombie 0x680ae2fffef06eca dropped, last try was 111 s ago
17:46:04:212 discovery for zombie 0x680ae2fffef06eca dropped, last try was 124 s ago
17:46:17:171 discovery for zombie 0x680ae2fffef06eca dropped, last try was 137 s ago
17:46:30:132 discovery for zombie 0x680ae2fffef06eca dropped, last try was 150 s ago
17:46:37:154 Current channel 25
17:46:37:163 Device TTL 1953 s flags: 0x7
17:46:43:091 discovery for zombie 0x680ae2fffef06eca dropped, last try was 163 s ago
17:46:55:589 [INFO] - No button map for: lumi.sensor_magnet.aq2 endpoint: 0x01 cluster: BASIC (0x0000) command: 0x0A payload[0]: 001
17:46:55:589 ZCL attribute report 0x00158D00041CE6F7 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
[... 90 lines cut ...]
18:03:37:165 Device TTL 932 s flags: 0x7
18:03:48:532 discovery for zombie 0x680ae2fffef06eca dropped, last try was 139 s ago
18:03:55:426 [INFO] - No button map for: lumi.sensor_magnet.aq2 endpoint: 0x01 cluster: BASIC (0x0000) command: 0x0A payload[0]: 001
18:03:55:426 ZCL attribute report 0x00158D0004451EB8 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
18:03:55:426 0x00158D0004451EB8 extract Xiaomi special attribute 0xFF01
18:03:55:426 	01 battery 2995 (0x0BB3)
18:03:55:426 	03 temperature 27 °C
18:03:55:426 	04 unknown 5032 (0x13A8)
18:03:55:426 	05 RSSI dB (?) 10 (0x000A)
18:03:55:426 	06 LQI (?) 4294967296 (0x0100000000)
18:03:55:426 	0a unknown 0 (0x0000)
18:03:55:426 	64 on/off 0
18:04:04:852 discovery for zombie 0x680ae2fffef06eca dropped, last try was 155 s ago
18:04:24:532 discovery for zombie 0x680ae2fffef06eca dropped, last try was 175 s ago
18:04:29:205 dev /dev/ttyAMA0
18:04:29:205 GW firmware version: 0x26580700
18:04:29:205 GW firmware version shall be updated to: 0x26660700
18:04:37:155 Current channel 25
18:04:37:165 Device TTL 872 s flags: 0x7
18:04:44:211 discovery for zombie 0x680ae2fffef06eca dropped, last try was 195 s ago
18:04:48:136 remove outdated neighbor 0x4067

21 Sensors in the system mostly Aqara multi sensors, 2 IKEA unit and one Develco Norwegian HAN. All worked OK before the updated.

Uncertain if I am to upgrade to latest firmware or use a backup to downgrade. Please advice

Mimiix, 2020-11-04:
Hi!

Your mesh is kind of "unhealthy". Zigbee is ment to be meshing thus a healthy mix of routers and sensors. With only 2 Routers and 21 sensors it isn't at this moment. To add to that: Xiaomi's tend to drop off when the host goes down. Most of the time they come back, sometimes they do not.

You might click the devices reset buttons and see what happens, sometimes they come back.

My advice: Add more routers so the mesh gets more healthy.

baunan, 2020-11-04:
Thanks for the advice. I am not sure what the recommended combination between routers and sensors is. My sensors are not located very far from the Conbee II unit, so they are all within a 10meter radius. For the reason you mentioned I added an IKEA powered unit to act as a repeater so the the unit farthest away could have an alternative connection route and then lately one more to use as a switch for a fan. The sensors I use are a mixture of door/window sensors and multi sensors from Aqara 7 of each. As mentioned in the main post, after the upgrade the IKEA units started to drop in and out of the network some hours on and some off. First the one farthest away and then the repeater. After the a reboot I have not been able to get contact with them  again. As this installation in on my cabin I have not been able to try a reintegration yet. But are mostly concern if there is a software issue as this problem started after an upgrade as I didn't have any problems before.

[image]

DibblesNL, 2020-11-07:
I am having the same issues. Upgraded to the latest Deconz and spent a day trying to get my sensors all to reconnect before giving up. Wish i had a more up to date backup of the previous version to restore.

MattWestb, 2020-11-07:
@baunan you sensors with green line have OK signal and they is connection is no problem but its better if they having some router near so not need emptying the battery so fast and getting problem with to low power. The IKEA power outlet have one good radio module but sadly its one large kondensator mounted behind it and doing the antenna not working optimal but still better then the OSRAM plugs. Xaiomi sensors dont like changing their parent and its not easy way forcing them to pairing with one nearby parent is they have shusing one. Deleting one sensor and re pairing then very near the parent and then its paired moving it to its end position its the easiest in deCONZ. Also trying repairing them then they is in the  end position can working but normally need many tries for success.

Also checking that your wifi is not overlapping with you zigbee network can being one reason way the sensors loosing connection to there parents.

baunan, 2020-11-15:
Thanks for all the advice given. 
What I did this weeken was to downgrade the deCONZ add-on to 6.3.0 which was the version before my problems started. Added an extension cable to the Conbee II to get it away from the computer and change the wifi 2.4 GHz channel to 1.  Added an additional IKEA repeater (USB version) The system is up again and looks healthy. The only issue I have now is these random dropout which makes the sensors unavailable for 15 seconds. I have quite a few dropouts during the day as one can see from the picture.

[image]
The log only says this:
2020-11-14 11:13:08 DEBUG (MainThread) [pydeconz.websocket] Websocket starting
2020-11-14 11:13:08 DEBUG (MainThread) [pydeconz.websocket] Reconnecting to deCONZ in 15.
2020-11-14 11:13:23 DEBUG (MainThread) [pydeconz.websocket] Websocket starting
2020-11-14 11:13:23 DEBUG (MainThread) [pydeconz.websocket] Websocket running

Could anyone please give an advice on how to investigate these dropout further? I this a plugin or a Home Assistant integration issue?

MattWestb, 2020-11-15:
Good that your Zigbee part have stabilisation and not making you crazy !!
I think its HA that have loosing the contact with deCONZ and trying reconnecting if i understanding the loog right.
But you is running deCONZ as one adon so its on the same Pi so something is locking the processes on the PI.
First making one backup of both HA and deCONZ if its starting being problems with your SD-Card (I have losing 3 SD-Cards that being killed on My Pi3B and Zero).
Also is good having one SD-Card that is fast for saving small data blocks and can handle it like the the A1 or A2 classes (Application Performance Class) but the Vx classes is not interesting for us (for video that saving large continues files and cant handle small blocks reading / writing fast).
HA have doing some rework of data storage (SQL-Light) that was blocking the processing and most is in place but the last is coming 0.118 so updating HA then the new release is coming in some days / week.

MattWestb, 2020-11-15:
For try getting the zigbee mesh working better you can also trying repairing sensors that is not using one router that is near them to using one better one so they is getting better signal but its not easy getting them doing that in deCONZ but in ZHA its possible forcing pairing devices thru one router that you like.

Mimiix, 2020-11-16:
@baunan Seems to be this: https://github.com/home-assistant/core/issues/37372

baunan, 2020-11-29:
Just to give an update.
What I did this weekend was to go to the step of installing Home Assistant on an Intel Nuc i3 gen10 using Proxmox. 
This setup looks solid and so far not a single dropout of the Conbee II connected sensors has occured. So it could be that a RPi with and an A1 SD card wasn't fast enough for this application or started to be broken. 
I am still on deconz version 6.3.0 (2.05.81).

closed 2020-12-20 by baunan

oywino, 2020-12-27:
Why is it that this post headline says: eCONZ 6.4.1 (2.05.84) ?
What's the difference in those version numbers?   What is 6.4.1 ? 
I have deCONZ 2.08  (never seen the number 6.4.x anywhere)

baunan, 2020-12-27:
Hi
The 6.4 is reffering to Home Assistant add-on version. As you can see from the attached picture of this addon's,  change log the deConz version is updated on regular basis.

[image]

oywino, 2020-12-27:
Please excuse my ignorance, but what is Home Assistant Add-on ??
I have had Home Assistant running for several years now, but I have never seen anything called "Add-on" !

baunan, 2020-12-28:
You can find it under the Supervisor menu and in the add-on store.

[image]

oywino, 2020-12-28:
I don't have any of those menus in my Home Assistant!