dresden-elektronik/deconz-rest-plugin #3587: Paulmann 500.67 ZigBee Remote Gent

Wurmi1986, 2020-11-03:
- If you're unsure whether device support was already requested, please ask for advise in our Discord chat: https://discord.gg/QFhTxqN
--> I asked in Discord an get the Info that are some Infos missing in the past. Now i try to give all required Infos.

## Device

- Product name: Smart Home Zigbee Fernbedienung Gent
- Manufacturer: Paulmann Licht GmbH
- Model identifier: RemoteControl 
- Device type : Please remove all unrelated device types. 
  - Remote
  

## Screenshots
- Node Info panel

[image]

  - Basic Cluster attributes in the Cluster Info panel.

[image]

- Level Control

[image]
[image]

-Scenes

[image]
[image]
[image]
[image]

-Color Control

[image]
[image]
[image]
[image]

-ZLL commissioning

[image]

If there are information missing let me know and i will try to provide them.

best regard

Wurmi1986, 2020-11-21:
Is there a chance to integrate the device? Or is there still information missing?
Best regards
Wurmi

BiberM, 2020-11-25:
I own the same remote and would like to use it with the Phoscon App.

I did some more screenshots from the attributes table of the above mentioned clusters:
- 0300 Color Control
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

- 0005 Scenes
[image]
[image]

- 0006 On Off
[image]
[image]

- 0008 Level Control
[image]
[image]

Addionally I turned on debug mode and dumped the output. While in debug mode I used the Signal Monitor PLugin and pressed some buttons on the remote.
[Paulmann_500.67.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5599146/Paulmann_500.67.txt)

What else do you need?

closed 2020-12-25 by stale[bot]

Artur-Fedjukevits, 2022-09-01:
I have the same remote control. I would also be very grateful to see this device as supported.

Kilowatt-W, 2022-10-23:
Same for me, hoping to get it online some day :-)

OderDochNicht, 2022-12-08:
Same for me! Would be nice if the Q remotes will work with deconz sometimes.