dresden-elektronik/deconz-rest-plugin #3588: Revolt NX-4911-675 Thermostat

kgorszczyk, 2020-11-04:
## Device

- Product name: Revolt NX-4911-675
- Manufacturer: (deCONZ GUI:) _TYST11_kfvq6avy
- Model identifier: (deCONZ GUI:) fvq6avy
- Device type : Thermostat
- Product page: [https://www.revolt-power.de/Weltweit-steuerbares-Heizkoerper-Design-Thermostat-NX-4911-919.shtml](https://www.revolt-power.de/Weltweit-steuerbares-Heizkoerper-Design-Thermostat-NX-4911-919.shtml)

## Screenshots

Node Info Panel:
[image]

Basic Cluster:
[image]

Mimiix, 2020-11-04:
@Smanar This looks like one of your funny tuya thingys. What interests me is that it's already given a name.

kgorszczyk, 2020-11-04:
After some Googeling it seems that it's the same device as mentioned in #1927
Original Brand Name: Siterwell
Original Model Number GS361A
According to the Manual (http://www.smart9.com.cn/wp-content/uploads/2019/10/Siterwell-ZigBee-Radiator-Thermostat-GS361A-H04-User-Manual.pdf) they do seem to be Tuya devices

kgorszczyk, 2020-11-06:
Duplicate of #3381

closed 2020-11-06 by kgorszczyk