dresden-elektronik/deconz-rest-plugin #3591: Danfoss Ally reporting NULL in windowopen?

CaCu15, 2020-11-04:
## Describe the question or issue you are having
I'm using the deconz API from Node-Red and have a question concerning the state reported by the Danfoss Ally thermostat (supported with #2983) .

If I open a window near the thermostat the state change is reported correctly with windowopen: "Open" in the state received from the DECONZ web socket. But after I close the window again, I receive a state containing windowopen: null  (not "null") for a long time of 40-50 min, see log output below.  

```
"{"lastupdated":"2020-11-04T08:57:56.784","on":false,"temperature":2119,"valve":1,"windowopen":"Hold"}"
"{"lastupdated":"2020-11-04T08:57:57.282","on":false,"temperature":2119,"valve":1,"windowopen":"Open"}"
"{"lastupdated":"2020-11-04T08:59:56.772","on":false,"temperature":2047,"valve":1,"windowopen":"Open"}"
"{"lastupdated":"2020-11-04T09:04:56.773","on":true,"temperature":2047,"valve":4,"windowopen":"Open"}"
"{"lastupdated":"2020-11-04T09:04:57.272","on":true,"temperature":2047,"valve":4,`"windowopen":null}`"
"{"lastupdated":"2020-11-04T09:06:56.774","on":false,"temperature":2047,"valve":3,"windowopen":null}"
"{"lastupdated":"2020-11-04T09:07:57.094","on":false,"temperature":2047,"valve":2,"windowopen":null}"
"{"lastupdated":"2020-11-04T09:08:57.416","on":false,"temperature":2047,"valve":1,"windowopen":null}"
"{"lastupdated":"2020-11-04T09:10:56.787","on":false,"temperature":2101,"valve":1,"windowopen":null}"
"{"lastupdated":"2020-11-04T09:26:56.800","on":false,"temperature":2151,"valve":1,"windowopen":null}"
"{"lastupdated":"2020-11-04T09:54:56.787","on":false,"temperature":2151,"valve":1,"windowopen":"Closed"}"
"{"lastupdated":"2020-11-04T10:27:15.845","on":false,"temperature":2180,"valve":1,"windowopen":"Closed"}"
"{"lastupdated":"2020-11-04T11:27:34.847","on":false,"temperature":2217,"valve":1,"windowopen":"Closed"}"
```

According to the ZigBee cluster specifications the attribute 0x04000 (eTRV Open Window Detection) can have the following values: 

- 0x00: Quarantine
- 0x01: Windows are closed
- 0x02: Hold ,Windows are maybe about to open
- 0x03: Open window detected
- 0x04: In window open state from external, but detected closed locally

And according to Danfoss user manual "When Open-window has been activated, the function is quarantined for 45 minutes." 

Hence my question: **Why is the API reporting `null` here?** Maybe the device reports "0x00" (Quarantine) and this is mapped to `null` erroneously and **should be "Quarantined" actually**? I saw states 0x01 - 0x03  reported by the device (see log above). I don't expect to see 0x04 because I'm not using external window sensor with the device currently. But I never saw the state 0x00 reported by the device and according the user manual of the device it should be in "Quarantine" for 45 min after the window-open has been activated. This timeframe fits quite well with the time frame that is reported as `null `in my log above.

## Screenshots

## Environment
Host System: Raspberry 3
Running Method: Rasbian Stretch
Device: RaspBee 1
Deconz Version: 2.05.86 / 15.10.2020
Firmware: 26390500

## deCONZ Logs

## Additional context

SwoopX, 2020-11-04:
Hi, could you please let me know where you got that one from?

> According to the ZigBee cluster specifications the attribute 0x04000 (eTRV Open Window Detection) can have the following values:
>
>     0x00: Quarantine
>     0x01: Windows are closed
>     0x02: Hold [...]

Other states than the one implemented I haven't seen so far. Regardless, I'm more than happy to have another look or amend where required.

CaCu15, 2020-11-05:
That is from the attachements in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2983:

There is a list of all attributes on a sheet in https://github.com/dresden-elektronik/deconz-rest-plugin/files/4818269/Draft.-.Zigbee.Cluster.Specifications.Danfoss.eTRV.xlsx

SwoopX, 2020-11-05:
Hm, thanks. Interestingly, the top entry seems to have been cut off in the PDF, so it has not been considered. You could add it to file `general.xml` yourself for the time being.

Will add it for the next release. thanks again for highlighting.

closed 2020-11-08 by manup