dresden-elektronik/deconz-rest-plugin #3593: IKEA Fyrtur problem / Current Position

MasterofKungfu9999, 2020-11-05:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: io.broker
 - Firmware version: 26660700
 - deCONZ version: 2.05.84 
 - Device:  ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

mtjoernelund, 2020-11-05:
I have had this issue as well. Have you tried to remove the battery from the blinds, wait 20-30 sec and reinsert. After that the blinds seem to report position. However to get Homebridge to pick this up, you need to restart homebridge. That seems to do the trick for me.

MasterofKungfu9999, 2020-11-05:
Thanks for your trick, but at my covering it didn't work. I need some other trick

Mimiix, 2020-11-06:
You probably need to setup bindings. Read here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-bind-server-clusters

MasterofKungfu9999, 2020-11-10:
@Mimiix 
I have a headless installation and so i can't acsess the gui. I have also tried to acess it via vnc on my raspberry but it didn't work

Mimiix, 2020-11-10:
@MasterofKungfu9999 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

MasterofKungfu9999, 2020-11-11:
i have tried my very best but i can't connect via xforwarding neither than vnc

Mimiix, 2020-11-12:
Than this is going to be a hard fix. 

I'd suggest waiting for the new beta then or re-pair the blinds a few times.

sinowine, 2020-11-16:
I have the same problem and only at closing. If I open the blinds within an automation the status shown without the problem. In other threads from the past I read that the problem was that the API sends a value of 254 instead of 255 and the blind thinks that is not fully closed. 

I hope that this problem could resolve without any bindings because I Have a headless setup too.

Mimiix, 2020-11-16:
To be fair: Without the deCONZ gui you are kind of in the dark with these things.

sinowine, 2020-11-18:
https://github.com/ebaauw/homebridge-hue/issues/630

"mtjoernelund commented on 11 Sep
Ok, solution was pretty simple. Remove the battery in the blinds and wait for a few seconds before putting them back. Then the blinds started to report state to Homebridge. After rebooting Homebridge, status also updated perfectly in Homekit.

So to summarize for anybody tripping over this thread in the future, these are the steps that made it work for me:

Pair the blind in Deconz Web GUI
Pair the remote in Deconz Web GUI
In the old web-app add the blinds to the group created by the remote
Restart Homebridge
Remove the battery from the blind for some seconds and then reinsert it
Restart Homebridge again"

After I removed the batttery yesterday out of my two (with deconz paired) fyrturs everything works like a charm and I see the level in percent on both blinds. I hope it will be stay on this status :)

MasterofKungfu9999, 2020-11-24:
@sinowine Your solution works.

closed 2020-11-24 by MasterofKungfu9999

Alex-joomla, 2020-11-26:
Same Problem here on HASS. The state of the blinds don't make it into Home Assistant :-(