dresden-elektronik/deconz-rest-plugin #3594: Device name

acme-szala, 2020-11-05:
Hi, I have a Lonsonho Tuya 3 gang light switch that is not fully working with Deconz.
I'm able to pair it but Vendor and Product information in Phoscon is partially wrong - only the first gang is identified correctly, other 2 gangs are shown as "Heiman" which is wrong.
Each gang can be switched on/off from Phoscon just fine but when switching a group that lights are part of it does not work at all.

If there is more info required, please let me know.

## Device

- Product name: Lonsonho (Tuya) X713A (3-gang, No Neutral, P/N: 11830066)
- Manufacturer: _TYZB01_xiuox57i
- Model identifier: TS0013
- Device type : Switch

## Screenshots

[image]
[image]
[image]
[image]
[image]

Smanar, 2020-11-05:
Hello, so first technical explanation, tuya switch use only 1 cluster for all switches, but worse only 1 endpoint too.
So It s not possible create 3 lights entries for this device, So I m using an ugly hack to create 2 more fake endpoint, the 02 and 03 don't exist in reality.

For the "heiman" it s already in the tuya PR, I bored too with that.
For the "unknow" as model id, I have corrected it, if you can compile the code to test ?

For the group feature, I think this device can't support it, it s not zigbee compliant, it don't support samee zigbee command than an ikea bulb for exemple.

acme-szala, 2020-11-05:
Thanks @Smanar for the explanation. My heart grows soo much when I see companies implement a standard with an individual "touch".
It's good that I only got one unit for testing :/

For the vendor name and model id data I'll just wait for your PR and current code to be released.
I'm using Home Assistant addon for Deconz, and even though I can compiling libde_rest_plugin.so from master just fine, it's just too much hassle to try to get it substituted in a Docker image without rebuilding new one.
BTW, I don't see my particular model id "TS0013" in "de_web_plugin.cpp" - is it required?

As for the group feature, this is something I'm most disappointed about.
I was hoping to be able to control all gangs simultaneously. Right now when I switch all gangs from HA there is a small but noticeable delay.

Smanar, 2020-11-06:
>I don't see my particular model id "TS0013" in "de_web_plugin.cpp" - is it required?

No it s not used for "router" fonctionnalities, a plug will have the on/off feature naturally but need to be written in the code to have the power measurement.

And TS0013 mean nothing for tuya, you have several device (with other type) with the same model ID.

closed 2020-11-15 by manup