dresden-elektronik/deconz-rest-plugin #3595: Zigbee switch module MS-104Z (Moes?)

petosak, 2020-11-05:
## Device

- Product name: Zigbee switch module MS-104Z (also MS-104BZ, MS-105Z, MS-105BZ) 
- Manufacturer: _TYZB01_iuepbmpv
- Model identifier: TS0121
- Device type : 
  - Switch - wall mount switch (https://zigbee.blakadder.com/Moes_MS-104Z-1.html)

## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
  - Node Info panel
[image]

  - Basic Cluster attributes in the Cluster Info panel.
[image]

[image]
[image]

Phoscon shows me as light with picture of plug...
[image]

But it's wall mount switch here:
https://zigbee.blakadder.com/Moes_MS-104Z-1.html

I have HASSIO on RPI4 with ConBee2

Mimiix, 2020-11-05:
Screenshots of the other clusters are missing. Please attach all of them.

petosak, 2020-11-06:
Sorry,
all clusters info here:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2020-11-06:
Thanks! that should do it.

Smanar, 2020-11-06:
But this device is working naturaly ?

The plug picture, is from the device itself, it s an information send by the device.

petosak, 2020-11-06:
It's different device.
It's light switch:
https://zigbee.blakadder.com/Moes_MS-104Z-1.html
And this is one side pf instruction manual:
[image]

Smanar, 2020-11-06:
But it s not our fault ^^, It s tuya that write in his device "PLUG", we are just reading this value, it change nothing in working mode ?

petosak, 2020-11-06:
Yeah. That's true. Tuya or Moes or WTF factory made this device clonning different device without sw changes. It works as they wrote, but in app looks weird. I have to suppress electrical measurement becase it really can't measure and change type of device.

closed 2020-11-06 by petosak

Thoky81, 2020-12-08:
Any idea how to pair this MS-104Z switch in deconz? I have followed the instruction: long press for beep twice then again long press for beep continuously and nothing is showing in phoscon or even in deconz UI. I am pairing it as a light. Thanks

petosak, 2020-12-08:
You'll see it as a light...
[image]

Thoky81, 2020-12-08:
@petosak  OK but I dont see it there. Can you pls advice how to set the device in pairing mode? Thanks

petosak, 2020-12-08:
When i connected device to line i heard sound and started pairing.
You should try turn on and off five times.
[image]

Thoky81, 2020-12-11:
Well I dont know what is wrong but still no success to pair the device in phoscon 😔

Smanar, 2020-12-11:
You don't see someting even in the deconz GUI ?
Hav you tried with logs enabled ? (Deconz / help / log)

Thoky81, 2020-12-12:
No. Long pressing the button on the switch for 10s then the sound beeps continuously so it should be in paring mode. But no success to connecting it to deconz..

Smanar, 2020-12-12:
Where have you see the long press during 10 s ?
You need to make 5 on/off, but to do that you need to add a physical switch, from this french forum on manual it s picture 4

https://easydomoticz.com/forum/viewtopic.php?f=28&t=10566&start=20

Thoky81, 2020-12-12:
I think I have different device.![Uploading 101988389...]![Uploading 101988391...]

Smanar, 2020-12-12:
Ha yes, right.
But you need to see something in deconz itself if the procedure is good, nothing special in logs (in deconz/help) ?

Thoky81, 2020-12-12:
I think I know why I am not able to pair it. I tried to pair new aqara motion sensor today and I didnt success. So the problem must be somewhere else. I am running the deconz in docker/synology and have recently installed ssd cache. I will try to install the deconz on rpi to see if it helps. Before I will check the logs one more time..

Thoky81, 2020-12-12:
So finally its working 🙏🏼. The issue was with last docker update for sure. I have loaded the previous network configuration from phoscon hiden menu and now its paired 😀 Thanks

Smanar, 2020-12-13:
Lol, finally ^^

Taemo, 2020-12-23:
> So finally its working 🙏🏼. The issue was with last docker update for sure. I have loaded the previous network configuration from phoscon hiden menu and now its paired 😀 Thanks

Hi I have a Tuya MS-104Z - pretty much identical (I think?) to the device you have here, I am trying to pair this with Home Assistant + Conbee ii Deconz - Rpi not Docker. I dont see to be able to pair it either. I have confirmed I am able to pair other devices.

Did you long press once on the modules button and get the Di-Di sound? or did you, long press-5sec-long press and get the long tone ?

Or did you use the power on-off  method ?

Cheers
Taemo

Taemo, 2020-12-27:
Ok all,

Posting in case anyone else finds this, I have managed to get this working with Home Assistant Conbee ii 2 and Deconz

The Tuya MS-104Z instructions are a bit misleading, the "reset" button of the device does not see to trigger pairing mode, or at least I cant get it too!

Instead wiring it all up and switching the switch input 5 time (with a rocker that is on-off so 10 switches, 5 power events) that puts into a continuous beeping tone, continuous being continuous beep forever :) - at that point it pairs fine with deconz.

Comes up as 

Vendor_TZ3000_zmy1waw6
ModelTS011F
Version 65

And the beeping stops.

The only annoying thing is that the switch input can be switched to fast (on and off) cycle and miss the transition, its not a problem if its only 1 transition ie on to off, but if you go on-off-on within a second it will not register the the off or on state well. Thus requires another switch event to take effect.

Hope all that helps someones else.

Jamie

DavidFFerreira, 2021-01-03:
Anyway to put it working with Zha?

Mimiix, 2021-01-03:
@DavidFFerreira that's not what this git is for.

Closing and locking this issue as it is resolved.