dresden-elektronik/deconz-rest-plugin #3596: IKEA KADRILJ After the deletion, fragments of the device are still visibile in iobroker and homebridge

MasterofKungfu9999, 2020-11-06:
## Describe the bug
After deleting the window cover in the phoscon-app the battery-status can be seen in iobroker and homebridge

## Steps to reproduce the behavior

## Expected behavior
no status can be seen furthermore and the device is clean deleted

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: 
 - Firmware version: 26660700
 - deCONZ version: 2.05.84
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? No

Mimiix, 2020-11-06:
@ebaauw @jey-cee is this a API issue or a implementation issue?

Jey-Cee, 2020-11-06:
Objects in ioBroker has to be deleted by hand. But if the objects are created again the state are reported by deConz.
Then it is an issue.

MasterofKungfu9999, 2020-11-06:
I have deleted it by hand. But the objects are created again.

Mimiix, 2020-11-06:
Are you able to provide RestAPI results?

MasterofKungfu9999, 2020-11-06:
sorry, in this way i'm a dau

Mimiix, 2020-11-06:
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

MasterofKungfu9999, 2020-11-06:
"config": {
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "5b3e9c6c10f7281b42d693dc2640dc9f",
  "lastseen": "2020-11-05T20:17Z",
  "manufacturername": "IKEA of Sweden",
  "modelid": "KADRILJ roller blind",
  "name": "KADRILJ roller blind ",
  "state": {
    "battery": 32,
    "lastupdated": "2020-11-05T11:17:12.542"
  },
  "swversion": "2.2.009",
  "type": "ZHABattery",
  "uniqueid": "00:0d:6f:ff:fe:a6:09:9d-01-0001"
}

MasterofKungfu9999, 2020-11-06:
It this right?

Mimiix, 2020-11-06:
That explains why you see it. I am not sure why it still is there tho. 

Can you try to delete it with the restapi?

MasterofKungfu9999, 2020-11-06:
How? is it described somewhere?

MasterofKungfu9999, 2020-11-06:
is this the method "delete"?

Mimiix, 2020-11-06:
Yes, on that same page  i linked earlier.

MasterofKungfu9999, 2020-11-06:
[{
  "success": {
    "id": "4"
  }
}]

Mimiix, 2020-11-06:
Is it gone now ?

MasterofKungfu9999, 2020-11-06:
yes