dresden-elektronik/deconz-rest-plugin #3598: HUE Apps and HUE Disco

MBj1703, 2020-11-06:
Hi,

I just started with deCONZ and want to move from IKEA and Philips Bridge to deCONZ.
So far, so good.

My question now:
I used HUE Disco App, which is not able to connect to deCONZ.
Also the Disco feature in HUE Essential App ist not working.

Is this feature not available in deCONZ?
This would mean, I have to stay at the Philips bridge 👎 

Also the original Philips HUE App doesn't seem to see the correct infos about Zones and Rooms on deCONZ.
If I setup a room in the App, it will be visible in Phoscon. Vice versa is not working.

At the moment im running Phoscon on a Nettop with Ubuntu 20.04 and the ConBee II Stick.
Version: 2.05.86 / 15.10.2020
Firmware: 26660700

Cheers
Markus

closed 2020-11-06 by close-issue-app[bot]

Mimiix, 2020-11-06:
As the template was not followed i normally not answer. But the answer on hue disco(entertainment) is already given [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2891#issuecomment-642201975)

MBj1703, 2020-11-06:
Thank you for your answer Mimiix, appreciate it, even I didn't follow the template.

Unfortunately the answer makes me not happy.
Even if only Philips HUE will be supported, would be fine for me.

Entertain/Disco is an essential key feature for me...and I think for many others as well.