dresden-elektronik/deconz-rest-plugin #3599: Known issues, problems, limitations w/ IKEA KADRILJ smart blinds?

mtressl, 2020-11-06:
## Describe the question or issue you are having

Okay, so first things first: the blinds were working before, either through deConz, Homebridge, or HomeKit.
But around 10 days ago they suddenly stopped working. So I did what I did before, tried to reconnect the blinds to deConz, reset the blinds, re-pair the blinds, delete them through deConz, Phoscon, even through the API directly.

Result is, the blinds connect, I can see them in deConz and Phoscon, can read stuff - but they do.not.move anymore.

Strangest thing is, that the battery indicator in deConz shows a low battery even though the battery is fully charged. I even got a new one from IKEA - same problem, see screenshot below.

I am running `deconz:latest` (2.05.86). Already downgraded several times to `deconz:stable` (2.05.84).
My RaspBee II is running the latest firmware (26660700) as well.

Ok, my question:

are there known issues that I did not find here in the reports that cause the behaviour I have?
I am really annoyed b/c of this behaviour, and before I throw the blinds away I wanted to check w/ you guys.

## Screenshots

[image]
[image]
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.05.86 / 2.05.84
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

wvuyk, 2020-11-06:
Did you try the battery trick? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-702055261

mtressl, 2020-11-06:
> Did you try the battery trick? [#1121 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-702055261)

Yes, you mean out-wait-in-again? I did... and that's why I got a new battery from IKEA, b/c I wanted to make sure it's not a defect battery pack.

wvuyk, 2020-11-06:
Not 100%. When you take it out, start a scan for new devices. Then put it back in. That at least is how I did it with success?

mtressl, 2020-11-07:
Just to completely understand your process:
- blinds are still added to the system (but faulty battery display and not moving /reacting)
- I take the battery out
- start new device discovery from Phoscon (or deConz?)
- re-insert the battery (but no reset on the blinds)

Correct?

wvuyk, 2020-11-07:
Yes, that is has been working for me in the past. Do not remove the device from deconz, the scan will refresh the device, that is all.
You could also try to read the basic cluster during the scan. For some devices this helps the refresh as well.

mtressl, 2020-11-08:
Unfortunately none of the proposed solutions do the trick @wvuyk - I cannot get the battery indicator to refresh and show a fully charged state (which the battery is). Any other hints and tipps how to fully reset the blinds, or make deConz completely starting from scratch with the blinds? Speaking of starting from scratch, removing the blinds through deConz, Phoscon, and through the API should be all to consider when doing this, correct?

wvuyk, 2020-11-10:
I think the only trustworthy removal is through Phoscon, but am not sure on that as I keep seeing ghost devices returning back in my installation in the deConz map here. Whatever I try, they keep comming back But that could also be remnants of old beta stuff running here...

I found an old remark here in my notes that for some blinds you would have to manually set up the attribute reporting, is this something you have tried?

Other then that, I am out of suggestions....

mtressl, 2020-11-10:
> manually set up the attribute reporting

No I have not tried that. Can you point me in any direction where I can read about it?

wvuyk, 2020-11-10:
I am sure it wil be somewhere here too, but here is a link to another post for a device: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3226#issuecomment-685960083, I do not know if this was posted in a wiki as well.
Also[ here on my own forum](https://forums.homeseer.com/forum/hs4-products/hs4-plugins/lighting-primary-technology-plug-ins-aa/jowihue-w-vuyk-aa/1398707-using-a-ikea-smart-blind-fyrtur-or-kadrilj)

mtressl, 2020-11-10:
Manual attribute reporting did not help. I also managed to re-flash the firmware of the blinds, but no success in controlling.
What I start seeing is the following line in my Homebridge logs:
`[11/10/2020, 8:15:48 PM] [Philips Hue & deCONZ (ZigBee)] Window covering: set homekit last boot from Tue Nov 10 2020 20:12:32 to Tue Nov 10 2020 20:15:48`

Does that point towards an endless reboot loop on the blinds? Might they actually be broken?

mtressl, 2020-11-11:
@ebaauw 
Sorry for tagging you out of context, please see my problems above.
As it seems the homebridge-hue plugin gives me the following log entry, not sure if this might be related to my general issue: `Window covering: set homekit last boot from Wed Nov 11 2020 17:20:33 to Wed Nov 11 2020 17:23:25`

Dodoooh, 2020-12-18:
is there a newer / better way. Unfortunately I still have problems with adding the roller shutter.
I have tried it as described here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-524617659

in home assistant i unfortunately only get the battery status displayed, but not the shutter itself. but i can open / close it in the deconz GUI

Smanar, 2020-12-18:
I think you just don't find it in HA.
If you can find it in phoscon, it mean it exist in APÏ, so need to be visible in HA too.

You have no fonction to resynchronise ?

Dodoooh, 2020-12-18:
> I think you just don't find it in HA.
> If you can find it in phoscon, it mean it exist in APÏ, so need to be visible in HA too.
> 
> You have no fonction to resynchronise ?

but it should be displayed in home assistant, i mean how else could i control the shutters? 

I also tried with the NodeRed - Deconz plugin which has direct api access, there it also does not find the shutter :/

Smanar, 2020-12-18:
How many device have you ?
Have you take a look in the light device list ? (with Node red if it show the complete list ?)

Dodoooh, 2020-12-18:
i have 70 nodes

I can't find anything in node red except the battery. i was able to integrate it half a year ago, and there i could find it in node red as well as in home assistant....

i switched to a new infrastructure and wanted to do everything from scratch. to eliminate first errors for all time. was apparently not the best idea....
[image]
[image]

Smanar, 2020-12-18:
You are not loking a good place, in the light section there is no battery device.

You have an API key that you are using in third app, just go on 

`http://IP:PORT/api/KEY/lights`

IP znd PORT are the same you are using for phoscon.

SwoopX, 2020-12-18:
The blinds do not seem to be paired correctly, you can see that on the "unusual" names.

Dodoooh, 2020-12-18:
@Smanar. no nothing like that can be found here. Only my lamps and other such devices.. 

@SwoopX how to pair them "even more" ? :) I can control them via Web GUI

SwoopX, 2020-12-18:
> I can control them via Web GUI

Any device can be controlled from the GUI once they successfully received the network settings and are allowed to join the network. But this is just the first half of the join processes. The second does the lights and sensors creation, which can be seen by the device NWK address as its name and confirmed by you not being able to find any REST resource.

Try entering light search and then open/close the blind. If that's unsuccessful, remove the batteries for 10 secs and try again.

Dodoooh, 2020-12-18:
thank you @SwoopX but it doesn't look like it solved the problem. :/ i've done your process several times now and i'm still in the same situation.

SwoopX, 2020-12-18:
Hm, if a sensor search also doesn't help, I'm out I'm afraid. Don't have any :/

Smanar, 2020-12-18:
But if it work on phoscon, it mean the API entry is somewhere, no ?

Dodoooh, 2020-12-23:
so I do not know what it was, possibly an early Christmas miracle ;) (I sat surely half a day with three shutters in a small storage room to be close enough to the deCONZ stick) 

After hours of various factory resets and re-adds I came across a Github post (link will come as soon as found): 

Once shortly press both buttons. 
Then press both buttons for about 30 seconds +
After that normal factory reset.

So it was found immediately in the Phoscon WebGUI and added I did not have to bind anything via VNC. 

Thank you again very much for your support! @SwoopX @Smanar

wvuyk, 2020-12-23:
Both buttons of what? the remote control? I do not remember the blinds having multiple buttons?

closed 2021-02-08 by Mimiix

Dodoooh, 2021-02-08:
> Both buttons of what? the remote control? I do not remember the blinds having multiple buttons?

from the blinds, these have two buttons