dresden-elektronik/deconz-rest-plugin #3602: REST API responds to requests when Zigbee messages are queued, rather than sent

hanskroner, 2020-11-06:
## Describe the bug

Rest API requests seem to be responded to once their payload is turned into Zigbee messages and these messages queued inside deCONZ - before the messages are sent over Zigbee. This gives API clients no visibility into whether the API is ready to accept new requests or not.

## Steps to reproduce the behavior

I came across this issue while using the excellent Homebridge-Hue <https://github.com/ebaauw/homebridge-hue> plugin for Hombridge. I'm connecting Homebridge-Hue to the deCONZ REST API and exposing my Hue bulbs to HomeKit. Inside HomeKit, I'm creating a scene with more than 6 bulbs. With those bulbs initially off, activating the scene results in various lights not reacting. This is caused because the number of Zigbee messages generated by the scene exceed what deCONZ's queue can hold, and the messages come in faster than deCONZ can send them out the Zigbee network - some messages get dropped. Homebridge-Hue has built-in mechanisms for limiting the number of parallel connections it opens to an API, but since the REST API is replying to all requests as soon as it has queued them - without providing an indication of the queue's size or how full it is - Homebridge-Hue's mechanisms never kick in and it continues to open new requests after every reply it gets.

Without Homebridge-Hue, the issue can be reproduced by sending more than 6 parallel PUT requests to turn on, set brightness, and set xy color to `lights/LIGHT_ID/state` for more than 6 bulbs that are initially off.

## Expected behavior

The expected behavior is that all sent requests get processed - it's acceptable that as the number of requests increases, the latency also increases.

I can think of a few scenarios that would've been preferable:

1) deCONZ's internal queue is greatly increased. As a new comer to the project, it's difficult to comprehend how a piece of software that requires such powerful hardware to run on, has an outbound message queue limited to 20 Zigbee messages. That allows controlling no more than 6 lights in quick succession. It's understandable and acceptable that latency in message delivery will increase along with the queue's size, but at least all messages would be delivered.

2) The REST API informs API clients when it's ready to process new requests by replying when the request has been sent out to the Zigbee network, instead of when it has been queued in the deCONZ firmware. This, along with the API being able to hold a reasonable amount of REST API requests for processing, would make deCONZ behave closer to a Hue Bridge. I don't have first-hand experience with this, but I was lucky enough to be assisted through root causing my issue by @ebaauw who does.

3) The REST API informs API clients about the state of its queue, for example by replying immediately to API requests informing clients whether a message was sent or queued. If a message was queued, the API also has the possibility of suggesting an estimated time in the future when the request will be serviced, given then state of its queue and the number of Zigbee messages that would result from the request's payload, and sending a confirmation back to the client once the request was processed. Something like this would provide clients with good insights into how many open connections they can keep, and how quickly they can send requests to the API - at the cost of requiring them to be more sophisticated.

## Screenshots

## Environment

 - Host system: PC
 - Running method: Debian
 - Firmware version: 26580700
 - deCONZ version: 2.05.85
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

Pastebin link for compactness: <https://pastebin.com/hm0GP3UG>

## Additional context

hanskroner, 2020-12-05:
Ping

kmplngj, 2020-12-31:
Ping

hanskroner, 2021-02-08:
Was this closed because the state bot labeled it stale, or is it closed as "won't fix"?

Mimiix, 2021-02-08:
I've closed all in batch as the bot seemed to malfunction.

hanskroner, 2021-02-08:
> I've closed all in batch as the bot seemed to malfunction.

Thanks for reopening!

hanskroner, 2021-03-08:
Ping