dresden-elektronik/deconz-rest-plugin #3604: Modify state while device is off

radi-ka, 2020-11-06:
## Describe the question or issue you are having
Hello there,

tldr: i want to be able to modify my lights state, while they are turned off. How can i achieve this?

I'd like to build an external service that updates the brightness & color temp throughout the day, you know, warm and dark at night cold and bright in the morning, but with a smooth daily transition.

But to avoid "flicker" when my motion sensors trigger a light group they need to be already aware of their new state -> the state needs to be modifiable while the light is off.

The REST Api does not allow to set the brightness, while the light is off. ("... bri, is not modifiable. Device is set to off.")
So I tampered with the code a bit. I removed lines 973 through [980](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/5d8aa096242dd69977ccf105cc22c3f816644a31/rest_lights.cpp#L980)
And... SUCCESS! the api responds with a success and the lights state is off but shows my updated value.
BUT: when i now send a simple "on" message, my state is ignored and the light turns on with bri=254

So i guess there is another state somewhere that is used here... :(

Any ideas how to solve my original problem in a different way or how to modify the state for good?
Or.. what am i missing?

thanks :)

## Screenshots

## Environment

 - Host system: PC
 - Running method: Marthoc Docker container
 - Firmware version: 0x26350500
 - deCONZ version: 2.05.85
 - Device: ConBee I
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context
device used: philips hue bulb

ebaauw, 2020-11-06:
As per the Zigbee standard, devices ignore requests to update their state while off.  By disabling the warning, the REST API plugin updates its cache copy of the light state, but the actual light state doesn't change.

radi-ka, 2020-11-06:
Thank you, I feared as much.

A workaround might be to have an incoming "on" message to append the state from that cache.

This would only work for api calls and not for zigbee messages as the motion sensor trigger. Hm...

radi-ka, 2020-11-09:
> As per the Zigbee standard, devices ignore requests to update their state while off. By disabling the warning, the REST API plugin updates its cache copy of the light state, but the actual light state doesn't change.

closed 2020-11-09 by radi-ka

radi-ka, 2020-11-10:
just for future reference:
it seems my goal can be achieved with rules. they allow to specify brightness and color settings, if configured manually.
i just need to keep them updated through the day.

radi-ka, 2020-11-13:
the easiest way to achieve what i intended is to use light groups. their state can be modified, even if the group is switched off

ebaauw, 2020-11-13:
Groups don’t have a state, except for `all_on` and `any_on`, which are aggregated from the member lights `state.on`.  The `action` is write-only and simply echoes back the last values PUT.

agg23, 2020-12-08:
@radi-ka Did you end up finding a solution to this? I have been dealing with the same problem for years and would love to make improvements.

radi-ka, 2020-12-11:
@agg23 ah, yes. I did find a workaround.

At first I wanted to modify the group behavior within deCONZ. It should read the "action" state of a group when I send a specific command. But then I would have to apply my patch each time that I wanted to use a new version of deCONZ. Even with proper automation... I am too lazy for that.

So I implemented a small service, that updates my rules once every X seconds with the new brightness and color temperature values. It also sends an action with thoso values to groups that are already on.

I ll try to put it up on github in the coming days, but it is heavily tuned to my needs and generates all my rules, so not sure how useful it is for you. Maybe I can separate some stuff but we'll see.