dresden-elektronik/deconz-rest-plugin #3605: Commands not executed but new state reported by Deconz with 26660700

ReX1983, 2020-11-06:
## Describe the bug

Commands sent to the switches don't get executed but the new state is updated in Deconz / Home Assistant.

E.g. an automation turns on a lumi.ctrl_neutral device but the switch doesn't trigger even if gets reported as on in the UI.
This leads to inconsistent states and often I don't know if my lights are on off (e.g. I realised that a garden light has been on for days).

If I turn it off and on again it actually switches on (so it still connected to Deconz).

It seems that the commands get lost, even if the devices are online.

## Steps to reproduce the behavior

1. The original state of the switch is off
2. Turn on the switch (via the Home Assistant UI or Home Assistant automation)
3. The switch remains off but Home Assistant reports the state as on

## Expected behavior

1. The original state of the switch off
2. Turn on the switch (via the Home Assistant UI or Home Assistant automation)
3. The switch turns on

## Screenshots

## Environment

 - Host system: (Raspberry Pi 3)
 - Running method: (Marthoc Docker container)
 - Firmware version: (26660700)
 - deCONZ version: (2.05.84)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II

### deCONZ Logs

## Additional context

Mimiix, 2020-11-07:
Can you please add logs?

ReX1983, 2020-11-11:
> Can you please add logs?

Will try to as soon as I have a chance.

Lindansaren, 2020-11-13:
I'm having the exact same behavior. Sometimes random lights and switches does not react to commands but the state is updated in gui (HA and deconz). Frustrating. 
Running the conbee 2 on a pi4. (Addon)

ReX1983, 2020-11-15:
I managed to get a log of the issue. Unfortunately it happens also with the latest 2.05.88 deCONZ version.

Some context: I am using one rocker of the "Kitchen Cooking" 0x00158D00015723E7 Aqara no neutral wall switch (double rocker) to toggle one line of the "Kitchen Dining" 0x00158D00014DF5B2 Aqara no neutral wall switch (double rocker). This is implemented via a Home Assistant automation.

This is the sequence of events:
1. Kitchen Dining was off until 12:23:21:667, when I pressed Kitchen Cooking to turn it on. This time it worked.
2. At 12:23:22:026 I pressed again Dining, but Cooking didn't turn off, even if Home Assistant reported it as off.
3. I pressed again Dining at 12:23:23:081, following this event Home Assistant reported Dining as on. 
4. Finally I pressed Cooking at 12:23:23:464, when it managed to turn off Dining for real (this time it worked).

> 12:23:20:452 ZCL attribute report 0x00158D000183A976 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F
> 12:23:20:454 0x00158D000183A976 extract Xiaomi special attribute 0xFF01
> 12:23:20:456    64 on/off 0
> 12:23:20:458    65 on/off 0
> 12:23:20:459    6e unknown 0 (0x00)
> 12:23:20:461    6f unknown 255 (0xFF)
> 12:23:20:462    01 battery 3300 (0x0CE4)
> 12:23:20:464    03 temperature 23 °C
> 12:23:20:466    05 RSSI dB (?) 117 (0x0075)
> 12:23:20:468    08 unknown 9749 (0x2615)
> 12:23:20:469    0a unknown 5915 (0x171B)
> 12:23:20:471    99 unknown 0 (0x00000000)
> 12:23:20:472    9b unknown 0 (0x0000)
> 12:23:21:667 ZCL attribute report 0x00158D00015723E7 for cluster: 0x0006, ep: 0x02, frame control: 0x18, mfcode: 0x0000
> 12:23:22:024 [INFO] - No button handler for: lumi.ctrl_neutral2 endpoint: 0x04 cluster: ONOFF (0x0006) command: 0x0A payload[0]: 000
> 12:23:22:026 ZCL attribute report 0x00158D00015723E7 for cluster: 0x0006, ep: 0x04, frame control: 0x18, mfcode: 0x0000
> 12:23:23:081 ZCL attribute report 0x00158D00015723E7 for cluster: 0x0006, ep: 0x02, frame control: 0x18, mfcode: 0x0000
> 12:23:23:180 delay sending request 16 dt 1 ms to 0x00158D00014DF5B2, ep: 0x02 cluster: 0x0006 onAir: 1
> 12:23:23:188 delay sending request 16 dt 1 ms to 0x00158D00014DF5B2, ep: 0x02 cluster: 0x0006 onAir: 1
> 12:23:23:224    0x00158D00014DF5B2 force poll (2)
> 12:23:23:429 ZCL attribute report 0x00158D00014DF5B2 for cluster: 0x0006, ep: 0x02, frame control: 0x18, mfcode: 0x0000
> 12:23:23:463 [INFO] - No button handler for: lumi.ctrl_neutral2 endpoint: 0x04 cluster: ONOFF (0x0006) command: 0x0A payload[0]: 000
> 12:23:23:464 ZCL attribute report 0x00158D00015723E7 for cluster: 0x0006, ep: 0x04, frame control: 0x18, mfcode: 0x0000
> 12:23:23:740    0x00158D00014DF5B2 force poll (2)
> 12:23:23:941 ZCL attribute report 0x00158D00014DF5B2 for cluster: 0x0006, ep: 0x02, frame control: 0x18, mfcode: 0x0000

ReX1983, 2020-11-20:
I have had again the same issue today: command sent to a switch, not executed and wrong state reported.

Is there anyone willing to look into this? Happy to provide the new log, if anyone is interested.

Smanar, 2020-11-20:
It's "almost" normal.

When you send a command, the api is updated before sending the command, an easy test you can done.
- send "off" order to a bulb
- the bulb is "off" in phoscon/third app
- unplug the bulb
- send "on" order
- the bulb is "on" on phoscon/third app

On normal working mode it s not a problem, because the device is sending report, but it can be a problem if your device have connexion problem for exemple (it miss command or don't send report)

closed 2020-12-25 by stale[bot]