dresden-elektronik/deconz-rest-plugin #3606: virtual daylight sensor triggers my outdoor lamps one hour early

bee-con, 2020-11-07:
## Describe the bug

Ever since daylight saving time changes (on the 25th of october), the built-in virtual daylight sensor triggers my outdoor lamps one hour early. The timezone setting is correct for my location, and the time on my raspberry Pi4 is also correct.
See also https://discord.com/channels/494922323518947329/714772122236354644/772029441647050793

## Steps to reproduce the behavior

Create a group of lamps that is controlled by motion sensor, and set this control to be active from dusk till dawn (thus using the built-in virtual daylight sensor). Set your timezone and time correctly, and then wait for daylight saving time to change from summertime to wintertime. Observe that the built-in virtual daylight sensor now triggers one hour early.

## Expected behavior

The built-in virtual daylight sensor should be aware of daylight saving time changes.

## Screenshots

See also https://discord.com/channels/494922323518947329/714772122236354644/772029441647050793

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.84
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

not applicable.

## Additional context

not applicable.

ebaauw, 2020-11-07:
Did you double-check `localtime` and `timezone` in `/config`?

bee-con, 2020-11-07:
> Did you double-check `localtime` and `timezone` in `/config`?

Insufficient information, please clarify. I'm guessing: Open a PuTTy session, goto where and issue what command?

ebaauw, 2020-11-07:
Do a GET of `/config` over the API.  See the README for a link to the API documentation.

bee-con, 2020-11-08:
> Do a GET of `/config` over the API. See the README for a link to the API documentation.

Thanks, got it. That was neither easy nor intuitive, but here's what I got;

```
    "UTC": "2020-11-08T08:48:57",
    "localtime": "2020-11-08T09:48:57",

```

That seems to be correct and corresponds to the setting in phoscon:
[image]

ebaauw, 2020-11-08:
Ok, that’s looking good.

Just to be sure, what are the `sunriseoffset` and `sunsetoffset` values in the `config` of the daylight sensor?  Typically it’s the first sensor, so do a GET of `/sensors/1/config`.

This wouldn’t explain the change on daylight savings time changing, though.  My best guess is that you’re using a schedule using UTC instead of local time.  I don’t think it’s possible to create a rule on UTC.

How did you setup the triggering?  In Phoscon?  Does it use a rule or schedule?

bee-con, 2020-11-08:
> Just to be sure, what are the `sunriseoffset` and `sunsetoffset` values in the `config` of the daylight sensor? Typically it’s the first sensor, so do a GET of `/sensors/1/config`.

GET of `/sensors/1/config didn't work, here's the beginning of the output of GET /sensors

```
{
	"1": {
		"config": {
			"configured": true,
			"on": true,
			"sunriseoffset": 30,
			"sunsetoffset": -30
		},
		"etag": "ad5fe91533b4a438012d13976eaa882e",
		"manufacturername": "Philips",
		"modelid": "PHDL00",
		"name": "Daylight",
		"state": {
			"dark": false,
			"daylight": true,
			"lastupdated": "2020-11-08T07:26:19.976",
			"status": 160,
			"sunrise": "2020-11-08T06:35:54",
			"sunset": "2020-11-08T15:55:31"
		},
		"swversion": "1.0",
		"type": "Daylight",
		"uniqueid": "00:21:2e:ff:ff:05:4f:03-01"
	},

```
> How did you setup the triggering? In Phoscon? Does it use a rule or schedule?

In Phoscon, sensor control of group. 

[image]

ebaauw, 2020-11-08:
So the offsets probably cause the trigger to happen half an hour before/after sunrise or sunset (I never remember the meaning of the sign).  The `sunrise` and `sunset` times (in UTC) seem correct for Berlin.  From the perspective of the REST API plugin, everything looks correct.

I think it's an issue with Phoscon creating the wrong rules (just remembered that schedules only take absolute time) or with CLIPDaylightOffset, if they already use that.

@Mimiix can you transfer this issue to the Phoscon repo?

bee-con, 2020-11-08:
> From the perspective of the REST API plugin, everything looks correct.
>
> I think it's an issue with Phoscon creating the wrong rules (just remembered that schedules only take absolute time) or with [...]

Thank you!

bee-con, 2020-11-08:
BTW, where is the Phoscon repo? I could only find one for beta versions, and absolutely nothing is going on there...
https://github.com/dresden-elektronik/phoscon-app-beta