dresden-elektronik/deconz-rest-plugin #3609: IKEA TRADFRI remote do not generate events

dn0sar, 2020-11-08:
## Describe the bug

Buttons in the IKEA TRADFRI remote do not work on the current most up to date rest-api plugin version (3a0751abaf9eb8014ebfa90f8d51127da3762b22).

## Steps to reproduce the behavior

  1. Use plugin version 2.05.84
  2. Click on `on` button
  3. Lights turn off
  4. Update to 3a0751abaf9eb8014ebfa90f8d51127da3762b22
  5. Click on `on` button
  6. Nothing happens
  7. Downgrade to 2.05.84
  8. Button works again

Note that on 3a0751abaf9eb8014ebfa90f8d51127da3762b22 the remote does not even pair correctly cause as a step of the pairing it is asked to press the `on` button for 10s, but the procedure fails there as no button press is detected.

## Expected behavior

## Screenshots

The button press is recognized and an event is generated from it.
## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian / Home Assistant deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: see above
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

MattWestb, 2020-11-08:
Is the remote bounded to one (light) group that the bulbs is in ?
Or do you using the remote as one HA controller that is doing all things thru deCONZ magic ?

dn0sar, 2020-11-08:
I'm using it through an automation in home assistant. It is not bound to any light group.

MattWestb, 2020-11-08:
Then its some problems with the parsing the light commands to HA events  in deCONZ ;-((

canim60, 2020-11-08:
> 
>     7. Downgrade to 2.05.84
> 

Can you say me how to downgrade in Rasperry pi? with commands?

dn0sar, 2020-11-08:
> Then its some problems with the parsing the light commands to HA events in deCONZ ;-((
> Way not binding it to one light groupe then its working then your HA / deCONZ is offline ala ZLL ?

I don't think that is the problem. The pairing in phoscon would work otherwise. Instead it gets stuck in the 'press the on button for 10s' phase and then it says error pairing.

MattWestb, 2020-11-08:
OK then you have problem getting it paired in deCONZ :-((
Some times I must using "other" then the IKEA as manufacturer is not working for IKEA controllers.
Also deleting unnamed devices (with name as 0xXXXX) in deCONZ GUI can helping but it can being very tricky is its cont like to pairing new devices.

dn0sar, 2020-11-14:
It was not loading properly the `button_maps.json` file. I had an old version in `/usr/share/deCONZ/devices/button_maps.json` that was not working well with the version I manually compiled. Solved by replacing the file manually with the most up to date version present in the repo.

closed 2020-11-14 by dn0sar