dresden-elektronik/deconz-rest-plugin #3610: ConBee II cannot be reflashed and cannot connect to IKEA devices since firmware 26660700

hai-nguyen-van, 2020-11-08:
## Describe the bug
After flashing the firmware to the latest version `deCONZ_ConBeeII_0x26660700.bin.GCF`, it seems that it is unable to control my IKEA devices. Hence, I tried to reflash again but it is now unable to open the device and reflash

## Screenshots
[image]

```bash
$ GCFFlasher_internal -l 
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM1     | 0x1CF1 | 0x0030  | DE2212065  | ConBee II
```

```bash
$ sudo GCFFlasher_internal -t 60 -d /dev/ttyACM1 -f deCONZ_ConBeeII_0x26660700.bin.GCF
...
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
Reboot device /dev/ttyACM1 (ConBee II)
retry, failed

785: Error: failed to open device, status: Permission error while locking the device (2)
```

## Environment
 - Host system: Debian Buster
 - Running method: command line
 - Firmware version: 26660700
 - deCONZ version: (2.xx.yy)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Z-Wave USB key, External Hard-drive

## deCONZ Logs
```
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
14:19:27:563 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
14:19:27:566 CTRL. 3.27.214:19:27:606 COM: /dev/ttyACM0 : 0200 (0x0658/0x0200)
14:19:27:606 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
14:19:27:606 COM: /dev/ttyS0 :  (0x0000/0x0000)
14:19:27:606 ZCLDB init file /home/domus/.local/share/dresden-elektronik/deCONZ/zcldb.txt
14:19:27:688 parent process -bash
14:19:27:688 gw run mode: normal
14:19:27:688 GW sd-card image version file does not exist: /home/domus/.local/share/dresden-elektronik/deCONZ/gw-version
14:19:27:689 DB sqlite version 3.27.2
14:19:27:689 DB PRAGMA page_count: 33
14:19:27:689 DB PRAGMA page_size: 4096
14:19:27:689 DB PRAGMA freelist_count: 0
14:19:27:689 DB file size 135168 bytes, free pages 0
14:19:27:689 DB PRAGMA user_version: 7
14:19:27:689 DB cleanup
14:19:27:690 DB create temporary views
14:19:27:690 DB view [0] created
14:19:27:690 DB view [1] created
14:19:27:690 DB view [2] created
14:19:27:691 DB view [3] created
14:19:27:691 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
14:19:27:691 sql exec SELECT key FROM config2
14:19:27:691 sql exec SELECT key,value FROM config2
14:19:27:691 Load config UTC: 2020-11-07T20:24:41 from db.
14:19:27:691 Load config announceinterval: 45 from db.
14:19:27:691 Load config announceurl: https://phoscon.de/discover from db.
[... 254 lines cut ...]
14:20:06:304 wait reconnect 5 seconds
14:20:07:303 wait reconnect 4 seconds
14:20:07:804 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:20:07:804 failed to reconnect to network try=4
14:20:08:303 wait reconnect 3 seconds
14:20:09:303 wait reconnect 2 seconds
14:20:10:304 wait reconnect 1 seconds
14:20:10:318 COM: /dev/ttyACM0 : 0200 (0x0658/0x0200)
14:20:10:318 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
14:20:10:318 COM: /dev/ttyS0 :  (0x0000/0x0000)
14:20:10:318 auto connect com /dev/ttyACM1
14:20:10:322 Serial com disconnected, reason: 4
14:20:10:322 device disconnected reason: 4, index: 0
14:20:11:304 wait reconnect 15 seconds
14:20:11:319 COM: /dev/ttyACM0 : 0200 (0x0658/0x0200)
14:20:11:319 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
14:20:11:319 COM: /dev/ttyS0 :  (0x0000/0x0000)
14:20:12:304 wait reconnect 14 seconds
14:20:12:804 failed to reconnect to network try=5
14:20:13:304 wait reconnect 13 seconds
14:20:14:303 wait reconnect 12 seconds
14:20:15:304 wait reconnect 11 seconds
^C14:20:15:842 shutdown after signal(2)
14:20:15:845 saved node state in 0 ms
```

SwoopX, 2020-11-08:
This looks like deconz is still running. Have you shut it down?

hai-nguyen-van, 2020-11-08:
Ok thanks. So I was able to reflash it to an older version. I use Jeedom to manage it and when I press the button "Retrieve API key", each time the value changes (in the past, it was unique). Is this normal ?

[image]

closed 2020-12-19 by stale[bot]