dresden-elektronik/deconz-rest-plugin #3611: Zemismart 4 buttons ZigBee Scene Switch

tr1plus, 2020-11-08:
Hi all - I just got this button and would like to connect it to my Deconz set-up. Let me know if any info is missing. This is my first device request 👍 
Thanks!

## Device

- Product name: Zemismart 4 buttons ZigBee Scene Switch
- Manufacturer: _TZ3000_vp6clf9d
- Model identifier: TS0044
- Device type : Please remove all unrelated device types. 
  - Switch
  - Other: (I assume Scene Switch is just a normal switch)

https://www.aliexpress.com/item/4001365407192.html?spm=a2g0o.store_home.hotSpots_6000239680572.0
https://www.zemismart.com/products/zemismart-4-gang-tuya-zigbee-wireless-switch

## Screenshots

  - Endpoints and clusters of the node
[image]
  - Node Info panel
[image]
  - Basic Cluster attributes in the Cluster Info panel.
[image]

[image]

[image]

Smanar, 2020-11-09:
Hello, I m trying to include your device, have you something to compile the code to test ?

Take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But :
- better to have the last deconz version available (you will have the next stable in 3/4 days) or can try with the beta
- you can make a backup of the file "libde_rest_plugin.so" (in another folder), so easy to make rollback if you have problem with my code.

tr1plus, 2020-11-09:
Hey

I haven't done this as I'm using the deconz add-on in homeassistant.

I do have a vmware esxi server so I can spin up a temporary VM. Could you just confirm if this makes no changes to the conbee II stick. With that I mean - when I link the stick back to my homeassistant VM it should all still work as before on that vm, right?

Also if I understand your comment I should wait for the week-end to spin up this VM to test? Or will your changes already be included in the new stable?

Just a quick question also - do I first need to install the deconz application as per here https://phoscon.de/en/conbee2/install and then execute your commands? Or are your commands sufficient?

Thanks for the support!

Smanar, 2020-11-09:
No, don't worry, settings are stored on the device, but you don't change them and other informations (devices) are stored in a DB stored locally.
But it can be harder to install deconz (and the dev environnement) on virtual machine, so not sure it will work.
You can too use the phoscon feature to make a backup, and make a copy of the lib file, to rollbacl fastly.

No the change will be not in the next Stable in 3/4 days, but in the beta  the 15 th, so if I m wrong the correction will be in the next one, 1 month after.

And Yes you need to install first deconz, as you need a recent version, you can wait for the next stable, else better to use beta version.

tr1plus, 2020-11-11:
Hey

I tried in a VM since I couldn't find another solution quickly 👍 
I think I did it with the stable release however - hope that is no big issue.

In the deconz applicationI do see it being a "switch" (see first screenshot), however in the phoscon front-end it's still just 4 lights and nothing is added in my switches. I did add it by selecting 'other' switch.

[image]
[image]

Not sure if this is of any help to you?

Thanks!

Smanar, 2020-11-11:
Same problem than here > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3524

Wireless device are working like wired one ...

I have updated the code, can you make another try pls ?

tr1plus, 2020-11-11:
For me it shows up in the deconz client but not at all on the phoscon web app (not even a light like before)

[image]

Smanar, 2020-11-11:
Nothing in "sensor" part ?
Can you take a look direclty in the api, using third app or `http://IP:PORT/api/API_KEY/sensors`

tr1plus, 2020-11-11:
Hey

For whatever reason I see it as a "light" now (but only 1 instead of 4). When I remove this light the switch node in deconz client is also gone.

Last modified date of the libde_rest_plugin.so is 2 hours ago so that should be the latest one you shared.

[image]

[image]

Item 22 and 23 in the json below look to be relevant. Not sure if this is of any help

Output of the rest api before deleting and adding the node. (after doing the same below)

```json
{
    "1": {
        "config": {
            "configured": true,
            "on": true,
            "sunriseoffset": 30,
            "sunsetoffset": -30
        },
        "etag": "4d5a471fe5fcc594892f432311156efb",
        "manufacturername": "Philips",
        "modelid": "PHDL00",
        "name": "Daylight",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2020-11-11T19:58:11.725",
            "status": 230,
            "sunrise": "2020-11-11T06:53:26",
            "sunset": "2020-11-11T16:03:33"
        },
        "swversion": "1.0",
        "type": "Daylight",
        "uniqueid": "00:21:2e:ff:ff:05:03:a2-01"
    },
    "10": {
        "config": {
            "battery": 100,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "1f4f1d8b4664cfd2cdce09ccb1c8a773",
        "lastseen": "2020-11-11T20:01Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.weather",
        "name": "Bedroom Sensor",
        "state": {
            "lastupdated": "2020-11-11T20:01:54.172",
            "temperature": 1972
        },
        "swversion": "20161129",
        "type": "ZHATemperature",
        "uniqueid": "00:15:8d:00:04:4a:47:c4-01-0402"
    },
    "11": {
        "config": {
            "battery": 100,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "1f4f1d8b4664cfd2cdce09ccb1c8a773",
        "lastseen": "2020-11-11T20:01Z",
[... 795 lines cut ...]
        "type": "ZHAHumidity",
        "uniqueid": "00:15:8d:00:04:4a:46:b4-01-0405"
    },
    "9": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "1739a60e8a9c96684491b9cf90bd62d4",
        "lastseen": "2020-11-11T20:22Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.weather",
        "name": "Living room Sensor",
        "state": {
            "lastupdated": "2020-11-11T20:22:05.306",
            "pressure": 1006
        },
        "swversion": "20161129",
        "type": "ZHAPressure",
        "uniqueid": "00:15:8d:00:04:4a:46:b4-01-0403"
    }
}
```

Smanar, 2020-11-11:
Ok so :

- I haven't find yet why you still have 1 light device ...
- The devices for switches are ID =  20 / 21 / 22/ 23 (look at their uniqueid, there is one by endpoint)
- The pressed button appear in "buttonevent".
- It seem from the code you will have 1 entry bu endpoint, I can group them if you prefer ?

tr1plus, 2020-11-11:
Hey, no light is appearing in phoscon client anymore - not sure why it's gone now (I keep switching between the VM and my main install, so maybe the stick is just seeing things). I do still see 4 lights with name TS0044 in the lights api call - not sure what they are about. It's possible they are also "old", I'm not fully sure how to verify that as I can't see them in deconz or phoscon (unless it's the 4 little lights  on the remote itself for indicating button press, but I doubt it).

I'm not sure what you mean with 1 entry by endpoint. I'm not too familiar with the inner workings of deconz/phoscon.

Let's take the ikea remotes as an example. I want to be able to have the device being known as 1 device in phoscon with 4 buttons, and if possible having each button have multiple functions (long press, single press, double press) as far as possible of course :) I hope that answers yours question?

Additional question - is there any reason why it's not showing up in phoscon? Will it work once this device gets included in the stable release or...?

Thanks! :)

```json
{
    "1": {
        "colorcapabilities": 16,
        "ctmax": 454,
        "ctmin": 250,
        "etag": "14b12afaa4275cda5fb7e10ae75231f6",
        "hascolor": true,
        "lastannounced": null,
        "lastseen": "2020-11-11T21:20Z",
        "manufacturername": "IKEA of Sweden",
        "modelid": "TRADFRI bulb E14 WS opal 400lm",
        "name": "Office Light",
        "state": {
            "alert": "none",
            "bri": 248,
            "colormode": "ct",
            "ct": 250,
            "on": false,
            "reachable": true
        },
        "swversion": "1.2.221",
        "type": "Color temperature light",
        "uniqueid": "00:0b:57:ff:fe:2f:f0:63-01"
    },
    "11": {
        "etag": "314ddc1881c39cc7afecb4ff90c0f9c2",
        "hascolor": false,
        "lastannounced": "2020-11-11T21:17:53Z",
        "lastseen": "2020-11-11T21:17Z",
        "manufacturername":"��U",
        "modelid": null,
        "name": "On/Off light 11",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-02"
    },
    "12": {
        "etag": "314ddc1881c39cc7afecb4ff90c0f9c2",
        "hascolor": false,
        "lastannounced": "2020-11-11T21:17:53Z",
        "lastseen": "2020-11-11T21:17Z",
        "manufacturername":"��U",
        "modelid": null,
        "name": "On/Off light 12",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-03"
    },
    "13": {
[... 634 lines cut ...]
    },
    "9": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "05e75bac6c1d23600229ba6731092789",
        "lastseen": "2020-11-11T21:15Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.weather",
        "name": "Living room Sensor",
        "state": {
            "lastupdated": "2020-11-11T20:22:05.306",
            "pressure": 1006
        },
        "swversion": "20161129",
        "type": "ZHAPressure",
        "uniqueid": "00:15:8d:00:04:4a:46:b4-01-0403"
    }
}
```

Thanks!

Smanar, 2020-11-12:
Ok, so I think it s because you have let the permit join.
Look at "light" device 11/ 12 / 13 / 14
>        "lastannounced": "2020-11-11T21:17:53Z",
>          "lastseen": "2020-11-11T21:19Z",
And incomplete data (strange deconz haven't tried to complete them .... )

But the sensor was created before
>         "lastseen": "2020-11-11T21:15Z"

So I have updated the code to prevent that (at least to try), I m retreiving the value from sensor, so if sensor is created, light will be blocked.

> Let's take the ikea remotes as an example. I want to be able to have the device being known as 1 device in phoscon with 4 buttons, and if possible having each button have multiple functions (long press, single press, double press) as far as possible of course :) I hope that answers yours question?

Short/ long / double are already supported
Have modified to code to trying to make only 1 device instead of 4.

>Additional question - is there any reason why it's not showing up in phoscon? Will it work once this device gets included in the stable release or...?

Nope, Phoscon is a different project, it s the phoscon dev that decide wich one switch will be visible or not.

tr1plus, 2020-11-13:
Sorry for the delay. I just updated the code based on your branch. This are the outputs of /lights and /sensors (I first deleted everything trough the api that was created earlier).

It seems a light is still created. Below also the deconz screenshot. Is there anyway I can verify via deconz if my button presses are registering? (Sorry I'm used to using phoscon)

Also- can we ask phoscon team to add the device or how is this usually done?

[image]

```json
{
    "1": {
        "colorcapabilities": 16,
        "ctmax": 454,
        "ctmin": 250,
        "etag": "735bf25a976ca8dbc83196798d92d003",
        "hascolor": true,
        "lastannounced": null,
        "lastseen": "2020-11-13T19:57Z",
        "manufacturername": "IKEA of Sweden",
        "modelid": "TRADFRI bulb E14 WS opal 400lm",
        "name": "Office Light",
        "state": {
            "alert": "none",
            "bri": 248,
            "colormode": "ct",
            "ct": 250,
            "on": true,
            "reachable": true
        },
        "swversion": "1.2.221",
        "type": "Color temperature light",
        "uniqueid": "00:0b:57:ff:fe:2f:f0:63-01"
    },
    "11": {
        "etag": "735bf25a976ca8dbc83196798d92d003",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-13T19:51Z",
        "manufacturername": "_TZ3000_vp6clf9d",
        "modelid": "TS0044",
        "name": "On/Off light 11",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01"
    },
    "2": {
        "colorcapabilities": 16,
        "ctmax": 454,
        "ctmin": 250,
        "etag": "92e545f8d087240f3ba3f37100a8d750",
        "hascolor": true,
        "lastannounced": "2020-10-13T20:33:27Z",
        "lastseen": "2020-11-13T19:57Z",
        "manufacturername": "IKEA of Sweden",
        "modelid": "TRADFRI bulb E27 WS opal 980lm",
        "name": "Living room Light Sofa",
        "state": {
            "alert": "none",
            "bri": 103,
            "colormode": "ct",
            "ct": 373,
            "on": true,
            "reachable": true
        },
        "swversion": "1.2.217",
[... 488 lines cut ...]
        "type": "ZHAHumidity",
        "uniqueid": "00:15:8d:00:04:4a:46:b4-01-0405"
    },
    "9": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "d0803524322851a2de96681b8dde3e1c",
        "lastseen": "2020-11-13T19:57Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.weather",
        "name": "Living room Sensor",
        "state": {
            "lastupdated": "2020-11-13T19:51:36.877",
            "pressure": 1004
        },
        "swversion": "20161129",
        "type": "ZHAPressure",
        "uniqueid": "00:15:8d:00:04:4a:46:b4-01-0403"
    }
}
```

Smanar, 2020-11-13:
>It seems a light is still created. Below also the deconz screenshot. Is there anyway I can verify via deconz if my button presses are registering? (Sorry I'm used to using phoscon)

Sure, look at field "buttonevent" in sensor 20. The first number is the button number and the second number is the action

>Also- can we ask phoscon team to add the device or how is this usually done?

Yep, It's possible too, there is a github for that too https://github.com/dresden-elektronik/phoscon-app-beta/issues
But he have lot of work ATM, and the device is not yet in deconz itslef ^^, so I think it s better to wait some time.

You have deleted API entries for your tests, but you have keep the deconz entry or not for your tests ? Because it s a long story but I m trying to get information from deconz database to reconize the device.
And I think if you delete the "light" entry it will delete too the "sensor" entry ?

tr1plus, 2020-11-13:
Hey. I deleted it via api and from the deconz screen itself before testing your latest code.

I tried validating button presses (I did it via API since I can't find the "buttonevent" in the deconz gui :( )
It mostly stays at 1002 - doesn't matter what buttons I press. In very rare cases it goes to 1004. So I don't think button handeling is working correctly (or the API Doesn't update frequently enough, not sure how that works)

When I (just now) deleted the light entry via api, the senor entry stayed behind.

Then I removed sensor via api and removed the node from deconz.

Then re-paired the switch (limiting to relevant entries for visibility)

```json
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "0f8e78898daf838b12e3c65e45cd073a",
    "lastseen": "2020-11-13T22:28Z",
    "manufacturername": "_TZ3000_vp6clf9d",
    "mode": 1,
    "modelid": "TS0044",
    "name": "Switch 21",
    "state": {
        "buttonevent": 1004,
        "lastupdated": "2020-11-13T22:28:04.699"
    },
    "type": "ZHASwitch",
    "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01-0006"
}
```

```json
{
    "etag": "b5fa90c83ec3b58356ddc2828e1953b0",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-11-13T22:27Z",
    "manufacturername": "_TZ3000_vp6clf9d",
    "modelid": "TS0044",
    "name": "On/Off light 10",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "On/Off light",
    "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01"
}
```

Smanar, 2020-11-14:
>When I (just now) deleted the light entry via api, the senor entry stayed behind.

Good news for later.
In fact, the deconz code need to be modified, ATM to reconize tuya device I m using the deconz database, so for the code works the device need to be in it. So you make a first inclusion, you will have bad entries, but the next one will work. But if you delete the deconz entry, you restart from scratch.

So the better way is to delete the "light" entry in the API, but never the device in deconz, to prevent "bad devices" creation.

For your button event return, I think you have only return for the first button, so I have updated the code again, if it still not working I will probably need special log when you press it.

The field need to be X00Y
- X = button Number
- Y = event

Events are
- 2 = short released
- 3 = long released
- 4 = double press

Smanar, 2020-11-14:
I m sorry but I need to finalize the code today if I want it be in next beta.
There is no one that can try the code ?
On "tuya" branch or "tuya_pr" (based on last code possible)

I just need to know if the "switches group" is working  (mean all buttons are working in the same entry), else I will make a roolback to have 1 sensor device by endpoint.

tr1plus, 2020-11-14:
Hey - I can only test in about 12 hours from now - was out for almost the whole day. I may try a bit earlier. Keep you posted

Do whatever needs to be done if it's too late - I can test after and see what is possible.

tr1plus, 2020-11-15:
FYI - I'll start testing now by pulling the tuya branch.

Keep you posted.

tr1plus, 2020-11-15:
Hey

I didn't delete anything this time around, just updated the code with your git pull & compile.

According to the "buttonevent" in the api the 3 press types (short, long, double) work for buttons 1 - 3. This device has 4 buttons so I'm not sure if you considered that or of the last button is just wonky 👍 

Apart from that the lights are still visible in the api of deconz. Let me know if I need more testing on that.

I will be available from time to time in the coming hours. I will keep an eye on this github/mail.

Smanar, 2020-11-15:
Thx, will take a look for button 4.
But you have 3 "sensor" entry, ot only one that show the 3 buttons ?

tr1plus, 2020-11-15:
Hey

In Deconz I see this (so 4 buttons I would assume)
[image]

In API /sensors I only see the 1 entry

```json
    "21": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "d06d1cd496471b839c8da74764582732",
        "lastseen": "2020-11-15T09:11Z",
        "manufacturername": "_TZ3000_vp6clf9d",
        "mode": 1,
        "modelid": "TS0044",
        "name": "Switch 21",
        "state": {
            "buttonevent": 1004,
            "lastupdated": "2020-11-15T09:11:12.646"
        },
        "type": "ZHASwitch",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01-0006"
    },
```

In /lights I see the 4 lights

```json
   "10": {
        "etag": "41ed218799c314771724001ea70cf91f",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-15T09:10Z",
        "manufacturername": "_TZ3000_vp6clf9d",
        "modelid": "TS0044",
        "name": "On/Off light 10",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01"
    },
    "12": {
        "etag": "41ed218799c314771724001ea70cf91f",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-15T09:10Z",
        "manufacturername":"�}�V",
        "modelid": null,
        "name": "On/Off light 12",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-02"
    },
    "13": {
        "etag": "41ed218799c314771724001ea70cf91f",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-15T09:10Z",
        "manufacturername":"�}�V",
        "modelid": null,
        "name": "On/Off light 13",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-03"
    },
    "14": {
        "etag": "41ed218799c314771724001ea70cf91f",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-15T09:10Z",
        "manufacturername":"�}�V",
        "modelid": null,
        "name": "On/Off light 14",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-04"
    },
```

hope this clarifies. I only get response in the "buttonevent" field in the `/sensors/5c:02:72:ff:fe:a4:ab:ad-01-0006` on button 1-2-3. 4 does't change anything for 'buttonevent' field.

Smanar, 2020-11-15:
Nice thx ^^. So you have all value, 3001 2002 and ect, in the sensor 21 ?
Unfornately, for the moment you need to delete yourself the "light" devices the first time, but one time they are stored, it will be ok.

If you want to try the 4 th button, edit the json file > https://github.com/Smanar/deconz-rest-plugin/commit/dbcf97a2fa51a6734c9d79bdf7cba774d44eef81

This file is in a special folder, on rapsbian it's 
`.local/share/dresden-elektronik/deCONZ/devices/`
or
`/usr/share/deCONZ/plugins/devices`

But don't worry if the 3 are working the 4 th will work too. Thx A lot, I will make the last PR now.

tr1plus, 2020-11-15:
Just to confirm - I should delete the 4 lights via api, an leave the rest (sensor & deconz gui node), correct?

Yes, some examples, all from the same sensor:

"buttonevent": 2003,
"buttonevent": 3002,
"buttonevent": 3003,
"buttonevent": 2002,
"buttonevent": 2004,

Finishing something up - will take a look at the file

Thanks for the help already :) Will this potentailly be in the next stable somewhere mid December?

Smanar, 2020-11-15:
Yep, you have said it don't delete the sensor entry, so yes you can.
The code look first in deconz database then in sensor database before creating a new "light" device, so they will not re appear after the deletion.

You will have them in the beta tommorow I think, so ofc in the future official.

tr1plus, 2020-11-15:
I just re-confirmed. I deleted the 4 light entries and the sensor entry is still there and responding to button presses (via the buttonevent field).

Thanks for the help. If I find some time later today I'll check the JSON file you mentioned for button 4.

tr1plus, 2020-11-15:
Hey

2 remarks:

In your commit that you linked you are removing 1 line from the button 3 actions (see red)

You are putting all this in "Tuya3gang". MY button is a 4 gang - not sure if this will be clear in the future why a 3 gang has 4 buttons ;)

Below JSON is tested and works with my 4Gang button

```json
        "Tuya3gangMap": {
            "modelids": [""],
            "map": [
                [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1 short"],
                [1, "0x01", "ONOFF", "0xfd", "1", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "B1 double"],
                [1, "0x01", "ONOFF", "0xfd", "2", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "B1 long"],
                [1, "0x02", "ONOFF", "0xfd", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2 short"],
                [1, "0x02", "ONOFF", "0xfd", "1", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "B2 double"],
                [1, "0x02", "ONOFF", "0xfd", "2", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "B2 long"],
                [1, "0x03", "ONOFF", "0xfd", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3 short"],
                [1, "0x03", "ONOFF", "0xfd", "1", "S_BUTTON_3", "S_BUTTON_ACTION_DOUBLE_PRESS", "B3 double"],
                [1, "0x03", "ONOFF", "0xfd", "2", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "B3 long"],
                [1, "0x04", "ONOFF", "0xfd", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4 short"],
                [1, "0x04", "ONOFF", "0xfd", "1", "S_BUTTON_4", "S_BUTTON_ACTION_DOUBLE_PRESS", "B4 double"],
                [1, "0x04", "ONOFF", "0xfd", "2", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "B4 long"]
            ]
        },
```

tr1plus, 2020-11-15:
Another remark.

I don't see the switch in the deconz UI anymore. This happened after I restarted deconz to validate the json change.
Is it possible that it is no longer there because the lights (the once I removed) were not available on start up?

IT's still availalbe in the API

[image]

```json
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "af43562570713f63e786d0f257b72228",
    "lastseen": "2020-11-15T13:04Z",
    "manufacturername": "_TZ3000_vp6clf9d",
    "mode": 1,
    "modelid": "TS0044",
    "name": "Switch 21",
    "state": {
        "buttonevent": 4002,
        "lastupdated": "2020-11-15T13:04:00.296"
    },
    "type": "ZHASwitch",
    "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01-0006"
}
```

Smanar, 2020-11-15:
Yep, it s a generic code for tuya, it work for 1 gang and 2 gand and 3 and 4.

>I don't see the switch in the deconz UI anymore

Ha ? not good. But if it still working, it is obligatorily in deconz, a device can be in deconz and not in the api, but not the reverse.

tr1plus, 2020-11-15:
I had previously deleted the 4 lights via API.

Then I restarted the whole machine after making changes to the json file to ensure it had the latest version.

After that I could still see the switch in the api and it was responding to button presses, but as you can see It is not visible in deconz ui (switch 21)

Smanar, 2020-11-15:
Can you take a look in the "node list" in deconz with the mac adress ?
The name can be different.

tr1plus, 2020-11-15:
Can't find it. Device is still responding to button presses

```json
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "b2fd070cd041eb497096cee07f35fd85",
    "lastseen": "2020-11-15T14:57Z",
    "manufacturername": "_TZ3000_vp6clf9d",
    "mode": 1,
    "modelid": "TS0044",
    "name": "Switch 21",
    "state": {
        "buttonevent": 2002,
        "lastupdated": "2020-11-15T14:57:39.517"
    },
    "type": "ZHASwitch",
    "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01-0006"
}
```

[image]
[image]

Smanar, 2020-11-15:
Ok its something definitively strange ^^
I have asked, it seem it s not possible, you have a magic network :)

Something to explore, but if I m right, if you have problem, now you have the manufacture number somewhere in the database, so if you don't delete the sensor device, you can make a new inclusion to correct this situation, the "light" device will be blocked, and the device will be added in deconz.

tr1plus, 2020-11-15:
Hey - I've had that before that weird things happen in my set-up ;)

I paired the switch again. AT frist the node in DeConz was called on/off light
A light entry did appear in the API
Then I restarted the VM.

The sensor has now this name
[image]

The Sensor entry is still available in the API (and is still working on button presses)

```json
    "11": {
        "etag": "20a7e20395888474a5c8815a5bd21739",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-15T17:20Z",
        "manufacturername": "_TZ3000_vp6clf9d",
        "modelid": "TS0044",
        "name": "On/Off light 11",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01"
    },
```

```json
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "5090d064f010163c71d08f9b873f37eb",
    "lastseen": "2020-11-15T17:26Z",
    "manufacturername": "_TZ3000_vp6clf9d",
    "mode": 1,
    "modelid": "TS0044",
    "name": "TS0044 21",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2020-11-15T17:21:01.638"
    },
    "type": "ZHASwitch",
    "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01-0006"
}
```

tr1plus, 2020-11-15:
NOTE: I am often switching back to another VM that doesn't have the latest deconz changes on it (as those control my lights across the house). Maybe this is interfering with things and breaking the pairing of this device?

Smanar, 2020-11-15:
No I don't think, if it s another VM I think there is another dabase.

BTW I have 1 month to solve this problem ^^, So ATM, on second inclusion, tuya device still create a "light" device, but with correct information.
And if we delete it the device disapear from deconz, but the device still working

And it s strange but on your capture the device seem "grayed" ?

tr1plus, 2020-11-15:
A lot of my devices are that way - as you can see the green light flashes on from time to time when I click a button

[image]

If you need it - I would be able to try/do some things over discord or something. Just let me know whatever you need.

Smanar, 2020-11-16:
Sure I will need some help soon.
Manup have already add in the code a "better device detection" that can avoid the "tuya hack", not enought time ATM, but will be my monthly objective ^^

tr1plus, 2020-11-16:
Sounds good - let me know if/when I need to test things :)

VoyteckPL, 2020-11-20:
Hi, is it working already? How to add support to this switch? Thanks!

Smanar, 2020-11-20:
For the moment you need to use the last deconz version (Beta 2.6.0)

VoyteckPL, 2020-11-20:
How do I install it in HA?

Smanar, 2020-11-20:
if you are using container, you need to wait an update from the container.
If you have a normal OS, just download the file and install it.

VoyteckPL, 2020-11-20:
I use Hassos. Is it possible to install beta?

tr1plus, 2020-11-30:
Hey

I think I just installed the beta version and tried testing the below (assuming this is the correct version).
[image]

I'm still getting 4 light entries when adding the switch

```json
    "10": {
        "etag": "0a39125225624e337eb4b28ef7f5a2ba",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-30T10:41Z",
        "manufacturername": "_TZ3000_vp6clf9d",
        "modelid": "TS0044",
        "name": "On/Off light 10",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01"
    },
    "15": {
        "etag": "9ec72e801859475270bf20796d80e387",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-30T10:41Z",
        "manufacturername":"P��ZU",
        "modelid": null,
        "name": "On/Off light 15",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-02"
    },
    "16": {
        "etag": "9ec72e801859475270bf20796d80e387",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-30T10:41Z",
        "manufacturername":"P��ZU",
        "modelid": null,
        "name": "On/Off light 16",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-03"
    },
    "17": {
        "etag": "9ec72e801859475270bf20796d80e387",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-30T10:41Z",
        "manufacturername":"P��ZU",
        "modelid": null,
        "name": "On/Off light 17",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-04"
    },
```

The sensor entry is still there and looks to be responding well to button event changes

```json
    "20": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "0a39125225624e337eb4b28ef7f5a2ba",
        "lastseen": "2020-11-30T10:42Z",
        "manufacturername": "_TZ3000_vp6clf9d",
        "mode": 1,
        "modelid": "TS0044",
        "name": "Switch 20",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "type": "ZHASwitch",
        "uniqueid": "5c:02:72:ff:fe:a4:ab:ad-01-0006"
    },
```
"buttonevent": 4002,
"buttonevent": 4003,
"buttonevent": 3004,
ect.

Let me know if I need to test some more. Is downloading the beta version via the apt beta repository sufficient?

Thanks!

turbouk2312, 2020-11-30:
Hi, I have the same issue with the same 4 way switch. My install is pretty much brand new so I'm happy to help with some messing around as I don't have much to lose if I need to wipe everything

Kusi-Schweiz, 2020-12-01:
Funktioniert leider nicht einwandfrei. Wird eine Taste gedrückt, dann wird der Buttonevent 2x hintereinander auf true gesetzt.
Dies spielt bei mir eine Rolle, da ich mit einer Taste die Funktion ein/aus realisiere mit einer Abfrage.

Mimiix, 2020-12-01:
@Kusi-Schweiz Please stick with english so everyone can participate in the discussion 😄

Kusi-Schweiz, 2020-12-01:
When i push a button, in ioBroker de State is two times true.

Smanar, 2020-12-01:
@Kusi-Schweiz it s the same issue than here ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3693
He have solved the issue with a new inclusion.

You have all events in double ?

Kusi-Schweiz, 2020-12-01:
@Smanar Yes, but only one second. I push the button and one second later, the state is again true.

[image]

[image]

Smanar, 2020-12-02:
Can you find the log during the double event ?

If you have the GUI, they are easy to find.
Else on your third app, but idk wich one you are using
Or if you can run deconz in debug mode for test ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

Only dbg-error=2 and dbg-info = 2

stevenwfoley, 2020-12-02:
Below is what I get in the deconz gui debug view for INFO and INFO_2. ERROR and ERROR_2 produced nothing.
17:44:38:494 [INFO] - Button 1002 B1 short TS0044
17:44:38:496 enqueue event state/buttonevent for /sensors/29
17:44:38:497 enqueue event state/lastupdated for /sensors/29
17:44:38:550 discard sensor state push for 29: state/lastupdated (already pushed)
17:44:39:502 [INFO] - Button 1002 B1 short TS0044

Smanar, 2020-12-03:
No lines like this one just before ?

```
18:37:58:340 Node data 0x00158d00027cf2b7 profileId: 0x0104, clusterId: 0x0012
18:37:58:345 0x00158D00027CF2B7: update ZCL value 0x02/0x0012/0x0055 after 0 s
18:37:58:350 	payload: 5500210200
```

stevenwfoley, 2020-12-03:
This looks more complete, and this is just for button 1 (lower left).

11:48:31:479 Node data 0xbc33acfffe28ba91 profileId: 0x0104, clusterId: 0x0001
11:48:31:482 0xBC33ACFFFE28BA91: update ZCL value 0x01/0x0001/0x0021 after 0 s
11:48:31:484 enqueue event config/battery for /sensors/29
11:48:31:493 [INFO] - No button handler for: TS0044 endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021
11:48:31:495 ZCL attribute report 0xBC33ACFFFE28BA91 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
11:48:31:498 	payload: 210020c8
11:48:31:801 [INFO] - Button 1002 B1 short TS0044
11:48:31:804 enqueue event state/buttonevent for /sensors/29
11:48:31:807 enqueue event state/lastupdated for /sensors/29
1:48:31:877 discard sensor state push for 29: state/lastupdated (already pushed)

Smanar, 2020-12-03:
I need more ^^, sorry.
When you press the button you have 2 events, so I would like the log during the double event, with the command send 2 time, to check if the problem is from deconz, the API or the device.

stevenwfoley, 2020-12-03:
This is a production zigbee environment, so my log is a bit noisy, but this is EVERYTHING from info and info_l2 recorded a second or so before the button press, and a few seconds after. The device id is 29, and the unique id is 0xBC33ACFFFE28BA91.

13:42:14:808 Current channel 25
13:42:14:832 Device TTL 3660 s flags: 0x7
13:42:15:355 read attributes of 0x00178801025C32D9 cluster: 0x0006: [ 
13:42:15:359 0x0000 
13:42:15:361 ]
13:42:15:363 add task 1181938 type 19 to 0x00178801025C32D9 cluster 0x0006 req.id 198
13:42:15:365 Poll APS request 198 to 0x00178801025C32D9 cluster: 0x0006
13:42:15:490 Poll APS confirm 198 status: 0x00
13:42:15:493 Erase task req-id: 198, type: 19 zcl seqno: 105 send time 0, profileId: 0x0104, clusterId: 0x0006
13:42:15:517 Node data 0x00178801025c32d9 profileId: 0x0104, clusterId: 0x0006
13:42:15:521 enqueue event attr/lastseen for /lights/6
13:42:15:525 0x00178801025C32D9: update ZCL value 0x0B/0x0006/0x0000 after 0 s
13:42:15:528 enqueue event event/checkgroupanyon for /groups/6996
13:42:15:532 enqueue event event/checkgroupanyon for /groups/7
13:42:15:534 enqueue event event/checkgroupanyon for /groups/2583
13:42:15:536 enqueue event event/checkgroupanyon for /groups/65520
13:42:15:581 Websocket 10.0.0.34:52828 send message: {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":153,"id":"6","lastannounced":"2020-10-10T20:46:31Z","lastseen":"2020-12-03T18:42Z","manufacturername":"Philips","modelid":"LTW004","name":"Living Room Lamp 2","swversion":"5.130.1.30000","type":"Color temperature light","uniqueid":"00:17:88:01:02:5c:32:d9-0b"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"00:17:88:01:02:5c:32:d9-0b"} (ret = 401)
13:42:15:586 Websocket 10.0.0.10:49992 send message: {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":153,"id":"6","lastannounced":"2020-10-10T20:46:31Z","lastseen":"2020-12-03T18:42Z","manufacturername":"Philips","modelid":"LTW004","name":"Living Room Lamp 2","swversion":"5.130.1.30000","type":"Color temperature light","uniqueid":"00:17:88:01:02:5c:32:d9-0b"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"00:17:88:01:02:5c:32:d9-0b"} (ret = 401)
13:42:15:729 read attributes of 0x00178801025C32D9 cluster: 0x0008: [ 
13:42:15:734 0x0000 
13:42:15:737 ]
13:42:15:740 add task 1181940 type 19 to 0x00178801025C32D9 cluster 0x0008 req.id 203
13:42:15:742 Poll APS request 203 to 0x00178801025C32D9 cluster: 0x0008
13:42:15:783 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
13:42:15:845 Poll APS confirm 203 status: 0x00
13:42:15:849 Erase task req-id: 203, type: 19 zcl seqno: 106 send time 0, profileId: 0x0104, clusterId: 0x0008
13:42:15:873 read attributes of 0x00178801025C32D9 cluster: 0x0300: [ 
13:42:15:878 0x0007 
13:42:15:881 0x0008 
13:42:15:883 0x4001 
13:42:15:885 ]
13:42:15:887 add task 1181942 type 19 to 0x00178801025C32D9 cluster 0x0300 req.id 205
13:42:15:890 Poll APS request 205 to 0x00178801025C32D9 cluster: 0x0300
13:42:15:896 Node data 0x00178801025c32d9 profileId: 0x0104, clusterId: 0x0008
13:42:15:908 0x00178801025C32D9: update ZCL value 0x0B/0x0008/0x0000 after 0 s
13:42:15:997 Node data 0x84182600000a1515 profileId: 0x0104, clusterId: 0x0006
13:42:16:015 enqueue event attr/lastseen for /lights/24
13:42:16:019 0x84182600000A1515: update ZCL value 0x03/0x0006/0x0000 after 0 s
13:42:16:022 enqueue event event/checkgroupanyon for /groups/65520
13:42:16:025 enqueue event event/checkgroupanyon for /groups/12
13:42:16:028 enqueue event event/checkgroupanyon for /groups/22137
13:42:16:031 ZCL attribute report 0x84182600000A1515 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
13:42:16:033 	payload: 00001000
13:42:16:082 Websocket 10.0.0.34:52828 send message: {"attr":{"colorcapabilities":24,"ctmax":370,"ctmin":153,"id":"24","lastannounced":null,"lastseen":"2020-12-03T18:42Z","manufacturername":"OSRAM","modelid":"LIGHTIFY A19 Tunable White","name":"Basement 4","swversion":"0x01020492","type":"Color temperature light","uniqueid":"84:18:26:00:00:0a:15:15-03"},"e":"changed","id":"24","r":"lights","t":"event","uniqueid":"84:18:26:00:00:0a:15:15-03"} (ret = 392)
13:42:16:087 Websocket 10.0.0.10:49992 send message: {"attr":{"colorcapabilities":24,"ctmax":370,"ctmin":153,"id":"24","lastannounced":null,"lastseen":"2020-12-03T18:42Z","manufacturername":"OSRAM","modelid":"LIGHTIFY A19 Tunable White","name":"Basement 4","swversion":"0x01020492","type":"Color temperature light","uniqueid":"84:18:26:00:00:0a:15:15-03"},"e":"changed","id":"24","r":"lights","t":"event","uniqueid":"84:18:26:00:00:0a:15:15-03"} (ret = 392)
13:42:16:092 Poll APS confirm 205 status: 0x00
13:42:16:097 Erase task req-id: 205, type: 19 zcl seqno: 107 send time 0, profileId: 0x0104, clusterId: 0x0300
13:42:16:143 Node data 0x00178801025c32d9 profileId: 0x0104, clusterId: 0x0300
13:42:16:148 0x00178801025C32D9: update ZCL value 0x0B/0x0300/0x0007 after 0 s
13:42:16:154 0x00178801025C32D9: update ZCL value 0x0B/0x0300/0x0008 after 0 s
13:42:16:157 0x00178801025C32D9: update ZCL value 0x0B/0x0300/0x4001 after 0 s
13:42:16:237 schedule 4 diff 33464, Sun Nov 29 23:00:00 2020
13:42:16:242 schedule 6 diff 11864, Mon Nov 30 17:00:00 2020
13:42:16:245 schedule 7 diff 8264, Mon Nov 30 16:00:00 2020
13:42:16:248 enqueue event config/localtime for /config/
13:42:16:250 poll node 00:17:88:01:02:5f:35:51-0b
13:42:16:252 Poll light node Basement Landing
13:42:16:255 Idle timer triggered
[... cut ...]
13:42:19:237 schedule 4 diff 33461, Sun Nov 29 23:00:00 2020
13:42:19:240 schedule 6 diff 11861, Mon Nov 30 17:00:00 2020
13:42:19:242 schedule 7 diff 8261, Mon Nov 30 16:00:00 2020
13:42:19:245 enqueue event config/localtime for /config/
13:42:19:270 rule event /config/localtime: 13:42:18.270 -> 13:42:19.245 (975ms)
13:42:19:313 poll node 7c:b0:3e:aa:00:a4:92:4a-03
13:42:19:317 Poll light node Plug 2
13:42:19:366 Poll APS request to 0x7CB03EAA00A4924A cluster: 0x0006 dropped, values are fresh enough

Smanar, 2020-12-04:
IDK, I realy don't see where the event is coming ...
For battery there is information
```
13:42:16:894 Node data 0xbc33acfffe28ba91 profileId: 0x0104, clusterId: 0x0001
13:42:16:897 0xBC33ACFFFE28BA91: update ZCL value 0x01/0x0001/0x0021 after 0 s
13:42:16:900 enqueue event config/battery for /sensors/29
13:42:16:908 [INFO] - No button handler for: TS0044 endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021
13:42:16:910 ZCL attribute report 0xBC33ACFFFE28BA91 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
13:42:16:913 payload: 210020c8
```

But for switches, absolutely nothing, just 
>13:42:17:219 [INFO] - Button 1002 B1 short TS0044

stevenwfoley, 2020-12-04:
Would a wireshark sniff be helpful?

Smanar, 2020-12-04:
Ha yes ! if you can yes, will be better than deconz log.

stevenwfoley, 2020-12-04:
How would I get this to you? The file contains my zigbee security key so I don't want it public.

[127 more replies]

closed 2021-06-29 by github-actions[bot]