dresden-elektronik/deconz-rest-plugin #3612: [Device Support Request] Lidl 3 way SilverCrest Smart USB Extension Lead TY-HG06338-DK (Tuya TS011F)

ACrazyConcept, 2020-11-08:
# Device

- Product name: Smart USB Extension Lead / Smart Connect Strip
- Manufacturer: In deCONZ: **_TZ3000_wzauvbcs** and **Heiman**, on the device: SilverCrest
- Model identifier: In deCONZ: **TS011F**, on the device: SPSZ 3 A1 and TY-HG06338-DK
- Device type :
  - Smart Plug / Power Strip

## Screenshots
[image]
[image]
[image]
[image]

This is one of a handful of devices just launched by Lidl in several countries I assume. The power strip comes with different regional socket types. The one I got is the DK version.
It is not that it is not working. I get 3 x On/Off plug-in unit as you can see, and they work fine and also work in home-assistant etc.
But it also has 4 USB power outlets. And I have so far not been able to conclude whether or not they have any zigbee representation hiding somewhere.
It looks like other smart plug Tuya modules do, but that's possibly not an indicator of anything: [Fixing ZigBee3.0 dual USB wireless socket plug #4503](https://github.com/Koenkk/zigbee2mqtt/issues/4503)

A couple of other references I could find online:
[Lidl Smart Home](https://www.lidl.dk/smart-home)
[TuYa TS011F via Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/TS011F.html)
[Smart Connect Strip on zigbeealliance.org](https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Smart+Connect+Strip)

Mimiix, 2020-11-08:
Hi,

Please include screenshots of all clusters.

ZeppDK, 2020-11-09:
have you seen per mejdals video? https://youtu.be/R1cLFzcDQNU
it seems to be only "normal" usb outlets...

how did you get deconz to add the normal outlets? im not having any luck adding them...

ZeppDK, 2020-11-09:
never mind... they turned up under lights for some reason... :-D

ACrazyConcept, 2020-11-09:
> 
> 
> have you seen per mejdals video? https://youtu.be/R1cLFzcDQNU
> it seems to be only "normal" usb outlets...
> 
> how did you get deconz to add the normal outlets? im not having any luck adding them...

Well that's dissapointing.

closed 2020-11-09 by ACrazyConcept

LeoeLeoeL, 2021-02-21:
Pay attention to LIDL smartplugs; The "PowerOn  OnOff" is configurated on PREVIOUS. In case of Blackout, if the switch was ON it will be ON.
All the other smartplugs, after a power loss become OFF.