dresden-elektronik/deconz-rest-plugin #3618: Danfoss Ally - Enable external temperature sensor

simoneras, 2020-11-09:
## Describe the question or issue you are having
Hello everyone!
I would like to enable the external sensor to control the valve.
In the / config section I don't find how to do with REST-API.
It can be done?
Any idea?

Thanks

google translate

## Screenshots

Environment
Host System: Raspberry 3
Running Method: Rasbian Pi OS
Device: ConBee II
Deconz Version: 2.05.84 / 14/9/2020
Firmware: 264A0700

## deCONZ Logs

## Additional context

SwoopX, 2020-11-09:
This cannot be done via REST API. You may try that via deconz GUI, but I doubt it will work at all. For any further questions in this regard, please reach out to Danfoss support

Mimiix, 2020-11-09:
@SwoopX Case closed?

SwoopX, 2020-11-09:
@Mimiix Nothing we will or can do here from my perspective.

Mimiix, 2020-11-09:
#case closed

closed 2020-11-09 by Mimiix

dowhiletrue, 2020-12-12:
I wonder if `0x4015 External Measured Room Sensor` might be set as writable attribute in `general.xml` since [Danfoss trv.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/5493076/Danfoss.trv.xlsx) states the attribute to be writable. I changed this for testing purposes in and wrote a value. The TRV does not honor the value. I'll contact Danfoss about this issue.