dresden-elektronik/deconz-rest-plugin #3620: Various motion sensors stopped sending status "motion detected"

vahaldor, 2020-11-09:
## Describe the bug

## Steps to reproduce the behavior

For me, even fresh reinstall does not change anything. I cannot reproduce this as it does not work completely.

## Expected behavior

Motion detected.

## Screenshots

## Environment

 - Host system: Raspberry Pi 4 4GB
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: (6.4.1)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs

These commands does not work in hassio SSH terminal.

## Additional context

petr-vytlacil, 2020-11-10:
I have same problem.

Mimiix, 2020-11-10:
Can you provide logs?

vahaldor, 2020-11-11:
Can you tell me how can I export them, please?

Smanar, 2020-11-11:
>deCONZ version: (6.4.1)

Can you check again your version in phoscon/gateway pls ?

vahaldor, 2020-11-11:
I updated the firmware.
If you mean the update of deconz I probably installed newest version after the fresh install of home assistant.

[image]

petr-vytlacil, 2020-11-11:
My versions, and I have same problem.
[image]

vahaldor, 2020-11-11:
Interesting, why the system does not propose an update to me?

Smanar, 2020-11-11:
One of you have the gui to check something ?

petr-vytlacil, 2020-11-11:
[image]

Smanar, 2020-11-11:
You are not on the good cluster.
- Click on the second circle on the node title, it will expand cluster list
- Click on cluster 0x0406, it will expend cluster tab
- The attribute table is on the bottom tab, but with bad values.
- Double clic on attribute 0x0000, it will show a window
- Press "Read" with making the device awake

And you will have a value or an error message.

vahaldor, 2020-11-11:
I dont even have the motion sensors visible in the deconz map. The multisensor with four sensors dont have 0x0406.

petr-vytlacil, 2020-11-11:
[image]

vahaldor, 2020-11-11:
I also miss the motion measurement which is 0406? Dont know.
[image]
[image]

vahaldor, 2020-11-11:
Xiaomi Motion Sensor detects motion after several resets, for now, I can report how quickly it will stop.

I would love to provide some logs but I need guidance how to do that. Thanks.

petr-vytlacil, 2020-11-11:
Plese primarily fix the issue with IMMAX Multisensor 4in1 worked which dont detect motion. Thank you .-)

Smanar, 2020-11-11:
Was for the Xiaomi, the Immax use cluster 0x0500, and not the same working mode.
And yes there is some logs for IAS part
>DBG_Printf(DBG_ZCL, "IAS Zone Status Change, status: 0x%04X, zoneId: %u, delay: %u\n", zoneStatus, zoneId, delay);

To enable debug https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

vahaldor, 2020-11-11:
Where do we use these commands? They dont work in HA SSH terminal

petr-vytlacil, 2020-11-11:
> Was for the Xiaomi, the Immax use cluster 0x0500, and not the same working mode.
> And yes there is some logs for IAS part
> 
> > DBG_Printf(DBG_ZCL, "IAS Zone Status Change, status: 0x%04X, zoneId: %u, delay: %u\n", zoneStatus, zoneId, delay);
> 
> To enable debug https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags
> 
> The usefull ones are `--dbg-info=2 --dbg-error=2` you can set all other to 0

Is any way how enable debug on Windows OS? THX

vahaldor, 2020-11-11:
Btw I opened a thread with a bug report related to a particular installation. This is offtopic and destroys this thread.

Smanar, 2020-11-11:
Not sure this bug was specific to an installation, if you have same ?

vahaldor, 2020-11-11:
He has Windows 10. So I have to install raspberry OS in order to produce logs?

SwoopX, 2020-11-11:
Reading the attribute configuration requires the device to be awake, so you need to push the physical button 1-2 seconds before pressing the read button.

vahaldor, 2020-11-11:
Are you talking about the IMMAX multisensor we both with Petr Vystrcil have? There is only one button

Smanar, 2020-11-11:
You can't use command line on Hassio ? There is no way to obtain log using HA ?
For windows it s worse, I know a method, but never tried it, and idk if it work ...

vahaldor, 2020-11-11:
The terminal in HA is a simple one. No /usr/ folder there, so the above mentioned way is not possible. They expect to run raspbian. There is a log though in deconz addon but I cannot alter the output:

20:06:04:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:09:539 [INFO] - No button map for: lumi.sensor_motion endpoint: 0x01 cluster: 0x0406 command: 0x0A payload[0]: 000
20:06:09:539 ZCL attribute report 0x00158D0003CF5132 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:06:09:893 [INFO] - No button map for: lumi.sensor_motion endpoint: 0x01 cluster: 0x0406 command: 0x0A payload[0]: 000
20:06:09:893 ZCL attribute report 0x00158D0003D11CF6 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:06:10:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:12:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:12:604 [INFO] - No button map for: 4in1-Sensor-ZB3.0 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
20:06:12:604 ZCL attribute report 0x000D6FFFFE40055E for cluster: 0x0400, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:15:943 [INFO] - No button map for: lumi.sensor_motion endpoint: 0x01 cluster: 0x0406 command: 0x0A payload[0]: 000
20:06:15:943 ZCL attribute report 0x00158D0003CE0A2C for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:06:18:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:20:033 [INFO] - No button map for: 4in1-Sensor-ZB3.0 endpoint: 0x01 cluster: 0x0402 command: 0x0A payload[0]: 000
20:06:20:033 ZCL attribute report 0x000D6FFFFE40055E for cluster: 0x0402, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:20:500 [INFO] - No button map for: 4in1-Sensor-ZB3.0 endpoint: 0x01 cluster: 0x0405 command: 0x0A payload[0]: 000
20:06:20:500 ZCL attribute report 0x000D6FFFFE40055E for cluster: 0x0405, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:25:220 ZCL attribute report 0xCCCCCCFFFE3CD09E for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:25:246 ZCL attribute report 0xCCCCCCFFFE3CD09E for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:28:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:28:493 [INFO] - No button map for: 4in1-Sensor-ZB3.0 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
20:06:28:493 ZCL attribute report 0x000D6FFFFE40055E for cluster: 0x0400, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:30:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:33:529 [INFO] - No button map for: 4in1-Sensor-ZB3.0 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
20:06:33:529 ZCL attribute report 0x000D6FFFFE40055E for cluster: 0x0400, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:37:223 [INFO] - Button 1002 On
20:06:37:232 rule event /sensors/17/state/lastupdated: 0 -> 0
20:06:37:232 trigger rule 46 - Rule PUSHDIMUP_ON
20:06:37:322 ZCL attribute report 0x000B57FFFED6B06C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:38:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:38:730 [INFO] - Button 2001 Move down
20:06:38:739 rule event /sensors/17/state/lastupdated: 0 -> 0
20:06:38:739 trigger rule 51 - Rule PUSHDIMDOWN_START
20:06:40:330 [INFO] - Button 2003 Stop
20:06:40:337 rule event /sensors/17/state/lastupdated: 0 -> 0
20:06:40:337 trigger rule 50 - Rule PUSHDIMDOWN_STOP
20:06:40:434 ZCL attribute report 0x000B57FFFED6B06C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:06:43:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:51:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:06:53:355 skip binding for attribute reporting of ep: 0x01 cluster 0x0001 (end-device might sleep)
20:06:56:870 Current channel 20
20:06:56:878 Device TTL 6899 s flags: 0x7
20:06:59:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:07:01:355 skip binding for attribute reporting of ep: 0x01 cluster 0x0001 (end-device might sleep)
20:07:02:928 dev /dev/ttyAMA0
20:07:02:929 GW firmware version: 0x26660700
20:07:02:929 GW firmware version is up to date: 0x26660700
20:07:04:780 [INFO] - No button map for: lumi.sensor_motion endpoint: 0x01 cluster: 0x0406 command: 0x0A payload[0]: 000
20:07:04:781 ZCL attribute report 0x00158D0003D11CF6 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:07:06:148 [INFO] - No button map for: lumi.sensor_motion endpoint: 0x01 cluster: 0x0406 command: 0x0A payload[0]: 000
20:07:06:148 ZCL attribute report 0x00158D0003CF5132 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:07:07:354 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:07:08:268 ZCL attribute report 0xEC1BBDFFFEF433B4 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:07:09:348 [INFO] - No button map for: 4in1-Sensor-ZB3.0 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
20:07:09:348 ZCL attribute report 0x000D6FFFFE40055E for cluster: 0x0400, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
[... cut ...]
20:07:46:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:07:52:086 [INFO] - No button map for: lumi.sensor_motion endpoint: 0x01 cluster: 0x0406 command: 0x0A payload[0]: 000
20:07:52:086 ZCL attribute report 0x00158D0003D11CF6 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:07:54:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:07:56:355 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
20:07:56:869 Current channel 20

SwoopX, 2020-11-11:
You got tons of motion reports in there. Not from the Immax (I'm surprised that it worked anyway), but from the Xiaomi's...

vahaldor, 2020-11-11:
I see that. I did reset all three xiaomi and they seem to work for now but as I said previously they lost motions in a couple of days. I did fresh install of HA about three times and it happened everytime after several days.

Immax does not work but also worked at the beginning when I linked it for the first time.

SwoopX, 2020-11-11:
Regarding the Immax: lucky you. Got one donated by a user and only got it to work with some unknown black magic. That thing simply seem to refuse to send the required IAS reports and there's nothing in the sniffer logs. I'd rather suspect a firmware issue and some feedback I read in some issue here just recently went into that direction as well.

vahaldor, 2020-11-19:
I reset the Conbee, deleted all entities, deleted integration and addon, restarted HA, set up all zigbee devices from scratch and found that one bulb was behaving unpredictably. So I did not add it to the network and the IMMAX motion sensor begun working. Right now, the luminance sensor seems to return different values. However, motion has been detected several times since then... I would love to uncover this behavior deeper but somebody has to lead me.

[image]

closed 2020-12-25 by stale[bot]