dresden-elektronik/deconz-rest-plugin #3621: IKEA TRADFRI / No color change

MrChiliCheese, 2020-11-09:
## Describe the question or issue you are having
Tonight some of my lights stopped responding to commands from Alexa. I connected to Phoscon and saw that 3 lights and 1 room disappeared from the configuration. Before re-adding them I updated the firmware to 26660700 and the version to 
2.05.86 / 15.10.2020.
I added one light and had Alexa search for devices. The light is detected and I can turn it on/off and set the brightness. However it is not possible to set the color using Alexs, this only works from Phoscon itself. 

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.86
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? No other device is connected 

## deCONZ Logs

## Additional context

Mimiix, 2020-11-10:
Seems like you have the same as this:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3387

Also, i recommend using a extension cable.

MrChiliCheese, 2020-11-10:
Does not seem be the same as the lights are responsive in the UI (Brightness/Color/Saturation), but with Alexa only the Brightness seems to work. Any additional data I can provide?

MrChiliCheese, 2020-11-10:
I have tried different older version (2.05.69/02.05.75/02.05.77) and I am currently on 02.05.84 (=latest stable).

1. Brightness/Color/Saturation can be controlled by UI and the light reacts in real life.
2. If Brightness is changed by Alexa the UI shows the change and the light reacts in real life.
3. If Brightness/Color/Saturation are changed by Alexa the UI show the change, but the light does not react.

acid115, 2020-11-10:
I have solved this issue 9 months ago by fix the code in rest-lights.cpp. Now i have updated to the newest version, and noone fixed this issue.... Ok. 
I tried to merge my changes from februray into the current rest-lights. But no chance. Someone changed the code so, i have to analyse it where the issue is. The part where the hue value is calculated wrong is commented out....dont know how to fix it at the moment :/ i also have a tradfri color bulb 600lm

Mimiix, 2020-11-11:
@acid115 You might want to post here if you want. Maybe @SwoopX Can help you out on integrating?

SwoopX, 2020-11-11:
When I got to know what the change should be, I can have a look.

MrChiliCheese, 2020-11-11:
What is the difference betwee the command set is send from Alexa (not working) and from the UI? Are the signals send in one command by ALexa or is it the transition time?

acid115, 2020-11-11:
9 month ago the issue was that the hue value for color change was completly wrong calculated.
We have to debug it.
I think the gateway and rest api receives the command from alexa. And the command is ok, i think. We have to find out how the json from alexa looks like and debug what way the values go through the code of rest-lights

acid115, 2020-11-15:
Ok, now it works for me.
I found the code we have to change.
Maybe we have to do this only for ikea bulbs controlled by echo.
Short: echo sends hue and sat, but ikea bulb only unterstands x and y .
So it is necessary to calculate x and y from given hue and sat. And then set x and y like the echo should sent x and y.

More details tomorrow

MrChiliCheese, 2020-11-16:
Sounds like the change from here: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/a4c712bf6d4cc60b376968b9f6d292b2b916755f

acid115, 2020-11-16:
maybe, i dont unterstand a4c712b :)

So here is my change:
Line 1208 - 1249 (was commented out, but was nearly correct :)
 addTaskSetXyColor(task, x, y); was not in.
additionally i fixed the yellow value.
can someone merge it?
`
diff --git a/rest_lights.cpp b/rest_lights.cpp
index 1a855d2..64314a9 100644
--- a/rest_lights.cpp
+++ b/rest_lights.cpp
@@ -1202,47 +1209,67 @@ int DeRestPluginPrivate::setLightState(const ApiRequest &req, ApiResponse &rsp)
             ok = addTaskSetHueAndSaturation(task, hue, targetSat); // FIXME
             // ok = addTaskSetEnhancedHueAndSaturation(task, targetHue, targetSat);
         }
         if (ok)
         {
+            if (taskRef.lightNode->manufacturerCode() == VENDOR_IKEA 
+                && hasHue && hasSat && !hasXy && !hasCt)
+            {
             // FIXME: do we need this?
-            // quint16 hue = hasHue ? targetHue : taskRef.lightNode->item(RStateHue)->toNumber();
-            // quint8 sat = hasSat ? targetSat : taskRef.lightNode->item(RStateSat)->toNumber();
-            //
-            // double r, g, b;
-            // double x, y;
-            // double h = (hue * 360.0) / 65535.0;
-            // double s = sat / 254.0;
-            // double v = 1.0;
-            //
-            // Hsv2Rgb(&r, &g, &b, h, s, v);
-            // Rgb2xy(&x, &y, r, g, b);
-            //
-            // if (x < 0) { x = 0; }
-            // else if (x > 1) { x = 1; }
-            // if (y < 0) { y = 0; }
-            // else if (y > 1) { y = 1; }
-            //
-            // x *= 65535.0;
-            // y *= 65535.0;
-            // if (x > 65279) { x = 65279; }
-            // else if (x < 1) { x = 1; }
-            // if (y > 65279) { y = 65279; }
-            // else if (y < 1) { y = 1; }
-            //
-            // item = task.lightNode->item(RStateX);
-            // if (item && item->toNumber() != static_cast<quint16>(x))
-            // {
-            //     item->setValue(static_cast<quint16>(x));
-            //     Event e(RLights, RStateX, task.lightNode->id(), item);
-            //     enqueueEvent(e);
-            // }
-            // item = task.lightNode->item(RStateY);
-            // if (item && item->toNumber() != static_cast<quint16>(y))
-            // {
-            //     item->setValue(static_cast<quint16>(y));
-            //     Event e(RLights, RStateY, task.lightNode->id(), item);
-            //     enqueueEvent(e);
-            // }
+             quint16 hue = hasHue ? targetHue : taskRef.lightNode->item(RStateHue)->toNumber();
+             quint8 sat = hasSat ? targetSat : taskRef.lightNode->item(RStateSat)->toNumber();
+             //DBG_Printf(DBG_INFO, "Tommy: hue: %d, sat:  %d\n", hue, sat);
+             double r, g, b;
+             double x, y;
+             double h = (hue * 360.0) / 65535.0;
+             double s = sat / 254.0;
+             double v = 1.0;
+            
+             Hsv2Rgb(&r, &g, &b, h, s, v);
+             Rgb2xy(&x, &y, r, g, b);
+            
+             if (x < 0) { x = 0; }
+             else if (x > 1) { x = 1; }
+             if (y < 0) { y = 0; }
+             else if (y > 1) { y = 1; }
+
+             //correction for color yellow and Ikea ColorBulb 
+             if (req.mode == ApiModeEcho && 
+                hue == 10923)
+             {
+                //real yellow
+                x = 0.5068;
+                y = 0.4715;
+             } 
+            
+             addTaskSetXyColor(task, x, y);
+
+             x *= 65535.0;
+             y *= 65535.0;
+             if (x > 65279) { x = 65279; }
+             else if (x < 1) { x = 1; }
+             if (y > 65279) { y = 65279; }
+             else if (y < 1) { y = 1; }
+             //DBG_Printf(DBG_INFO, "Tommy: x: %f, y: %f\n", x, y);
+            
+             ResourceItem *item = taskRef.lightNode->item(RStateX);
+             if (item && item->toNumber() != static_cast<quint16>(x))
+             {
+                 item->setValue(static_cast<quint16>(x));
+                 Event e(RLights, RStateX, task.lightNode->id(), item);
+                 enqueueEvent(e);
+             }
+             item = taskRef.lightNode->item(RStateY);
+             if (item && item->toNumber() != static_cast<quint16>(y))
+             {
+                 item->setValue(static_cast<quint16>(y));
+                 Event e(RLights, RStateY, task.lightNode->id(), item);
+                 enqueueEvent(e);
+             }
+            }
             // End FIXME
 
             if (hasHue)

`

acid115, 2020-11-16:
And if someone has an CT-Bulb from Ikea and uses alexa to control it, knows that the temperature of the bulb is set to low by alexa. You can fix it with the following and everythingbis nice. Not everyone uses Philips or whatever. But you are right, most of the people uses ikea bulbs, does not control it by a conbee, they uses the ikea gateway
```

else if (hasCt)
    {
        TaskItem task;
        copyTaskReq(taskRef, task);
        quint16 targetCtForChange = targetCt;

        //correct ikea ct 
        if (task.lightNode->manufacturerCode() == VENDOR_IKEA && 
                req.mode == ApiModeEcho )
        {    
            targetCtForChange += 65;
        }

        if (!isOn)
        {
            rsp.list.append(errorToMap(ERR_DEVICE_OFF, QString("/lights/%1/state").arg(id), QString("parameter, ct, is not modifiable. Device is set to off.")));
        }
        else if (taskRef.lightNode->isColorLoopActive())
        {
            rsp.list.append(errorToMap(ERR_PARAMETER_NOT_MODIFIEABLE, QString("/lights/%1/state").arg(id), QString("parameter, ct, is not modifiable. Colorloop is active.")));
        }
        else if (addTaskSetColorTemperature(task, targetCtForChange))
        {
```

Claus1985, 2020-11-26:
I’ve got the same problem with changing the color of a ikea tradfri bulb. Same configuration works for hue bulbs and for others but not for ikea. 

@Mimiix Please include the fix in next release.

Thx and brgds,
Claus

Claus1985, 2020-12-09:
Why does this fix not get merged? IKEA Bulbs still only work partially (no Color).

MattWestb, 2020-12-09:
Im very sorry but @manup !!!

MattWestb, 2020-12-10:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3765#issue-752455629

Mimiix, 2020-12-10:
@Claus1985 That's not how it works unfortunately 😅. You need to create a Pull request on the corresponding code section.

Claus1985, 2020-12-10:
@Mimiix thx for clarification 👍 
@acid115 can you create a pull request for the code section? Thx! :-)

acid115, 2020-12-11:
> @Mimiix thx for clarification 👍 
> @acid115 can you create a pull request for the code section? Thx! :-)

Hi, sorry, but i said, that i have no time for that :(  git and me is not so good. I am happy if i can merge the current changes into my branch, make my changes and build the application.

Claus1985, 2020-12-11:
@Mimiix can you or someone from the development team help out here? Does not have to be super fast just give us a hint that it will be integrated would be great.
This bug is really making the use of ikea color bulbs pointless since they can only be used for on/of and not for color.

Mimiix, 2020-12-11:
I have no clue how to do that 😅 . I can't force someone to do it, if a dev (or anyone else ) reads along and willing to do a PR on this that's fine.

ebaauw, 2020-12-11:
I don't like seeing Alexa-specific stuff end up in the REST API Plugin code.  This should be fixed in the Alexa skin (or whatever they call it).

I think it's a good idea (eventually) to accept RGB-colour-space values in the API, and translate these to `xy` values in the CIE 1931 colour space used by Zigbee.  Currently, each API client (including Homebridge Hue) needs to do that transformation themselves.  Not sure if we should accept HSL or RGB values (or both).  The HSL values would be [0° - 360°, 0% - 100%, 0% - 100%], and the RGB values would be [0 - 255, 0 - 255, 0 - 255] (or maybe [0.0 - 1.0, 0.0 - 1.0, 0.0 - 1.0]), in line with RGB colour space conventions.  This is probably more an API v2 discussion, and would definitely be incompatible with the Hue API. 

Note that, even after fixing the scale, the HSL values in the RGB colour space are very different from the Zigbee `hue` and `sat` values, as currently exposed by the API.  The Zigbee values are utterly useless, as they mean different colours (`xy` values) to different light models.  Setting `hue` and `sat` to (re-scaled) Hue/Saturation values from a computer or phone will result in mismatched colours.  Effectively, API clients, including the Alexa skin, should use `xy`, not `hue` and `sat`.

So far, I've only come across two colour lights that don't support `xy`.  The first one is the Philips LivingColours (from before the Hue era), that cannot be paired with deCONZ anyways.  The second one is the notorious LIDL Xmas light strip, which seems to use the RGB colour space, defying (almost) every Zigbee standard.  I've seen plenty of lights that support `xy`, but not `hue` and `sat`, like the Trådfri.

Claus1985, 2020-12-16:
@ebaauw I don´t get what you mean. For me this has nothing to do with alexa but only with deconz and tradfri bulbs.
Facts are: 
1. Tradfri bulbs need color commands in xy values 
2. Tradfri bulbs are supported by deconz (at least according to the deconz documentation) 
3. Deconz does not give the correct command to the bulbs else we wouldn't have a problem
> Please fix deconz code to probably support color change in ikea tradfri color bulbs.

Regarding the alexa stuff..
I control my lights through node red (which is controlled by loxone).
Now if I use the exact same color values which work flawlessly with philips hue bulbs through deconz on ikea tradfri bulb (also through deconz) it does not work. There is no alexa in between. Just node red sending the exact same command to deconz. So problem is with deconz and how it handles tradfri bulbs.

MGrotz, 2020-12-23:
@Claus1985
For me a combination of HSV > RGB and RGB > XY worked in NodeRed

`/*
/   Input from Payload: 
/    msg.payload.hue (0-360)
/    msg.payload.sat (0-255)
/    msg.payload.bri (0-255)
*/

var input_H = msg.payload.hue/360;
var input_S = msg.payload.sat/255;
var input_V = msg.payload.bri/255;

var rgb = HSVtoRGB(input_H,input_S,input_V);

msg = {}; 

msg.payload = "{\"on\":true,\"xy\":[" + rgb_to_cie(rgb.r,rgb.g,rgb.b) + "]}";
node.status({fill:"green",shape:"Dot",text: input + " / " + rgb_to_cie(rgb.r,rgb.g,rgb.b)});
node.send(msg);
return msg;

// Functions:

/* accepts parameters
 * h  Object = {h:x, s:y, v:z}
 * OR 
 * h, s, v | 0 <= h, s, v <= 1
*/
function HSVtoRGB(h, s, v) {
    var r, g, b, i, f, p, q, t;
    if (arguments.length === 1) {
        s = h.s, v = h.v, h = h.h;
    }
    i = Math.floor(h * 6);
    f = h * 6 - i;
    p = v * (1 - s);
    q = v * (1 - f * s);
    t = v * (1 - (1 - f) * s);
    switch (i % 6) {
        case 0: r = v, g = t, b = p; break;
        case 1: r = q, g = v, b = p; break;
        case 2: r = p, g = v, b = t; break;
        case 3: r = p, g = q, b = v; break;
        case 4: r = t, g = p, b = v; break;
        case 5: r = v, g = p, b = q; break;
    }
    return {
        r: Math.round(r * 255),
        g: Math.round(g * 255),
        b: Math.round(b * 255)
    };
}

/* accepts parameters
 * red, green, blue | 0 <= red, green, blue <= 255
*/
function rgb_to_cie(red, green, blue)
{
	//Apply a gamma correction to the RGB values, which makes the color more vivid and more the like the color displayed on the screen of your device
	red 	= (red > 0.04045) ? Math.pow((red + 0.055) / (1.0 + 0.055), 2.4) : (red / 12.92);
	green 	= (green > 0.04045) ? Math.pow((green + 0.055) / (1.0 + 0.055), 2.4) : (green / 12.92);
	blue 	= (blue > 0.04045) ? Math.pow((blue + 0.055) / (1.0 + 0.055), 2.4) : (blue / 12.92); 

	//RGB values to XYZ using the Wide RGB D65 conversion formula
	var X 		= red * 0.664511 + green * 0.154324 + blue * 0.162028;
	var Y 		= red * 0.283881 + green * 0.668433 + blue * 0.047685;
	var Z 		= red * 0.000088 + green * 0.072310 + blue * 0.986039;

	//Calculate the xy values from the XYZ values
	var x 		= (X / (X + Y + Z)).toFixed(4);
	var y 		= (Y / (X + Y + Z)).toFixed(4);

	if (isNaN(x))
		x = 0;
	if (isNaN(y))
		y = 0;	 
	return [x, y];
}
`

I send it through a json node and set in the deCONZ out-node Command to object:json:
[image]

Hope this helps.

Now I am looking for an option to convert back [x,y] to RGB or HSV, since the return values for HUE and SAT always provides same values e.g:

`{"alert":null,"bri":254,"colormode":"xy","effect":null,"hue":10,"on":true,"reachable":true,"sat":254,"xy":[0.1555,0.0498]}`

Same way of input also works with Philips HUE Go for me - and there the status (hue and sat) is provided correctly

MGrotz, 2020-12-23:
here the code for direct import to Node-red:

[{"id":"2e9a3b63.50bae4","type":"function","z":"b459556.3abafa8","name":"HSV > XY","func":"/*\n/   Input from Payload: \n/    msg.payload.hue (0-360)\n/    msg.payload.sat (0-255)\n/    msg.payload.bri (0-255)\n*/\n\nvar input_H = msg.payload.hue/360;\nvar input_S = msg.payload.sat/255;\nvar input_V = msg.payload.bri/255;\n\nvar rgb = HSVtoRGB(input_H,input_S,input_V);\n\nmsg = {}; \n\nmsg.payload = \"{\\\"on\\\":true,\\\"xy\\\":[\" + rgb_to_cie(rgb.r,rgb.g,rgb.b) + \"]}\";\nnode.status({fill:\"green\",shape:\"Dot\",text: input + \" / \" + rgb_to_cie(rgb.r,rgb.g,rgb.b)});\nnode.send(msg);\nreturn msg;\n\n\n\n// Functions:\n\n/* accepts parameters\n * h  Object = {h:x, s:y, v:z}\n * OR \n * h, s, v | 0 <= h, s, v <= 1\n*/\nfunction HSVtoRGB(h, s, v) {\n    var r, g, b, i, f, p, q, t;\n    if (arguments.length === 1) {\n        s = h.s, v = h.v, h = h.h;\n    }\n    i = Math.floor(h * 6);\n    f = h * 6 - i;\n    p = v * (1 - s);\n    q = v * (1 - f * s);\n    t = v * (1 - (1 - f) * s);\n    switch (i % 6) {\n        case 0: r = v, g = t, b = p; break;\n        case 1: r = q, g = v, b = p; break;\n        case 2: r = p, g = v, b = t; break;\n        case 3: r = p, g = q, b = v; break;\n        case 4: r = t, g = p, b = v; break;\n        case 5: r = v, g = p, b = q; break;\n    }\n    return {\n        r: Math.round(r * 255),\n        g: Math.round(g * 255),\n        b: Math.round(b * 255)\n    };\n}\n\n/* accepts parameters\n * red, green, blue | 0 <= red, green, blue <= 255\n*/\nfunction rgb_to_cie(red, green, blue)\n{\n\t//Apply a gamma correction to the RGB values, which makes the color more vivid and more the like the color displayed on the screen of your device\n\tred \t= (red > 0.04045) ? Math.pow((red + 0.055) / (1.0 + 0.055), 2.4) : (red / 12.92);\n\tgreen \t= (green > 0.04045) ? Math.pow((green + 0.055) / (1.0 + 0.055), 2.4) : (green / 12.92);\n\tblue \t= (blue > 0.04045) ? Math.pow((blue + 0.055) / (1.0 + 0.055), 2.4) : (blue / 12.92); \n\n\t//RGB values to XYZ using the Wide RGB D65 conversion formula\n\tvar X \t\t= red * 0.664511 + green * 0.154324 + blue * 0.162028;\n\tvar Y \t\t= red * 0.283881 + green * 0.668433 + blue * 0.047685;\n\tvar Z \t\t= red * 0.000088 + green * 0.072310 + blue * 0.986039;\n\n\t//Calculate the xy values from the XYZ values\n\tvar x \t\t= (X / (X + Y + Z)).toFixed(4);\n\tvar y \t\t= (Y / (X + Y + Z)).toFixed(4);\n\n\tif (isNaN(x))\n\t\tx = 0;\n\tif (isNaN(y))\n\t\ty = 0;\t \n\treturn [x, y];\n}\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":3110,"y":80,"wires":[["ef9dd77b.1fd8d8"]]}]

closed 2021-02-08 by Mimiix