dresden-elektronik/deconz-rest-plugin #3623: New Conbee II keeps bootloading (Windows and Raspbian)

HaythamB, 2020-11-10:
## Describe the bug
Got a fresh Conbee II stick, and attempted to initialize it in Windows 10. Device keeps connecting/disconnecting repeatedly and won't stay connected.

For the brief window that it is connected (about 10 seconds), deCONZ recognizes the device just fine, and attempts to connect. It then drops off and reconnects over and over again.

Also tried the devices on a Raspberry Pi 4 (which also has a working Raspbee II), and the same issue persists. lsusb detects the device (and so does the GCFFlasher) but it disconnects after 10 seconds and reconnects.

I have a feeling its a hardware issue with the stick I have, just wondering if anyone else has seen this before I ask for a replacement (in case this is a known issue)

## Steps to reproduce the behavior
1. Insert conbee II USB
2. See it connect
3. See it disconnect
4. Goto #2 :)

## Expected behavior
Should connect and remain connected.

## Screenshots

[image]
[image]

## Environment
Host: Raspberry pi, and Windows 10, both native installs
Firmware version: unknown, since it won't connect long enough to check
 - deCONZ version: (2.05.86)
 - Device: ConBee II
 - Do you use an USB extension cable: tried with and without, also removed all usb devices under windows except for keyboard and mouse. On the pi, I have a boot SSD on USB 3, conbee was plugged into USB 2 ports.

## deCONZ Logs
11:14:28:699 HTTP Server listen on address 0.0.0.0, port: 8090, root: /usr/share/deCONZ/webapp/
11:14:28:724 CTRL. 3.27.211:14:28:858 dev /dev/ttyAMA0 (/dev/ttyAMA0)
11:14:28:859 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
11:14:29:143 parent process bash
11:14:29:143 gw run mode: normal
11:14:29:143 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
11:14:29:145 DB sqlite version 3.27.2
11:14:29:146 DB PRAGMA page_count: 37
11:14:29:146 DB PRAGMA page_size: 4096
11:14:29:146 DB PRAGMA freelist_count: 2
11:14:29:146 DB file size 151552 bytes, free pages 2
11:14:29:146 DB PRAGMA user_version: 7
11:14:29:146 DB cleanup
11:14:29:146 DB create temporary views
10:14:29:176 started websocket server at port 20877
10:14:29:180 [INFO] - Found file containing button maps. Parsing data...
10:14:29:193 [INFO] - Button maps loaded.
10:14:29:197 dlg action: Read binding table
10:14:29:198 found node plugin: libde_rest_plugin.so - REST API Plugin
10:14:29:201 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
10:14:29:228 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
10:14:29:278 dev /dev/ttyAMA0 (/dev/ttyAMA0)
10:14:30:228 UPNP socket not bound, state: 0
10:14:30:340 dev /dev/ttyAMA0 (/dev/ttyAMA0)
10:14:30:348 COM failed to open /dev/ttyACM0: Device or resource busy
10:14:30:358 device disconnected reason: 4, index: 0
10:14:31:294 wait reconnect 15 seconds
10:14:31:342 dev /dev/ttyAMA0 (/dev/ttyAMA0)
10:14:32:295 wait reconnect 14 seconds
10:14:33:294 wait reconnect 13 seconds
10:14:33:990 dev /dev/ttyAMA0 (/dev/ttyAMA0)
10:14:33:991 GW update firmware not found: 
10:14:34:257 Announced to internet https://phoscon.de/discover
10:14:34:257 discovery server date: Tue, 10 Nov 2020 10:14:34 GMT
10:14:34:257 	 local time seems to be ok
10:14:34:257 discovery found version 2.04.35 for update channel stable
10:14:34:294 wait reconnect 12 seconds
10:14:34:965 New websocket 192.168.1.200:36512 (state: 3) 
10:14:35:294 wait reconnect 11 seconds
10:14:36:294 wait reconnect 10 seconds
10:14:37:294 wait reconnect 9 seconds
10:14:38:294 wait reconnect 8 seconds
10:14:39:294 wait reconnect 7 seconds
10:14:40:294 wait reconnect 6 seconds
10:14:41:294 wait reconnect 5 seconds
10:14:42:294 wait reconnect 4 seconds
10:14:43:294 wait reconnect 3 seconds
10:14:43:843 dev /dev/ttyAMA0 (/dev/ttyAMA0)
10:14:43:843 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
10:14:44:294 wait reconnect 2 seconds
10:14:45:294 wait reconnect 1 seconds
10:14:45:338 dev /dev/ttyAMA0 (/dev/ttyAMA0)
10:14:45:341 COM check bootloader
10:14:47:346 COM timout query bootloader, assume application
10:14:48:798 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:14:48:798 device state timeout ignored in state 2
10:14:49:294 start reconnect to network
10:14:50:144 shutdown after signal(2)
10:14:50:170 saved node state in 0 ms
10:14:50:186 sync() in 16 ms
10:14:50:187 REST API plugin shutting down
10:14:50:209 sync() in 10 ms

## Additional context
I have a fully functioning raspbee II on the pi, which I disconnected for testing.

Mimiix, 2020-11-10:
On windows, can you follow this guide? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

See if it works.

HaythamB, 2020-11-10:
Managed to catch it when it was identified and firmware flashed properly. Its working now. Thanks a million! Seems like the old firmware doesn't like Windows 10 ;)

closed 2020-11-10 by HaythamB

Mimiix, 2020-11-10:
@HaythamB You said it came like this out of the box. Can you confirm that after flashing manually it worked as intended?

HaythamB, 2020-11-10:
Yes that's exactly right. Out of the box was not working at all and kept reconnecting. By flashing it while it was still detected under Windows (about a 10 second period to do this), it managed to keep the bootloader connected and flashed it successfully. Just added some devices to it and they're working flawlessly. Maybe worth an FAQ note somewhere?