dresden-elektronik/deconz-rest-plugin #3625: ROB_200-014-0 Zigbee Rotary dimmers (r49) not sending ZCL report

SanderTRBRT, 2020-11-10:
## Describe the question or issue you are having
Since October 19th Robbshop is shipping the ROB_200-014-0 Zigbee Rotary dimmers with a different FW and MAC. These new dimmers do not update their state "directly" anymore. They ship with a new connection to also support a switch and have a new firmware 2.5.3_r49 (old one was 2.5.3_r47) and a new MAC starting with: BC33AC in stead of EC1BBD.

Setting the Min Report Interval to 1 and Max to 900 for On/Off(like their other dimmers) and Level Control to "Reportable change" = 1 does not help. I've set the same settings for the new dimmers as the old dimmers had. All the old dimmers send a "ZCL attribute report" in the logging every few seconds or so, but for these dimmers nothing is visible in the logging. The last seen message updates every 30 seconds (and then so does the state onoff, level, etc.), but afaik not based on a ZCL attribute since nothing is in the logging. (Does deConz do some regular polling every 30 seconds?)

Could it be this version (with different FW and MAC but same model identifier ROB_200-014-0) is not support yet in deCONZ?

## Screenshots

## Environment
Conbee2 with 2.05.86 on windows and 2.05.84 on a RPi4 (Home Assistant)

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on / Windows
 - Firmware version: (26660700)
 - deCONZ version: 2.05.84
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No
## deCONZ Logs

## Additional context

SwoopX, 2020-11-10:
Can you please share a screenshot of the node info panel, also including the expanded cluster list of the device +  a screenshot of the basic cluster where the attributes have been read?

SanderTRBRT, 2020-11-10:
Node info:
[image]
[image]
[image]

SanderTRBRT, 2020-11-10:
BTW I also tried in deCONZ .86 on windows, it has the same result. The blue dot only flashes once every 30 seconds or if I make a manual action in the deCONZ GUI. Never from an action on the device itself.

SwoopX, 2020-11-11:
Thanks! Tbh, I'm a bit puzzled that the devices would not send any reports here. Anticipating there might be other devices with a slightly changed modelID, I made the code changes pretty generic for this one here. Taken a look at the code already and all should come in nicely.

Could you please try to manually bind on/off and level control cluster as described here?
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-bind-server-clusters

After that, set the attribute reporting as before, just to be sure.

SanderTRBRT, 2020-11-11:
Thanks! What I did was bind it manually according to the wiki and it started working and directly reporting the state. I can see the ZCL reporting coming in in the logging (with dbg-info = 2).

Like mentioned the new version is 0xBC33ACFFFEBB2D21 (0x2a54), timestamp: 21:50:07:645 where it now receives a ZCL report.
The old model is also in the log down below everything starting with 0xec1bb* which is ZCL reporting automatically.

`00011876	417.55282593	[5672] 21:50:07:489 APS-DATA.request id: 178, addrmode: 0x02, addr: 0x2393, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00	
00011877	417.60723877	[5672] 21:50:07:544 APS-DATA.indication srcAddr: 0x0070, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 255, rssi: -50	
00011878	417.60745239	[5672] 21:50:07:544 Node data 0xec1bbdfffe940f3f profileId: 0x0104, clusterId: 0x0B04	
00011879	417.60763550	[5672] 21:50:07:544 ZCL attribute report 0xEC1BBDFFFE940F3F for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 	
00011880	417.60769653	[5672] 21:50:07:544  payload: 080521bf03	
00011881	417.60781860	[5672] 21:50:07:545 APS-DATA.request id: 181, addrmode: 0x02, addr: 0x0070, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x00	
00011882	417.65921021	[5672] 21:50:07:596 APS-DATA.confirm id: 178, status: 0x00 SUCCESS	
00011883	417.65930176	[5672] 21:50:07:596 APS-DATA.confirm request id: 178 -> erase from queue	
00011884	417.66815186	[5672] 21:50:07:605 APS-DATA.indication srcAddr: 0x2393, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0702, lqi: 255, rssi: -51	
00011885	417.66821289	[5672] 21:50:07:605 APS-DATA.request id: 178 erase from queue	
00011886	417.66833496	[5672] 21:50:07:605 Node data 0xec1bbdfffe941080 profileId: 0x0104, clusterId: 0x0702	
00011887	417.66848755	[5672] 21:50:07:605 ZCL attribute report 0xEC1BBDFFFE941080 for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000 	
00011888	417.66851807	[5672] 21:50:07:605  payload: 000025b55dd3060000	
00011889	417.66860962	[5672] 21:50:07:605 APS-DATA.request id: 184, addrmode: 0x02, addr: 0x2393, profile: 0x0104, cluster: 0x0702, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x00	
00011890	417.68615723	[5672] 21:50:07:623 APS-DATA.confirm id: 181, status: 0x00 SUCCESS	
00011891	417.68621826	[5672] 21:50:07:623 APS-DATA.confirm request id: 181 -> erase from queue	
00011892	417.68664551	[5672] 21:50:07:623 aps request id: 181 finished, erase from queue	
00011893	417.70825195	[5672] 21:50:07:645 APS-DATA.indication srcAddr: 0x2a54, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -50	
00011894	417.70846558	[5672] 21:50:07:645 Node data 0xbc33acfffebb2d21 profileId: 0x0104, clusterId: 0x0006	
00011895	417.70870972	[5672] 21:50:07:645 ZCL attribute report 0xBC33ACFFFEBB2D21 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 	
00011896	417.70877075	[5672] 21:50:07:646  payload: 00001001	`

So what does this exactly mean what is happening? Is the binding not working correctly and can this be fixed? Can I provide more information to help?

If I remove the binding, nothing is showing for 0xBC33ACFFFEBB2D21 / 0x2a54 only some neighbor information:

```
00004985	184.84542847	[5672] 21:46:14:782 APS-DATA.request id: 162, addrmode: 0x02, addr: 0x2393, profile: 0x0104, cluster: 0x0702, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00	
00004986	184.89077759	[5672] 21:46:14:827 APS-DATA.confirm id: 162, status: 0x00 SUCCESS	
00004987	184.89086914	[5672] 21:46:14:827 APS-DATA.confirm request id: 162 -> erase from queue	
00004988	184.91503906	[5672] 21:46:14:851 Node 0xBC33ACFFFEBB2D21 is known by 4 neighbors, last seen 0 s	
00004989	184.91528320	[5672] 21:46:14:851 aps request id: 162 finished, erase from queue	
```

SwoopX, 2020-11-13:
Well, the point I wanted to make is that there's no reason for the newer devices to behave differently, as they meet all the generically set checks as well. Additionally, there's a mechanism in place that should create missed bindings after 0.5 - 1 hrs.

Have you waited that long? Maybe delete the manually set binding before going to bed and check the next morning? Other than that, I could change one of the checks to go for the manufacturer code instead of the manufacturer identifier (in case that could be the issue, which it shouldn't).

SanderTRBRT, 2020-11-23:
It seems to work now. Had to change the update parameters though so it would more often send updates. However, that device is not the only one of all the dimmers which is giving these info message:

```
16:12:53:642 [INFO] - No button handler for: ROB_200-014-0 endpoint: 0x01 cluster: ONOFF (0x0006) command: 0x0A payload[0]: 000
16:12:54:405 [INFO] - No button handler for: ROB_200-014-0 endpoint: 0x01 cluster: 0x0B04 command: 0x0A payload[0]: 008
```

I know info isn't a bad thing, just strange that it is giving these messages all of a sudden, the dimmers with the older firmware do not cause this info message.

SwoopX, 2020-11-24:
Those messages are normal and nothing to worry about. Command 0x0A is attribute reporting which is not processed by the button handler function.

Mimiix, 2020-12-09:
@SanderTRBRT What did you change exactly? I have another user having the same issue ^^

closed 2021-02-08 by Mimiix