dresden-elektronik/deconz-rest-plugin #3627: OWON Zigbee Thermostat

leonardpitzu, 2020-11-10:
## Device

- Product name: Zigbee Smart Thermostat
- Manufacturer: OWON Technology Inc.
- Model identifier: PCT502
- Device type : Thermostat

- Product name: Zigbee Smart Thermostat
- Manufacturer: OWON Technology Inc.
- Model identifier: PR412C
- Device type : Thermostat (relay i guess?!?!)

The device bought comes in 2 pieces - one "desk clock" which measures temperature and humidity and is the actual "brain" of the thermostat (stores date/time/schedule) - and one relay connected to mains (it has 2 internal relays for various boilers and heaters - most basic central heating system in Europe is a on/off system). Product is labeled on the box as "OWON Zigbee Smart Thermostat PCT502-Z" without mentioning the second device product code.
The "desk clock" contains also a rechargeable battery.

https://www.owon-smart.com/2019-wholesale-price-zigbee-temperature-humidity-sensor-zigbee-combi-boiler-eu-smart-thermostat-temperature-controller-502-owon-product/

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-11-22:
Tbh, I got no clue how to properly support that thing, especially remotely and without traffic sniffing. I could give it a shot but you should provide additional screenshots of the thermostat cluster's attributes while those have been read and be able to compile a test version of the REST API plugin yourself to test.

leonardpitzu, 2020-11-22:
Compiling would be no issue as is debugging. Sniffing in zigbeeland I do not know but I am willing to start working on it

SwoopX, 2020-11-22:
Sounds good. Then don't forget about the screenshots. I guess I could give you something at hand almost as good as a bare zigbee sniffer ;)

leonardpitzu, 2020-11-23:
Basic cluster
[image]

leonardpitzu, 2020-11-23:
identify cluster
[image]

leonardpitzu, 2020-11-23:
groups cluster
[image]
[image]

leonardpitzu, 2020-11-23:
scenes cluster
[image]
[image]
[image]
[image]

leonardpitzu, 2020-11-23:
thermostat cluster

set & get weekly schedule result in FAILURE (yet there is a saved schedule); clearing the schedule works in deconz and in reality
get relay status log results in unsup_cluster_command
the rest is depicted below

[image]
[image]
[image]
[image]
[image]

leonardpitzu, 2020-11-23:
thermostat user interface configuration cluster
[image]

leonardpitzu, 2020-11-23:
relative humidity measurement cluster
[image]

leonardpitzu, 2020-11-23:
unknown cluster
[image]

leonardpitzu, 2020-11-23:
on/off cluster
[image]

leonardpitzu, 2020-11-23:
time cluster
[image]

leonardpitzu, 2020-12-20:
Is there anything more i can provide or do in order to help with the issue?

SwoopX, 2021-01-17:
I'd need toe remaining thermostat cluster attributes from attribute 0x0012 downwards. I'm afraid you'll probably need to manually read each of them via a double click on the value.

leonardpitzu, 2021-01-20:
[image]
[image]
[image]
[image]

closed 2021-02-15 by manup