dresden-elektronik/deconz-rest-plugin #3628: Device name

marval, 2020-11-10:
## Device

- Product name: SilverCrest Smart Motion Sensor
- Manufacturer: _TZ1800_fcdjzz3s
- Model identifier: TY0202
- Device type : 
  - Sensor

## Screenshots
  - Endpoints and clusters of the node
[image]
  - Node Info panel
[image]
  - Basic Cluster attributes in the Cluster Info panel.
[image]

Mimiix, 2020-11-10:
@marval Missing screenshots of all clusters. 

A

marval, 2020-11-10:
Just added them. I was not aware they were required since they were not part of the template.

Mimiix, 2020-11-11:
You and I are both right. They are in the template, but a bit down:

[image]

I might change that to show a bit earlier.

Thanks for adding :)!

closed 2020-11-28 by manup