dresden-elektronik/deconz-rest-plugin #3629: Samsung Leak Sensor First Generation

adawalli, 2020-11-11:
## Device

- Product name: Samsung SmartThings Water Leak Sensor
- Manufacturer: SmartThings
- Model identifier: moisturev4
- Device type : Please remove all unrelated device types. 
  - Water Sensor
  - Thermostat

## Screenshots

### Basic
[image]

### Power
[image]

[image]

### Binary Input (Basic) Cluster
[image]

### Poll Control
[image]

### Temperature Measurement
[image]

### IAS Zone
[image]

### OTAU
[image]

adawalli, 2020-11-15:
Is there any other information I can provide that would be of use?

Mimiix, 2020-11-16:
Nope, all is there. You need to wait for a dev to integrate it :)

SwoopX, 2020-11-22:
@adawalli The node info panel screenshot is missing here.

closed 2020-11-30 by manup