dresden-elektronik/deconz-rest-plugin #3631: Xiaomi Motion sensors - same model, different behavior

jan-dewit, 2020-11-11:
## Describe the bug
I have multiple Aqara/Xiaomi motion sensors (human body sensors) in my Zigbee network (Conbee I), I am having issues with especially one (which I already replaced, but the didnt help).

For example this one is really slow in updating both motion and lux values, it's the only sensor in my network which reports last reporting times of more than 16 minutes, while all others are way more faster.
This "faulty" sensor is also missing motion from time to time, while others are NEVER EVER missing any motion event.

## Steps to reproduce the behavior
It's happening interemittent, this motion sensor is just not working as expected, even not after replacing the sensor with a brand new model I experience the same behavior for this one, while all others are working as expected.

## Expected behavior
Report motion if I am within reach of 1 meter and report lux readings (I know these arent accurate).

## Screenshots

## Environment

 - Host system: Intel NUC Hassio VM
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26390500
 - deCONZ version: 2.05.84 / 9/14/2020
 - Device: ConBee I
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: RFLink 433Mhz gateway

## deCONZ Logs

## Additional context

SwoopX, 2020-11-11:
Can you please provide the node info panel for at least one "ever" working and the "faulty one + a screenshot of the expanded node clusters? Also, please provide a mesh screenshot where we can see to which parent (router) the devices are connected to and which device that is. Thanks!

jan-dewit, 2020-11-13:
The mesh screenshot containing two motion sensors:
Zolder Motion One = the "slow" one which is connected to Light 5, which is a TRADFRI bulb GU10 W 400lm
Zolder Motion Two = the normal operating one is connected directly to the Conbee controller

I am not really sure what you mean with the expanded node clusters screenshot.
I am happy to share more information but this is a first time for me ;-)

[image]

[image]
[image]

Smanar, 2020-11-13:
Not the same issue, but similar problem, End device using Ikea router https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3647

jan-dewit, 2020-11-13:
> Not the same issue, but similar problem, Xiaomi end device using Ikea router #3647

Interesting, thanks for the mention. To further pinpoint it would be cool to test the "slow" sensor while connected directly to the Conbee. These two motion sensors are close to the Conbee so the range is no issue.

Is there any way to force this motion sensor to pair directly with the Conbee instead of a Tradfri bulb?

SwoopX, 2020-11-13:
Unfortunately, only by cutting power to all other routers.

closed 2020-12-19 by stale[bot]