dresden-elektronik/deconz-rest-plugin #3632: Device name

thomasrenes, 2020-11-11:
## Device

- Product name: Sonoff ZBMINI
- Manufacturer: SONOFF
- Model identifier: 01MINIZB
- Device type: Switch

## Screenshots

[image]

[image]

Mimiix, 2020-11-11:
Hi,

Screenshots of other clusters are missing.

thomasrenes, 2020-11-11:
Sorry about that, didn't thought it would be relevant. Other screenshots added. Need anything else?

Mimiix, 2020-11-11:
Not that i can figure.  Need to wait for a dev :)

SwoopX, 2020-11-22:
I see no to do here...

Mimiix, 2020-11-23:
So it can be closed :) @SwoopX ?

SwoopX, 2020-11-24:
I'd say yes. I case any other issues appear, please raise a seperate issue.

closed 2020-11-24 by SwoopX

mkuutti, 2020-11-30:
I wonder if ZBMINIs work out of the box or why was the request closed?

SwoopX, 2020-11-30:
It works out of the box.

malfroid, 2020-12-05:
I tried adding a Sonoff ZBMINI to deCONZ. It works for me to switch dumb lights.

However, for smart bulbs, I would like to disconnect the switch from the relay (e.g. Shelly switches call this "detached mode"). Does anyone know if this is possible with Sonoff ZBMINI?

raddi, 2020-12-20:
I'm using Phoscon v2.7.1 and i can add the ZBMini without a problem. But: It is not usable. I can switch it only one time. After that switching it on will instantly turn it off again. I have 4 ZBMinis, none of them works.

JoeArny, 2020-12-22:
> 
> 
> I'm using Phoscon v2.7.1 and i can add the ZBMini without a problem. But: It is not usable. I can switch it only one time. After that switching it on will instantly turn it off again. I have 4 ZBMinis, none of them works.

Same here, the device is recognized but deconz can not even blink it when I click on its icon.

raddi, 2020-12-22:
Updated to latest beta (2.8) and restarted the raspberry, now everything works fine!

crusaderflex, 2021-01-09:
Same here, Version 2.9, device could be paired. I can switch device on, but with the phoscon app it would not go to off!

spiff42, 2021-01-22:
I have the same issue. I just wanted to note that in Phoscon my ZBMINI is reporting:
Vendor Heiman
Model 01MINIZB
Version 20200927
I bought it on AliExpress, so I don't know if the device is genuine or a knock-off.

LeoeLeoeL, 2021-01-23:
I don't know if this can help somebody.
It seems that ZBMINI in its default configuration works as a relay.
I connected ZBMINI to a PUSH BUTTON and works.
Short pressure to switch on and short pressure to switch off.
If you make a long pressure, it switches on but when you release it switches off.

jensorium, 2021-02-01:
with the Phoscon App it only allows to switch on, but the "light-icon" in the app stays dark. if i switch the light with the push button i am able to switch it on again

LeoeLeoeL, 2021-02-02:
This device has a very strange behaviour.
At first it worked as you wrote. Suddenly, without any external action, it begins to pair (green light is blinking). It pairs again with Conbee alone and after that the Sonoff works good with ON and OFF in Home Assistant and Phoscon.
Then suddenly "again", it starts for pairing again. It pairs again alone and this time it works as you wrote!!!!
This turnover happens mostly everyday.

nukleuz80, 2021-02-03:
Same issue here!

epoc83, 2021-02-21:
I have the same issue. The device turns on but in phoscon remains off (turns on and off very quick) If I assign to a group of lights, I can turn off the group and the zbmini also turns off. If I try to turn on and off using homebridge and HomeKit, I can turn it on, but not off. It instantly reports being off.

Mimiix, 2021-02-22:
Please stick to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4016

Im locking this one too until Manup replies there.