dresden-elektronik/deconz-rest-plugin #3633: Glitch when turn on light on specified brightness

Rihan9, 2020-11-11:
## Describe the bug
Hi,
when I call the light.turn_on service on home assitant application, often the light turning on slightly and after move to full brightness. 
After a little bit off analysis, this problems seems to be due to the service api/\<apikey\>/lights/\<lightnumber\>/state.
If the light is currently off when i call the service on the endpoint using the following json:
{"on": true, "transitiontime": 0, "bri": 254}
the light turn on with less brigtness (like 20 or 40%) and after a few seconds (or even less a second) it turn on at full brigtness.
It seems to happen with both osram and ikea bulbs.

I excluded an home assistant problem calling the endpoint directly via terminal and curl.

## Steps to reproduce the behavior
- turn off the light
- call /api/\<apikey\>/lights/\<lightnumber\>/state endpoint with this payload:  {"on": true, "transitiontime": 0, "bri": 254}

## Expected behavior

the light turn on at full brigtness directly
## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.05.84
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes, an external hdd powered with external power supply

## deCONZ Logs

nothing relevant on addon logs

## Additional context

js-chris, 2020-12-03:
Hi,

I noticed exactly the same thing for my RaspBee ||, after updating deCONZ from 2.05.52 to 2.7.0-beta and Firmware to deCONZ_RaspBeeII_0x26680700.bin.GCF.

It seems to me, that only my Lamps from Neuhaus Lighting Group (NLG-RGBW light) are effected.

`{
        "colorcapabilities": 31,
        "ctmax": 295,
        "ctmin": 275,
        "etag": "fc46b58ea93e682cd1e6e8d3c03a838b",
        "hascolor": true,
        "lastannounced": null,
        "lastseen": "2020-12-03T07:56Z",
        "manufacturername": "Neuhaus Lighting Group",
        "modelid": "NLG-RGBW light",
        "name": "HaustuerFlurDeckenlampe",
        "state": {
            "alert": "none",
            "bri": 75,
            "colormode": "ct",
            "ct": 295,
            "effect": "none",
            "hue": 6949,
            "on": false,
            "reachable": true,
            "sat": 179,
            "xy": [
                0.417,
                0.3988
            ]
        },
        "swversion": "1001-0007",
        "type": "Extended color light",
        "uniqueid": "00:15:8d:00:02:57:57:07-02"
    }`

The effect is as described above, but I got some new insights:

If I switch the lamp on (with predefined value higher 50) OR dim to an value higher than 50, I noticed the lamp dims up to 50, stops for a (milli)second and dims further to 75.
PUT /api/xxx/lights/2/state 
{"on" : true }
{"on" : true, "bri": 75, "transitiontime": 0 }

if I dim to 30, it seems to me vice versa, the lamp dims to 50 and then down to 30.

It is really hard to say, if 50 is the right value, but it is very clear, that it dims to a certain value, stops, and dims further.

Not really nice and it is a new behaviour for me, I did not have this before I updated.

Thanks in advance
Chris

PS: Everything else worked perfect with and after the update

closed 2020-12-31 by stale[bot]

js-chris, 2021-01-04:
Hi,

I updated to 2.7.1-stable and have still the same Problem. Paul Neuhaus Lamps react slow and if I switch them on, they dim to about 20/30% and then up to their final value.

How can this be changed?

thanks in advance
Chris