dresden-elektronik/deconz-rest-plugin #3634: Firmware cannot connect to Gateway after a period or setting wrong parameter

boeserwatz, 2020-11-11:
My previous Zigbee stick ran superbly immediately under all environments (Win10/CCU3) – it didn't make the 30m and is supposed to be sloppy with more than 20 net participants – so I thought I would take the conbee 2
Last week I ordered my first Conbee 2 – I never got it installed.

My target was to make it run on a ccu3 with redmatic and the node node-red-contrib-zigbee.
On the ccu3 I was only able to connect to the adapter - but never get a pairing.
I then tried to run this first Conbee 2 on a PC with Windows 10. DeConz installed and adapter plugged in. But the software has never displayed more than "joining".

For testing I used an old notebook with USB2.0 and also a USB extension.
On this notebook I couldn't get any further than "joining" status. 
The Flashtool provides the correct data - the Flashtool could also be used to update the FW. This FW was also displayed correctly in the ccu3 – but more than the coordinator with the new FW could not be displayed.
I also tested the adapter on a PI3B+. Image loaded from the homepage. desktop version. Starts, starts immediately deConz and the Conbee 2 – but shows no more than the status "joining".

Now I thought - well, broken.
So yesterday I ordered my second conbee 2 and got it today.
I operated this with the USB extension on a notebook with pure **USB 3.0** ports and was able to set up and "join" a network. Two tests were carried out with an IKEA lamp and an OSRAM plug and these could be paired and operated. The network was displayed correctly in the deCONZ and the devices could be operated via the Gateway app. The gateway and firmware were also loaded and connected. The first conbee 2 couldn't do that. 
I thought that was it (sorry for my bad english). 

Then the net and the nodes in the deCONZ are deleted and the stick is plugged into the ccu3 with extension. But only the coordinator was displayed there and a pairing did not work!?
So, the stick back to the notebook where it already worked. But there now the same behavior as in the first conbee 2. 
No connection to firmware. The deConz software does not come to the status "Network Connected". 

What can I do – a trick? 
It seems the ccu3/redmatic/node-red-contrib-zigbee set a parameter that the FW no longer lets starts? 
Can I perhaps return to the previous conbee 2 setting or factory setting?

## Screenshots
[image]

[image]

[image]

## Environment
 - Host system: Raspberry Pi / PC / CCU3
 - Running method: redmatic/zigbee
 - Firmware version: deCONZ_ConBeeII_0x26660700
 - deCONZ version: 2.05.84
 - Device: ConBee II
 - Do you use an USB extension cable: yes

Smanar, 2020-11-11:
Hello

>I operated this with the USB extension on a notebook with pure USB 3.0 ports

Avoid USB 3.0 for Zigbee gateway.

>Then the net and the nodes in the deCONZ are deleted and the stick is plugged into the ccu3 with extension. But only the coordinator was displayed

It' s not normal ?

when you are making the test on PC / Win10 you don't have USB disconnections in loop ?

Where have you take the deconz capture ? The node was probably out of network during it.

The notebook is on windows too ? Have yout tried to run deconz with it ? Can you compare the setting ith this one https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

Mimiix, 2020-11-11:
The stick does not store the network. The software does. So plugging it into something else wouldn't help.

boeserwatz, 2020-11-11:
ok - the network is stored on my notebook with the usb 3.0 connection - works fine till i take the conbee 2 to my ccu3 - in the ccu3.  i only saw the coordinator with the correkt firmware but could not pair any device. 
i think the implemetation is not complete [https://github.com/Koenkk/zigbee-herdsman/issues/72](url). But thats ok. 
My second plan ist to bulid a home assist an provid the zigbee net from there to the ccu3. 
But now the conbee 2 seems to hang or can not start correct.

Mimiix, 2020-11-11:
I have no clue what ccu is. Also, that repo is not deconz.

boeserwatz, 2020-11-11:
For Homematic look at this [https://www.homematic-ip.com/en/products/detail/smart-home-central-control-unit-ccu3.html](url)

Mimiix, 2020-11-12:
I really have no clue how this should run deconz. You can't simply plug the Stick in there.

boeserwatz, 2020-11-12:
Thats may be the problem for the software in the ccu3. 
But this will not explain the behavior that the conbee 2 works first and then is no longer reachable.

Mimiix, 2020-11-12:
So for my understanding:
YOu see the lights in Windows but once in CCU it doesnt show the lights/nodes anymore.

boeserwatz, 2020-11-12:
First i saw all lights and a plug in in deconz with the conbee 2 - the i delete all notes and the net - because i thougth the stick is ok and now it can work in my ccu3 - but more than the coordinator was not shown - then back to pc and also there the stick is not reachable - and one thing more i have massiv problmes with me usb - devices connect and disconnect when the conbee ist connected

Mimiix, 2020-11-12:
Okay, 2 things here.

The stick does save NOTHING. So your windows had them paired, but that won't go for the CCU: That needs to pair too. The software stores the devices that are known.

To fix that connect/disconnect:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Follow that.

boeserwatz, 2020-11-12:
ok - stick store no info of devices - no problem - the plan is that the ccu3/redmatic/zigbee-node will do this. 

the FM-update is done - but i saw it in no tool (deconz or phosconApp) because the software can not connect to fw. i saw the fw number in the ccu3 software and it was the highest version. Also the flashingtool reported success with deCONZ_ConBeeII_0x26660700

Mimiix, 2020-11-12:
Is it working in Windows now again?

Once again: I can't speak for CCU3 on how to use deCONZ with that : We don't support that.

boeserwatz, 2020-11-12:
it is ok - no support for ccu3 - but i have connected two conbee 2 sticks to ccu3 and after that the communication to the firmware was not possible - i will try a retest on my pi3b+ (flashing the newest firmware and check to start the gateway) - this will take a while - i'll be back 👍

boeserwatz, 2020-11-12:
ok now i made a second test on a pi3b+ - only usb 2.0 ports. newest software and firmware. but the conbee 2 can not connected.

here my cmd log:
`pi@phoscon:~ $ sudo systemctl stop deconz-gui
pi@phoscon:~ $ wget https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
--2020-11-12 12:43:25--  https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
Auflösen des Hostnamens deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)... 84.19.195.242
Verbindungsaufbau zu deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)|84.19.195.242|:443 ... verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet ... 200 OK
Länge: 164120 (160K) [application/x-graphing-calculator]
Wird in »deCONZ_ConBeeII_0x26660700.bin.GCF.1« gespeichert.

deCONZ_ConBeeII_0x2 100%[===================>] 160,27K   692KB/s    in 0,2s    

2020-11-12 12:43:26 (692 KB/s) - »deCONZ_ConBeeII_0x26660700.bin.GCF.1« gespeichert [164120/164120]

pi@phoscon:~ $ sudo GCFFlasher_internal -l
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2xxxxxxx  | ConBee II 
pi@phoscon:~ $ sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26660700.bin.GCF
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 26660700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164106 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
deCONZ firmware version 26660700
pi@phoscon:~ $ wget https://deconz.dresden-elektronik.de/raspbian/beta/deconz-latest-beta.deb
--2020-11-12 12:44:54--  https://deconz.dresden-elektronik.de/raspbian/beta/deconz-latest-beta.deb
Auflösen des Hostnamens deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)... 84.19.195.242
Verbindungsaufbau zu deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)|84.19.195.242|:443 ... verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet ... 200 OK
Länge: 6139204 (5,9M) [application/x-debian-package]
Wird in »deconz-latest-beta.deb.1« gespeichert.

deconz-latest-beta. 100%[===================>]   5,85M  2,89MB/s    in 2,0s    

2020-11-12 12:44:56 (2,89 MB/s) - »deconz-latest-beta.deb.1« gespeichert [6139204/6139204]

pi@phoscon:~ $ sudo dpkg -i deconz-latest-beta.deb
(Lese Datenbank ... 148326 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von deconz-latest-beta.deb ...
Entpacken von deconz (2.05.86-raspbian-stretch-beta) über (2.05.86-raspbian-stretch-beta) ...
deconz (2.05.86-raspbian-stretch-beta) wird eingerichtet ...
Trigger für gnome-menus (3.31.4-3) werden verarbeitet ...
Trigger für desktop-file-utils (0.23-4) werden verarbeitet ...
Trigger für mime-support (3.62) werden verarbeitet ...
pi@phoscon:~ $`

deconz shows this:
[image]
[image]

phoscon app 
[image]

Mimiix, 2020-11-12:
Can you use a extension cable and check again?

boeserwatz, 2020-11-12:
i tested without anything between conbee 2 and pi. 
then with a 2.0 usb hub (powered).
and then with a extension cable.
nothing works - no connection to firmware

boeserwatz, 2020-11-12:
you write that you not support the ccu3 platform - But your colleague has offered support for the converter here [https://github.com/Koenkk/zigbee-herdsman/issues/72](url) – maybe he could help us

boeserwatz, 2020-11-12:
another question - is it possible that i brick that conbee 2?

Smanar, 2020-11-12:
From my memory, no, the loader is protected, so you can't overwrite it, and if it work you can update the firwmare again.

On the pi3, are you sure it s the good ip ? Are you sure you haven't 2 deconz instance/service in same time ? Have you tried to run it in debug mode ? 

`deCONZ --dbg-error=2 --dbg-info=2`

boeserwatz, 2020-11-12:
Good ip? What do mean?

I use this image:
[image]
When i restart the Raspberry and deconz starts i select conbee 2 and push connect. 
I think deConz runs only one instance. The taskmanager shows one instance.

my next step is to get a log from deconz with your hint for the debug mode.

I think the PANID is one of the signs. My working machine parameter for the panid was 0xffff. And i think this parameter is set by my ccu3. I try to set it to another value (for example 0x1234) - but i can not set any parameter since the adapter is not connecting - and the phosconApp show the text "nicht verbunden" not connected. May be it is an impossible parameter combination and the firmware or the bootloader crashes.

boeserwatz, 2020-11-12:
here the log from deconz with debug on. It starts and then i try to change a parameter(read/save/done)

[deconzlog01.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5532624/deconzlog01.txt)

boeserwatz, 2020-11-12:
i read here about my problem [https://github.com/Koenkk/zigbee2mqtt/issues/4746#issuecomment-717912736](url)

so i tried to reset my conbee 2. But i get this: 

[image]

boeserwatz, 2020-11-12:
hmm - when i retry the reset - the reset works - i think the reboot changes the device from ...ACM1 to ...ACM0 - so the error raises.

After successfull reset - my conbee 2 is still not reachable - ich dreh bald durch :(

Smanar, 2020-11-13:
On your log, you had problem, but the device finally connected, but it can disconnect too later..

>I think the PANID is one of the signs. My working machine parameter for the panid was 0xffff. And i think this parameter is set by my ccu3

But how it can change this value ....

Mimiix, 2020-11-13:
Forwarded to @manup , but i still suspect user error.

The uart often happens on VM's / docker's.

boeserwatz, 2020-11-13:
ok - but i can not change the parameters - when i set them and read them they seems stored - but when i remove the stick and put it back in the usb port the setting are gone - seems they never write to the stick/conbee 2.
it is bricked

Mimiix, 2020-11-13:
You mean that if you put it in the same device? Or antoher one?

manup, 2020-11-13:
When changing parameters in deCONZ and hitting save you also need to "Leave" + "Join" in order to get parameters applied and saved in NVRAM.

boeserwatz, 2020-11-13:
@Mimiix: the same device/conbee 2.
@manup: the software show me only "Joining" - i could not hit "Leave" or "Join".

manup, 2020-11-13:
The configuration looks ok, but it looks a bit that the Zigbee stack can't open the network. I think this might be interference and therefore it's stuck in trying to join.

Can you please start deCONZ from a terminal as follows:

`deCONZ --dbg-info=1 --dbg-error=1`

And show the logs, the status codes should bring some insights whats happening.

boeserwatz, 2020-11-13:
I installed deCONZ 2.05.88 with debug view. Is it sufficient to activate the function?

manup, 2020-11-13:
Yes this works as well shows the same logs if enabled

boeserwatz, 2020-11-13:
i started deCONZ and for full log of connect process i hit on "reboot device"

20:20:34:632 wait reconnect 6 seconds
20:20:35:632 failed to reconnect to network try=7
20:20:35:634 wait reconnect 5 seconds
20:20:35:747 dev /dev/ttyS0 (/dev/ttyS0)
20:20:36:633 wait reconnect 4 seconds
20:20:37:633 wait reconnect 3 seconds
20:20:38:632 wait reconnect 2 seconds
20:20:39:633 wait reconnect 1 seconds
20:20:39:682 dev /dev/ttyS0 (/dev/ttyS0)
20:20:39:690 COM check bootloader
20:20:40:633 failed to reconnect to network try=8
20:20:41:703 COM timout query bootloader, assume application
20:20:41:732 Device firmware version 0x26660700 ConBee II
20:20:41:766 unlocked max nodes: 200
20:20:41:848 Device protocol version: 0x010C
20:20:41:863 new node - ext: 0x00212effff065cde, nwk: 0x0000
20:20:41:948 Current channel 11
20:20:41:969 CTRL got nwk update id 0
20:20:41:979 CTRL ANT_CTRL 0x03
20:20:42:027 Device protocol version: 0x010C
20:20:42:087 Current channel 11
20:20:42:102 CTRL got nwk update id 0
20:20:42:108 CTRL ANT_CTRL 0x03
20:20:45:707 dev /dev/ttyS0 (/dev/ttyS0)
20:20:45:711 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
20:20:45:714 GW firmware version: 0x26660700
20:20:45:714 GW firmware version is up to date: 0x26660700
20:21:26:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:26:634 device state timeout ignored in state 3
20:21:29:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:29:633 device state timeout ignored in state 3
20:21:32:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:32:633 device state timeout ignored in state 3
20:21:35:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:35:633 device state timeout ignored in state 3
20:21:38:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:38:634 device state timeout ignored in state 3
20:21:41:632 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:41:633 device state timeout ignored in state 3
20:21:44:632 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:44:633 device state timeout ignored in state 3
20:21:47:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:47:634 device state timeout ignored in state 3
20:21:50:632 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:50:633 device state timeout ignored in state 3
20:21:53:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:53:634 device state timeout ignored in state 3
20:21:56:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:56:634 device state timeout ignored in state 3
20:21:59:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
20:21:59:634 device state timeout (handled)
20:21:59:707 dev /dev/ttyS0 (/dev/ttyS0)
20:21:59:714 device disconnected reason: 1, index: 1
20:21:59:749 device disconnected reason: 1, index: 2
20:22:00:635 failed to reconnect to network try=9
20:22:00:636 wait reconnect 15 seconds
20:22:01:132 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
20:22:01:133 device state timeout ignored in state 1
20:22:01:633 wait reconnect 14 seconds
20:22:02:633 wait reconnect 13 seconds
20:22:03:633 wait reconnect 12 seconds
20:22:04:633 wait reconnect 11 seconds
20:22:05:633 failed to reconnect to network try=10
20:22:05:634 wait reconnect 10 seconds
20:22:06:633 wait reconnect 9 seconds
20:22:07:633 wait reconnect 8 seconds
20:22:08:632 wait reconnect 7 seconds
20:22:09:633 wait reconnect 6 seconds
[... 48 lines cut ...]
20:23:00:633 failed to reconnect to network try=3
20:23:00:634 wait reconnect 6 seconds
20:23:01:633 wait reconnect 5 seconds
20:23:02:633 wait reconnect 4 seconds
20:23:03:633 wait reconnect 3 seconds
20:23:04:633 wait reconnect 2 seconds
20:23:05:632 failed to reconnect to network try=4
20:23:05:633 wait reconnect 1 seconds
20:23:05:687 dev /dev/ttyS0 (/dev/ttyS0)
20:23:05:700 COM check bootloader
20:23:05:711 COM detected application
20:23:05:818 Device firmware version 0x26660700 ConBee II
20:23:05:827 unlocked max nodes: 200
20:23:05:916 Device protocol version: 0x010C
20:23:05:996 Current channel 11
20:23:06:023 CTRL got nwk update id 0
20:23:06:031 CTRL ANT_CTRL 0x03
20:23:06:079 CTRL got stack debug assert code: 0x8000, type: 0x02
20:23:06:091 Device protocol version: 0x010C
20:23:06:169 Current channel 11
20:23:06:196 CTRL got nwk update id 0
20:23:06:210 CTRL ANT_CTRL 0x03
20:24:45:713 dev /dev/ttyS0 (/dev/ttyS0)
20:24:45:721 GW firmware version: 0x26660700
20:24:45:723 GW firmware version is up to date: 0x26660700

manup, 2020-11-13:
Not quite yet, I'm looking for "NET ZDO start network status 0x.." this should appear when it is running a while.

boeserwatz, 2020-11-13:
I wait for 30 min but there is no log entry with NET ZDO...
Should i perform another function?

BTW the log say 
"dev .../ttyS0" but my conbee 2 is mount on ttyAM0 - is that correct?

[image]

boeserwatz, 2020-11-13:
no sorry mount to ttyAM1

manup, 2020-11-13:
The /dev/ttyACM1 change like happend after the firmware rebooted and USB reenumeration.

Not sure why the network status code isn't shown, but I've looked into the firmware.
Can you please try to enable channels 11,15,20,25 in network settings and press save.

The firmware should try to connect on each of these channels.

manup, 2020-11-13:
Ah one more thing, uncheck predefined PanID

boeserwatz, 2020-11-13:
Set parameter and hit save
21:00:45:681 dev /dev/ttyS0 (/dev/ttyS0)
21:00:45:689 GW firmware version: 0x26660700
21:00:45:692 GW firmware version is up to date: 0x26660700
21:04:40:930 Announced to internet https://phoscon.de/discover
21:04:40:931 discovery server date: Fri, 13 Nov 2020 21:04:40 GMT
21:04:40:932 	 local time seems to be ok
21:04:45:722 dev /dev/ttyS0 (/dev/ttyS0)
21:04:45:730 GW firmware version: 0x26660700
21:04:45:734 GW firmware version is up to date: 0x26660700
21:08:45:713 dev /dev/ttyS0 (/dev/ttyS0)
21:08:45:723 GW firmware version: 0x26660700
21:08:45:727 GW firmware version is up to date: 0x26660700
21:12:45:702 dev /dev/ttyS0 (/dev/ttyS0)
21:12:45:710 GW firmware version: 0x26660700
21:12:45:714 GW firmware version is up to date: 0x26660700
21:16:45:684 dev /dev/ttyS0 (/dev/ttyS0)
21:16:45:692 GW firmware version: 0x26660700
21:16:45:695 GW firmware version is up to date: 0x26660700
21:20:45:718 dev /dev/ttyS0 (/dev/ttyS0)
21:20:45:729 GW firmware version: 0x26660700
21:20:45:731 GW firmware version is up to date: 0x26660700
21:21:32:459 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:32:460 device state timeout ignored in state 3
21:21:35:611 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:35:612 device state timeout ignored in state 3
21:21:38:632 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:38:633 device state timeout ignored in state 3
21:21:41:632 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:41:633 device state timeout ignored in state 3
21:21:44:632 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:44:633 device state timeout ignored in state 3
21:21:47:632 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:47:633 device state timeout ignored in state 3
21:21:50:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:50:634 device state timeout ignored in state 3
21:21:53:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:53:634 device state timeout ignored in state 3
21:21:56:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:56:634 device state timeout ignored in state 3
21:21:59:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:21:59:634 device state timeout ignored in state 3
21:22:02:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:22:02:634 device state timeout ignored in state 3
21:22:05:633 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
21:22:05:634 device state timeout (handled)
21:22:05:707 dev /dev/ttyS0 (/dev/ttyS0)
21:22:05:710 device disconnected reason: 1, index: 1
21:22:05:743 device disconnected reason: 1, index: 2
21:22:06:633 wait reconnect 15 seconds
21:22:07:132 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
21:22:07:133 device state timeout ignored in state 1
21:22:07:632 wait reconnect 14 seconds
21:22:08:632 wait reconnect 13 seconds
21:22:09:633 wait reconnect 12 seconds
21:22:10:633 failed to reconnect to network try=5
21:22:10:634 wait reconnect 11 seconds
21:22:11:632 wait reconnect 10 seconds
21:22:12:632 wait reconnect 9 seconds
21:22:13:633 wait reconnect 8 seconds
21:22:14:632 wait reconnect 7 seconds
21:22:15:632 failed to reconnect to network try=6
21:22:15:633 wait reconnect 6 seconds
21:22:16:633 wait reconnect 5 seconds
21:22:17:633 wait reconnect 4 seconds
21:22:18:633 wait reconnect 3 seconds
21:22:19:632 wait reconnect 2 seconds
21:22:20:633 failed to reconnect to network try=7
21:22:20:635 wait reconnect 1 seconds
21:22:20:689 dev /dev/ttyS0 (/dev/ttyS0)
[... 81 lines cut ...]
21:48:45:686 dev /dev/ttyS0 (/dev/ttyS0)
21:48:45:689 GW firmware version: 0x26660700
21:48:45:690 GW firmware version is up to date: 0x26660700
21:49:40:941 Announced to internet https://phoscon.de/discover
21:49:40:942 discovery server date: Fri, 13 Nov 2020 21:49:40 GMT
21:49:40:943 	 local time seems to be ok
21:52:21:649 saved node state in 0 ms
21:52:21:674 sync() in 23 ms
21:52:45:688 dev /dev/ttyS0 (/dev/ttyS0)
21:52:45:697 GW firmware version: 0x26660700
21:52:45:699 GW firmware version is up to date: 0x26660700
21:53:17:416 deCONZ configuration changed
21:53:21:876 Device protocol version: 0x010C
21:53:21:961 Current channel 11
21:53:21:989 CTRL got nwk update id 0
21:53:22:003 CTRL ANT_CTRL 0x03
21:53:53:523 Device protocol version: 0x010C
21:53:53:592 Current channel 11
21:53:53:612 CTRL got nwk update id 0
21:53:53:622 CTRL ANT_CTRL 0x03
21:54:14:107 deCONZ configuration changed
21:54:20:487 Device protocol version: 0x010C
21:54:20:566 Current channel 11
21:54:20:590 CTRL got nwk update id 0
21:54:20:597 CTRL ANT_CTRL 0x03

manup, 2020-11-13:
> CTRL got stack debug assert code: 0x8000, type: 0x02

Ah that's something, need to look that up later today, these are assert codes within the firmware at a specific place.

boeserwatz, 2020-11-16:
Are there any ideas left? 
I already spend more than 20 hours searching for a solution. 
I'm almost ready and send the adapter back. :(

manup, 2020-11-16:
Sorry for the late response. I'm working on an firmware update to mitigate the issue, hope to have it ready on Wednesday.
So far I've investigated the code and have found one invalid state where it can be stuck in endless joining, I think this is the case here.

boeserwatz, 2020-11-16:
ok - i'll wait for it - happy debugging

manup, 2020-11-18:
After a little testing I could re-create the bug by setting Predefined PANID to true and PANID to `0xffff` which is an invalid combination. The following firmware fixes this when the network is started. It sets Predefined PANID to false when PANID is `0xffff` and a new random PANID will be selected.

http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26670700.bin.GCF

boeserwatz, 2020-11-18:
@manup: you got it - thanks a lot
I installed the new firmware and deCONZ was able to get over the "joining" status. The status "In Network" has been reached.
The PhosconApp can also connect to the firmware.
A lamp could also be paired and controlled in the raspberry environment (PI3B+/Your Image/and USB without extension cable).

@Mimiix: i think it was a bug - not a user error :) - thank you too

I want to do another test and check if the adapter is now running in the ccu3 - i will report here

[image]

[image]

[image]

[image]

boeserwatz, 2020-11-18:
now it is perfect - also "ccu3/redmatic/with a special node for conbee" runs. And the stick is much more stronger and faster then the lowprice one i used befor - thanks for your support

[image]

manup, 2020-11-18:
Cool, thanks for the feedback.
The bug is very rare, I now just of two cases from the past where this happened.
Glad that it works now and the bug is finally fixed :)

Mimiix, 2020-11-18:
>@Mimiix: i think it was a bug - not a user error :) - thank you too
@boeserwatz I'm glad you've proven me otherwise. Thanks for the report and effort!

chris4my, 2020-11-18:
HI, I have a problem with connection of conbee it says in phoscon NO CONNECTION.
i can not access to downgrade or upgrade in phoscon. I have ver.2.05.88 / 2020-10-15 and it says ConBee 2 but I have the old conbee. has switched to another USB RPi3 + and also switched to a longer usb cable but it will be the same error NO CONNECTION .have tried to connect with VNC viewer but NO CONNECTION. Has / dev / ttyUSB0 in the configuration for deConz. What can you do?
[image]
[image]
[image]

Mimiix, 2020-11-20:
@chris4my please open a own issue if the issue persist after trying this: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

**Closing this issue as OP is helped.**

closed 2020-11-20 by Mimiix