dresden-elektronik/deconz-rest-plugin #3636: No Long Press Actions on IKEA TRADFRI 5-Button Remote's LEFT/RIGHT Buttons

perfecblue, 2020-11-12:
## Describe the question or issue you are having
IKEA's Tradfri 5-button round remote's Previous (LEFT) and Next (RIGHT) buttons do not generate long press even though it is shown as an available action in homekit.

The Dim Up (UP) and Dim Down (DOWN) buttons are generating long press event completely fine in homekit but strangely the LEFT and RIGHT buttons don't do so at all.

Upon long press, the LEFT and RIGHT buttons simply send out repeating actions instead of a long press action.

Is this a technical limitation of the remote or is this a possible bug?

Thank you!

## Screenshots

## Environment
Host system: Raspberry Pi 4 with Homebridge
Running method: Raspbian
Firmware version: 26660700
deCONZ version: 2.05.84
Device: ConBee II
Do you use an USB extension cable: YES
Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs
```
[12/11/2020, 19:55:44] [Hue] 主臥室 IKEA Remote Dim Up: homekit button single press
[12/11/2020, 19:55:49] [Hue] 主臥室 IKEA Remote Dim Up: homekit button long press
[12/11/2020, 19:55:51] [Hue] 主臥室 IKEA Remote Dim Down: homekit button single press
[12/11/2020, 19:55:57] [Hue] 主臥室 IKEA Remote Dim Down: homekit button long press
[12/11/2020, 19:56:00] [Hue] 主臥室 IKEA Remote Previous: homekit button single press
[12/11/2020, 19:56:02] [Hue] 主臥室 IKEA Remote Previous: homekit button single press
[12/11/2020, 19:56:03] [Hue] 主臥室 IKEA Remote Previous: homekit button single press
[12/11/2020, 19:56:06] [Hue] 主臥室 IKEA Remote Previous: homekit button single press
[12/11/2020, 19:56:08] [Hue] 主臥室 IKEA Remote Next: homekit button single press
[12/11/2020, 19:56:09] [Hue] 主臥室 IKEA Remote Next: homekit button single press
[12/11/2020, 19:56:11] [Hue] 主臥室 IKEA Remote Next: homekit button single press
[12/11/2020, 19:56:12] [Hue] 主臥室 IKEA Remote Next: homekit button single press
```

## Additional context

Smanar, 2020-11-12:
Hello, perhaps https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2618 ?

perfecblue, 2020-11-12:
> 
> 
> Hello, perhaps #2618 ?

Thanks for pointing me to the right thread, I can now confirm that the LEFT/RIGHT long press function is back after doing the battery removal trick!

Thank you very much!

closed 2020-11-12 by perfecblue