dresden-elektronik/deconz-rest-plugin #3639: Device requeste for VIMAR 14592 Switch

rafuz, 2020-11-12:
## Device

- Product name: Vimar - 14592.0 - 2-way switch IoT connected mechanism
- Manufacturer: VIMAR
- Model identifier: On_Off_Switch_v1.0
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]

rafuz, 2020-11-12:
As said in #3194 right now the switch works but Turn ON/OFF commands are applied as TOGGLE.
Vimar advertises a Scene selection button but no correspondig cluster in zigbee.

https://www.vimar.com/en/int/catalog/product/index/code/14592.0

LeoeLeoeL, 2020-11-13:
Probably Scene button can't be managed under Zigbee. 
It could be the same as for RGB Led. Vimar answered me via email it works under BT only.

Mimiix, 2020-11-13:
@SwoopX ?

SwoopX, 2020-11-22:
No clue what's going on there. One thing that is striking though: cluster 0x0007 (on/off switch configuration) is exposed as client cluster (grey) here, but it should be exposed as server cluster (blue). The server cluster has some attributes to change the switch behaviour only, also if it toggles. This is nothing deconz can change or influence, hence a device specific topic.

From my perspective, only the vendor can explain if and how that behaviour can be changed.

testorp, 2020-12-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

@Mimiix  these of from my device today

LeoeLeoeL, 2020-12-08:
I don't remember cluster 0007 in my device. Even on firmware 2.0
I can't check now because I returned the switch due to the toggle problem.

rafuz, 2020-12-08:
@testorp I have updated firmware today and re-added device and now I have the same clusters as you.
It is added as a light for me too.

rafuz, 2021-01-03:
I don't have any update, I have checked if there is a new firmware but nothing has changed.

LeoeLeoeL, 2021-01-03:
The situation's going worse.
I gave up and migrated alle tue HUE bulbs (14) and one 3A to the HUE bridge.
After that, I had the problem only one time per day so  I didn't solve.

TheBisk, 2021-01-10:
same problem here also with the latest firmware, the switch works but Turn ON/OFF commands are applied only as TOGGLE.
this bug was not present in some previous firmware version.

double1968, 2021-01-27:
I contacted Vimar, they announced that an update of the VIEW Wireless App (v1.2.3) has been released with which it is possible to update the switch to Firmware version 2.01. They invite you to update and verify operation.

testorp, 2021-01-27:
Do you have information as to improvements?
Any release notes

Thx
Paolo

double1968, 2021-01-27:
I updated the toggle switch issue was fixed

monkeypr00f, 2021-01-27:
> I updated the toggle switch issue was fixed. the changelog is not available

Did it works? I updated but in deconz is listed as light and it works just in toggle mode...
If you toggle from the button HA is not updated, if you toggle from HA it toggles without any state.

rafuz, 2021-01-27:
I will update tomorrow and report back.

rafuz, 2021-01-28:
New Firmware 2.01

[image]

[image]

[image]

Toggle bug seems fixed now

testorp, 2021-01-28:
I have returned mine, does it work without load now?

Thx
Paolo

monkeypr00f, 2021-01-28:
I can confirm it works with load! 
But not without, so useful in a traditional scenario, less if the switch is used to load a "remote" scene or similar without a load directly connected.

SwoopX, 2021-01-30:
One remark in this regard, as I've seen the cluster 0x0007 obviously changed from client (grey) to server (blue). Via the blue server cluster, it should be possible to configure the switching behaviour if desired.

TheBisk, 2021-01-30:
with the new firmware 2.0.1 everything is working fine now.

cafedo, 2021-01-30:
> One remark in this regard, as I've seen the cluster 0x0007 obviously changed from client (grey) to server (blue). Via the blue server cluster, it should be possible to configure the switching behaviour if desired.

So, will it be possible integrate completely a Philips Hue Bulb connected to the switch? I mean: leaving the bulb with a constant load and listening the Vimar switch event in order to send to the Philips Hue gateway the virtual “on” and “off” of the bulb..

closed 2021-03-12 by github-actions[bot]