dresden-elektronik/deconz-rest-plugin #3640: How do I set timeout on Xiaomi Aqara motion sensors?

fribse, 2020-11-12:
## Describe the question or issue you are having

I've just gotten a bunch of these very nice compact motion sensors.
They seem to be quite sensitive, and even has some lux sensor in it, as well as a temperature.
It does howver seem to me that the timout for these are way too long, something like 60s.
I want that to be more like 10 or 5 s.
So I went to the deconz interface, opened the cluster, and tried to write to it, but even though the field is marked rw, it just says 'writing' in the interface, but it still stays at 0.
Is it the right place I'm trying to write?

## Screenshots

[image]

## Environment

 - Host system: Intel NUC with PVE
 - Running method: Home Assistant
 - Firmware version: (26580700)
 - deCONZ version: (2.05.84)
 - Device: ConBee II
 - Do you use an USB extension cable: yes USB3
 - Is there any other USB or serial devices connected to the host system? If so: Which? Aeotec zwave usb stick

## deCONZ Logs

## Additional context

Mimiix, 2020-11-13:
Hi,

That's normal behaviour. There is a hack to get it in 5 sec,  but you need to google that.

SAOPP, 2020-11-13:
U can set between 60 and 90 seconds via rest-api.

Smanar, 2020-11-13:
IDK what you are doing with it, but you can use it as presence detection, not mouvement detection, increase the timeout, and turn off the light when "presence" go to "off".

fribse, 2020-11-14:
Ok, got it, thankyou all for the input, very much appreciated. I'm going to redesign the node-red flows to make them handle it instead of the 'old' way.

closed 2020-11-14 by fribse