dresden-elektronik/deconz-rest-plugin #3645: Latest deCONZ release 2.05.87 breaks HA integration

indomus, 2020-11-13:
## Describe the bug
After deCONZ today's update (2.05.87), something strange happens on HA 0.117.06 (latest):

discovered so far:

- xiaomi magic cube is working under phoscon web app, but no device event is passed to HA;
- aqara vibration sensor detects vibrations on phoscon app, but state won't change on HA;
- aqara motion sensor detects motion on phoscon app, but state won't change on HA;
- axiaomi window/door sensor detects opening/close on phoscon app, but state won't change on HA.

If you stop HA, remove and repair the sensors, restart HA, it seems working; after that, if you restart bot deCONZ/HA, it stops working again.

## Steps to reproduce the behavior
-

## Expected behavior
-

## Screenshots
-

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.87
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
nothing relevant

## Additional context
nothing relevant

SAOPP, 2020-11-13:
I think u need Hotfix for v2.5.87, it's already released.

indomus, 2020-11-13:
Tried, no change.

SAOPP, 2020-11-13:
Well, now we need to collect more feedbacks and wait for comments from the developers side also. I have updated and I don't have the described issue of u. I'm running deconz in docker at different platforms and sticks (rpi/x86/raspbee/conbee2).

Mimiix, 2020-11-13:
Hi. Hotfix is already out.

HA needs to fix it on their end. 

@kane610

Kane610, 2020-11-13:
What should be done @Mimiix? What is fixed in deconz and what is decided to not be fixed in deconz and expected to be handled in hass? To me it sounds like an isolated regression in deconz.

Mimiix, 2020-11-13:
It seems the integration breaks, so i have no clue whats going on. Hence me tagging you.

Kane610, 2020-11-13:
I'll be awaiting more details then

Mimiix, 2020-11-13:
Without any logs/details to go on or any probable causes: I'll close this issue. The integration is having issues and it should figure out where it goes wrong.

closed 2020-11-13 by Mimiix

Kane610, 2020-11-13:
You should stay away from making unfounded claims like this

indomus, 2020-11-13:
Kane is right, Mimiix: what changed in the last 24 hrs is deCONZ, not his integration on HA...

Mimiix, 2020-11-13:
As i said: At this point in time there is no proof on whatever was broken on deCONZ side. The HA integration is a user of the API. If that broke somehow: Provide pointers on **what** broke.

I really am not a fan of all this new stuff being included on "stable" and i keep telling that to @manup. However, having issues without any pointers wont help either.

SAOPP, 2020-11-13:
Why close this topic, if at least 1 complaint about this problem has already followed, the topic for this must exist in order to collect statistics and come to a solution. My point of view.

Above, the user writes that the problem occurred after the Deconz update, which in turn means that the problem is in the type of installation / interaction, but not in the HA side. There were no changes in HA regarding integration and work with Deconz at latest releases.

@indomus if ya get back to previous _working_ version of Deconz, the issue is gone?

indomus, 2020-11-13:
Didn't. But I won't do it until i get some evidences about the issue - but I cannot do it now.

Kane610, 2020-11-13:
@indomus what are the state of the devices in hass when this happen, are they unavailable? If they are try using the device refresh service before restarting anything

SAOPP, 2020-11-13:
@indomus In any case, u need to check this, because now your post looks more than 50%, if u return to the previous release and still face this problem, then I think u understand everything yourself in this case...

SAOPP, 2020-11-13:
And by the way, your problem looks like the deconz doesn't have time to start before the start of the HA.

indomus, 2020-11-13:
Can't be this. Tried starting deconz at first and THEN home assistant: same issue.

Kane610, 2020-11-13:
@indomus so what does the logs in hass say?

SAOPP, 2020-11-13:
@indomus ok last, have you tried re-add the integration?

indomus, 2020-11-13:
Will be the very last try to do :) I don't what any problem about entity names (will be there for sure)

indomus, 2020-11-13:
@Kane610 As I said, I'll do that as soon I can - can't now :(

Kane610, 2020-11-13:
You where responding so I thought you where on it

SAOPP, 2020-11-13:
> Will be the very last try to do :) I don't what any problem about entity names (will be there for sure)

in order not to be afraid of problems with entity names, they should be put in order in both Deconz and HA. Okay, the last word in yours.

SAOPP, 2020-11-13:
Also, as I understood u don't use deconz addon right?

indomus, 2020-11-13:
Raspbian, with HA Core on venv and Phoscon deCONZ installed on the same host.

So, I tried with:

```
logger:
  default: info
  logs:
    homeassistant.components.http.ban: warning
    homeassistant.components.deconz: debug
    homeassistant.components.zha: debug
```

nothing, no-thing. when one of the sensors written above changes state, this state is collected on phoscon app but NOT on Home Assistant, and nothing is written on HA logs.

Kane610, 2020-11-13:
You should also add deconz: debug

Mimiix, 2020-11-13:
> Why close this topic, if at least 1 complaint about this problem has already followed, the topic for this must exist in order to collect statistics and come to a solution. My point of view.
>
> Above, [...]

If a user can't provide any logs or pointers there's no reason for a issue to be open. Nevertheless : his report is the only one after hot fix 88 and the user isn't even running it. I am not saying there's no issue, but this issue won't be beneficial as long as no evidence is provided.

indomus, 2020-11-18:
```
Nov 18 10:01:17 raspberrypi hass[11052]: 2020-11-18 10:01:16 DEBUG (MainThread) [pydeconz.deconzdevice] Hall Motion created as
Nov 18 10:01:17 raspberrypi hass[11052]: {'config': {'battery': 98,
Nov 18 10:01:17 raspberrypi hass[11052]:             'on': True,
Nov 18 10:01:17 raspberrypi hass[11052]:             'reachable': True,
Nov 18 10:01:17 raspberrypi hass[11052]:             'temperature': 3100,
Nov 18 10:01:17 raspberrypi hass[11052]:             'tholddark': 12000,
Nov 18 10:01:17 raspberrypi hass[11052]:             'tholdoffset': 7000},
Nov 18 10:01:17 raspberrypi hass[11052]:  'ep': 1,
Nov 18 10:01:17 raspberrypi hass[11052]:  'etag': '62f691bee068cf3a8898f7ef3ad460b2',
Nov 18 10:01:17 raspberrypi hass[11052]:  'lastseen': '2020-11-18T11:58Z',
Nov 18 10:01:17 raspberrypi hass[11052]:  'manufacturername': 'LUMI',
Nov 18 10:01:17 raspberrypi hass[11052]:  'modelid': 'lumi.sensor_motion.aq2',
Nov 18 10:01:17 raspberrypi hass[11052]:  'name': 'Hall Motion',
Nov 18 10:01:17 raspberrypi hass[11052]:  'state': {'dark': False,
Nov 18 10:01:17 raspberrypi hass[11052]:            'daylight': True,
Nov 18 10:01:17 raspberrypi hass[11052]:            'lastupdated': '2020-11-18T08:47:45.018',
Nov 18 10:01:17 raspberrypi hass[11052]:            'lightlevel': 29709,
Nov 18 10:01:17 raspberrypi hass[11052]:            'lux': 935},
Nov 18 10:01:17 raspberrypi hass[11052]:  'swversion': '20170627',
Nov 18 10:01:17 raspberrypi hass[11052]:  'type': 'ZHALightLevel',
Nov 18 10:01:17 raspberrypi hass[11052]:  'uniqueid': '00:15:8d:00:04:9f:e0:ae-01-0400'}

This is one log about one sensor that has updated values on Phoscon but NOT on Home Assistant (  'lastupdated': '2020-11-18T08:47:45.018' is erroneous, so are the other state parameters)
```

SwoopX, 2020-11-18:
Sorry to say that, but this is going nowhere. The information above provided is no relevant log data. It is basically what you get when querying a specific device for its current status via deconz REST API.

> 'lastupdated': '2020-11-18T08:47:45.018' is erroneous, so are the other state parameters

What is erroneous there? Are you aware how the mechanism behind this works?

Don't get me wrong here, I'm still open that deconz might have whatever hickup here, but you're providing very strong indications that the REST API is doing fine and not the cause of the experienced issue.

We need more comprehensive deconz log data over a longer time (like 15 mins) where you did some stuff that causes deconz to fire websocket events.

indomus, 2020-11-18:
No, I'm not aware - but, if on phoscon got the correct values updated "some seconds ago" and starting HA I see, about that device, a 'lastupdated' field with data about time ago, something is going wrong.
But for sure I'm the only one with this issue.

indomus, 2020-11-18:
Btw: can you help me finding the correct deconz log? what I published are just HA's.

Kane610, 2020-11-18:
Can you share a longer bit of log starting from start of hass all the while you play around with one specific device

indomus, 2020-11-18:
> Can you share a longer bit of log starting from start of hass all the while you play around with one specific device
That's the point. NOTHING is written on log while some devices changes (phoscon side) state.

This is a grep log of HA starting:

`Nov 18 13:37:50 raspberrypi hass[15928]: 2020-11-18 13:37:50 INFO (MainThread) [homeassistant.setup] Setting up deconz
Nov 18 13:37:50 raspberrypi hass[15928]: 2020-11-18 13:37:50 INFO (MainThread) [homeassistant.setup] Setup of domain deconz took 0.0 seconds
Nov 18 13:37:50 raspberrypi hass[15928]: 2020-11-18 13:37:50 DEBUG (MainThread) [pydeconz.gateway] Sending "get" "None" to "192.168.1.99 "
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: {'config': {'UTC': '2020-11-18T12:37:53',
Nov 18 13:37:56 raspberrypi hass[15928]:                                          'name': 'pydeconz'},
Nov 18 13:37:56 raspberrypi hass[15928]:                                          'name': 'pydeconz'},
Nov 18 13:37:56 raspberrypi hass[15928]:                                          'name': 'pydeconz'},
Nov 18 13:37:56 raspberrypi hass[15928]:                                          'name': 'pydeconz'},
Nov 18 13:37:56 raspberrypi hass[15928]:                                          'name': 'pydeconz'},
Nov 18 13:37:56 raspberrypi hass[15928]:                                          'name': 'pydeconz'},
Nov 18 13:37:56 raspberrypi hass[15928]:                                          'name': 'pydeconz'},
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Home created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Landing created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Configuration tool 4 created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Daylight created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Kitchen Sink Flooding created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Washing Machine Flooding created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Main Valve Flooding created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Smoke created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Bedroom Ambient created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Bedroom Ambient created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Bedroom Ambient created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Kitchen Window Opening created as
Nov 18 13:37:56 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Bed Vibration created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Bathroom Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Letto (2) created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Hall Motion created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Terrace Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Terrace Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Terrace Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Fridge Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Hall Motion created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Fridge Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Entrance Vibration created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Bathroom Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Landing Motion created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Landing Motion created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Letto (1) created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Bathroom Ambient created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Mi Magic Cube created as
Nov 18 13:37:57 raspberrypi hass[15928]: 2020-11-18 13:37:56 DEBUG (MainThread) [pydeconz.deconzdevice] Kitchen Window Flooding created as
[... cut ...] 4:d9:d2-01"}
Nov 18 13:46:43 raspberrypi hass[15928]: 2020-11-18 13:46:43 DEBUG (MainThread) [pydeconz.websocket] {"config":{"battery":100,"on":true,"reachable":true,"temperature":2500},"e":"changed","id":"30","r":"sensors","t":"event","uniqueid":"00:15:8d:00:04:27:f8:f7-01-0006"}`

as you can see, the last line is a aqara temperature sensor correctly sending measurement. if one of the listed not-working sensors changes their states, you can see that cange on phoscon web app but not on this log.

SwoopX, 2020-11-18:
Yeah, this is more handy. The last lines show that deconz is firing the events. So, based on your reports, HA does not seem to pick them up or they do not reach HA.

@Kane610 as I don't know HA, I'm out at this point. What do you think?

indomus, 2020-11-18:
Plz, read what I wrote.

> The last lines show that deconz is firing the events.

No, non every single event that is collected by deconz gateway - as previously told.

indomus, 2020-11-18:
@Mimiix Please can you reopen the issue?

SwoopX, 2020-11-18:
> xiaomi magic cube is working under phoscon web app, but no device event is passed to HA;
>     aqara vibration sensor detects vibrations on phoscon app, but state won't change on HA;
>     aqara motion [...]

This is what you wrote in the issue description. You mentioned 4 times Phoscon gets updated correctly, but **NOT** HA. Now, Phoscon gets it's updates from deconz REST API, which HA is using as well. The log information above also shows that the deconz REST API is generating the events. Here, we can primarily try to help with anything related to the REST API.

Now, if the story now is: sensors should have updated their state, but this has **NOT** happened in Phoscon, than that's a whole different situation. If that is the case, than please raise a new issue and ensure that this is appropriately expressed.

indomus, 2020-11-18:
> Here, we can primarily try to help with anything related to the REST API.

This is what we're supposing: something changed on rest api on 0.87/0.88 release.

Kane610, 2020-11-18:
> as you can see, the last line is a aqara temperature sensor correctly sending measurement. if one of the listed not-working sensors changes their states, you can see that cange on phoscon web app but not on this log.

That temperature value is probably from a door/window sensor, it is only the internal temperature of the device and nothing used by HA.

Can you while catching the log play around with your magic cube or trigger your door/window sensor a few times to show the main events from those devices. Preferably keep more of the logs intact it would be nice to see the device definitions from the rest api call

indomus, 2020-11-18:
> That temperature value is probably from a door/window sensor, it is only the internal temperature of the device and nothing used by HA.

Partially wrong. I got several Aqara ambient sensor working fine.

> Can you while catching the log play around with your magic cube or trigger your door/window sensor a few times to show the main events from those devices. Preferably keep more of the logs intact it would be nice to see the device definitions from the rest api call

As previsoulsy told, no events are reported while listed sensors changes their states. Only phoscon app shows changes.

Kane610, 2020-11-18:
The devices which are not posting state changes, are they listed during the start up when doing the request for all rest api data?

I don't know if deconz logs is the next step here if there are no logs of sensor events from pydeconz there is nothing from the rest api in deconz. What do you think @SwoopX

indomus, 2020-11-18:
> The devices which are not posting state changes, are they listed during the start up when doing the request for all rest api data?

Yes. That's the very strange thing here. Entities are created. Simply, they don't are keeped updated.

Kane610, 2020-11-18:
> Yes. That's the very strange thing here. Entities are created. Simply, they don't are keeped updated.

This typically means somethings not right with the pairing if they don't signal state updates over websocket.

indomus, 2020-11-18:
on Phoscon they work fine.

Kane610, 2020-11-18:
Yes well, there is nothing I can do based on the current data

SAOPP, 2020-11-20:
> > The devices which are not posting state changes, are they listed during the start up when doing the request for all rest api data?
> 
> Yes. That's the very strange thing here. Entities are created. Simply, they don't are keeped updated.

request a new api key, and re-add deconz integration

indomus, 2020-11-20:
Already done. No change.

SAOPP, 2020-11-20:
let's go back to the old question, did u go back to previous versions?

...And maybe, did u update it with new ones? is the situation changing?

indomus, 2020-11-20:
yep, I tried .84, works like a charm.

SAOPP, 2020-11-20:
ok, what 'bout 2.06.00 / 15.11.2020?

indomus, 2020-11-20:
every version after .85 is buggy

SAOPP, 2020-11-20:
a r u had trying to make a brand new docker container?

SAOPP, 2020-11-20:
I am sorry, there are a lot of messages, which method of installing deconz do u use?

indomus, 2020-11-20:
this one https://phoscon.de/en/conbee/install#raspbian