dresden-elektronik/deconz-rest-plugin #3647: Aqara door sensor doesnt show correct state everytime

swissmaster1, 2020-11-13:
## Describe the question or issue you are having
I'm using Deconz with Conbee II Stick together with Home Assistant. Every lamp is working as expected as well as every switch or motion sensor. Since a few days I'm testing Aqara door sensor. It was no problem to include that sensor in deconz and as well in Home Assistant. It is recognizing also door openings / closings. But my problem is that maybe about every second time the door gets opened the sensor doesn't register and says the door is closed instead of open. When the door stays open for a while the Aqara still doesn't register. When I close the door and open it again it register it again correctly. I don't see a pattern in when it happens or not. About 2m away from the sensor is a zigbee router in form of a philips hue lamp to which the sensor is connected according to the mesh. So it shouldn't be a connection problem.
Does anybody have an idea?

## Screenshots

## Environment
Host system: Synology DS1815+ NAS
Running method: Marthoc Docker container 
Firmware version: 26660700
deCONZ version: 2.05.87
Device: ConBee II
Do you use an USB extension cable: yes USB3
Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

pdcmoreira, 2020-11-21:
I have the same problem, but I think it's actually because of poor hardware quality.

MattWestb, 2020-11-21:
The detector is 2 magnetic metal parts in one vacuum glass tube (or some edel gas). If the glas have some micro cracking its coming oxygen in the tube and the contacts is being very bad fast and can also gluing together. If have getting hard mechanical stress or to hot then souldring the glas can getting micro crackings and is not sealed OK = broken.
The second thing its if mounting it on or near metal that is shielding or is self magnetick it can loosing (magnetic short circuit) its funktion or must having them very near for detecting or opposite if some magnets is in the near.

I using the Aqara door windows contacts sensor as transmitters for my smoke detectors (without the detector elements) and getting them working as real zigbee ones but much cheaper :-)))

closed 2020-12-25 by stale[bot]